jotego/jtcores #500: Update all cores to support new DirectVideo DV1 SPD Infoframe

c0d3h4x0r, 2024-01-19:
To get correct auto-cropping behavior with RetroTink 4K.

jotego, 2024-01-19:
Could you give a bit more of information about this?

c0d3h4x0r, 2024-01-19:
Here is my understanding of it all. I'm not a MiSTer developer, so you should verify my understanding with the experts. I am just a RetroTink 4K owner who is eager to see all cores updated to add this feature.

The RetroTink 4K is a recently-released 4K scaler. The best way to connect MiSTer to it is via MiSTer's HDMI output running in "DirectVideo" mode.

DirectVideo mode disables MiSTer's built-in scaler and sends unscaled HDMI output (which must be 640x240 or higher, for reasons). Most MiSTer cores output content resolutions lower than 640x240, so the core framework apparently adds black borders around the content to make a 640x240 HDMI signal.

For RetroTink 4K to automatically crop away the black bars and get the true content resolution (so it can draw scanlines at the correct scale, etc.), MiSTer must tell it the size of the black borders that got added. To do that, the core must tell MiSTer that info.

MiSTer (unstable builds, and soon, the stable build) asks the core for the info and puts it into the HDMI SPD Infoframe. RetroTink 4K latest official firmware already contains full support for reading that info out of the HDMI SPD Infoframe and using it.

An unstable branch of the MiSTer core template has been created to implement providing the needed info to MiSTer, and a few unstable cores have already been built against it for testing.

Soon, the stable branches of MiSTer and the core template (containing the needed sys framework changes) will be updated to officially add this feature. When that happens, all cores will need to rebuild against the latest template to pick up support for it.

c0d3h4x0r, 2024-01-26:
MiSTer main stable and the core template repo were both updated on January 10, so this work can proceed now.

jotego, 2024-02-01:
I have updated the MiSTer template files.

@c0d3h4x0r, can you check the attached cores, please?

[jtcps1.rbf.zip](https://github.com/jotego/jtcores/files/14121774/jtcps1.rbf.zip)
[jt1942.rbf.zip](https://github.com/jotego/jtcores/files/14121779/jt1942.rbf.zip)

c0d3h4x0r, 2024-02-01:
RetroTink4K auto-cropping does not work with either of those test cores.

Auto-cropping does work with the MegaDrive and TurboGrafx-16 test cores that were made available via Discord. I don't know who built them, but you might want to find out and ask them what they did to make it work.

c0d3h4x0r, 2024-02-01:
You can also talk to birdybro in the MiSTer Discord. They recently updated the Tropical Angel arcade core to enable the auto-cropping feature.

jotego, 2024-02-01:
Good. But can you reply to the first question? Can you tell that this change has an effect at all?

c0d3h4x0r, 2024-02-01:
It's 1:17am here, so I need to get to bed. I'll have more time to test and compare this weekend.

cobhc2019, 2024-02-01:
Auto cropping and auto decimation works fine for me with the CPS1 core on the RT4K and looks great.

1942 doesn't seem to auto decimate or crop for me.

cobhc2019, 2024-02-02:
Ran an update all and then copied the cores over again and 1942 loads now, but doesn't auto crop or decimate like the CPS1 core does.

jotego, 2024-02-02:
Thank you for the feedback. I will release all cores updated to this today. Let's see what the feedback is. From talking to other developers, it looks like what I did is what needs be done so I will close this for now.

closed 2024-02-02 by jotego

cobhc2019, 2024-02-02:
Did some testing with the updated cores and it seems to be a bit of a mixed bag. Kind of burnt out on doing any more testing, as I have to power cycle the MiSTer for every core that the automatic scaling doesn't work on (for some reason), but this is what I have so far. I don't know if there's any link between those that do and don't work with automatic scaling on the RT4K.

1942 works
1943 works
Aliens doesn't work
Bionic Commando doesn't work
Black Tiger works
Bubble Bobble doesn't work
Contra doesn't work
CPS1 no Ionger works
CPS1.5 doesn't work
CPS2 doesn't work
DD doesn't work
DD2 doesn't work
Exed Exes doesn't work
Fast Lane works
Final Round doesn't work 
GNG doesn't work
Gun Smoke doesn't work
Karnov doesn't work
Karate Champ doesn't work

I'll see if I can test more next week if you need me to.

roninniagara, 2024-02-02:
same here. the screen on cps1/2 is split in half. even hitting the crop option on the rt4k doesn't fix it. completely unusable now.

simpsons/tmnt do nothing also.

black tiger works.

Same results. 

(also getting a patreon error when trying to start golden axe?!).

Jotego, if you want to reach out to me on discord or twitter (the handlebar gamer) I can test stuff pretty quickly for you also.

Thanks for all you do! Long time patreon subscriber! (CPS1 days!)

> Did some testing with the updated cores and it seems to be a bit of a mixed bag. Kind of burnt out on doing any more testing, as I have to power cycle the MiSTer for every core that the automatic [...]

jotego, 2024-02-03:
Thank you for testing and reporting. This is a bit of a side quest, to be honest. It looks like there is some core-specific thing missing. I need to get more information about it. I hope this can be automated as I do not want to have to enter specific stuff for each core (or even game) for this feature.

Some games in the same core even draw a a black frame around the game that is independent on hardware signals. That means that cropping derived from hardware blanking signals won't work for all cores. So I think there will always be a need for some configuring on the Tinker side.

cobhc2019, 2024-02-03:
No problem, I don't think anyone expects you to look at this on a core by core basis. Hopefully this can be automated...as when it works, it's really nice with the Tink 4K.

Let me know if you want/need me to test anything else.

paulb-nl, 2024-02-07:
Some of these cores like cps1 have trouble with HDMI and i2c. This error is in the log:
`*** ADV7513 not found on i2c bus! HDMI won't be available!`

This prevents it from sending any HDMI initialization or other data to the ADV7513 chip.