jotego/jtcores #512: jtcps1: Audible pops on sound samples playback.

Left-Empty, 2024-01-26:
Hello,

After today's (January 26th, 2024) update, I can hear a slight "pop" at the beginning of most samples in the CPS1 games I tried so far.

That pop, which sounds like a mike pop, can be better isolated by turning on a low pass audio filter.

The pop seems to be dependent on the sample. For instant, in Knights of the Round, you can hear it very distinctly landing any jump, and in the music. Swinging the sword doesn't produce it though.
In the King of Dragons, it's audible in most of the music tracks.
In Strider, Japanese version, it doesn't happen if you attack once, but is very audible when you spam the attacks. Possibly it hints it might be related to multiple samples playback at once?

Thank you for all your work once again, and cheers!

Left-Empty, 2024-01-31:
Hello Jose,

I just tested it and it improves things in the vast majority of the cases. The single sound effects pops seem to be totally gone in all the CPS1 games I tried, especially those which had issues before.

Two CPS1 games that still stood out to me were Knights of the Round and the King of Dragons, both of which got still some very low pops that can be heard during the massive ammounts of samples played within the background musics of both games.

Also, I had forgotten to check in the previous version, but the infamous track of King of Dragons stage 4 (Code No 04 in the test mode) is still having troubles in both versions of the core: it is being mostly devoured by the cymbal sample, while said sample should really be barely heard.
Cheers!

TesterNes, 2024-02-01:
I can confirm about the thunderous volume by cymbals which can be heard easily at stage 4. And sadly the entire set of drums and cymbals are in other frequency, outputting a lower quality on them in comparison to real hardware.

jotego, 2024-02-01:
Please start a different issue for topics other than the one in the title and add evidence to support the claims and be used as reference.

About the pop sound, I removed a dc filter that could be causing it. @Left-Empty may you test the attached file, please?

[jtcps1.rbf.zip](https://github.com/jotego/jtcores/files/14121218/jtcps1.rbf.zip)

Left-Empty, 2024-02-01:
Hello,

Thank you for your reply. To me it seems to be totally fixed, but I listened so much to the games in the past 24 hours I'm not sure my ears aren't muddy anymore XD

Closing, and reopening the old cymbal issue that got closed last week!

closed 2024-02-01 by Left-Empty