jotego/jtcores #514: select button doesn't work although game runs (MiSTer)

jtmiki, 2024-01-26:
select button doesn't work although game runs
- [ ]  Super Hang-on
- [ ] The Simpson's
- [ ] Time Scanner

Commit: 923324f

danmacpherson, 2024-01-29:
Select also isn't working for Bubble Bobble on Analogue Pocket -- doesn't insert coin.

jotego, 2024-01-29:
> Select also isn't working for Bubble Bobble on Analogue Pocket -- doesn't insert coin.

It does work. Read [this](https://github.com/jotego/jtbin/wiki/Why-I-only-see-the-on-screen-credits%3F)

danmacpherson, 2024-01-30:
Hi @jotego,

So I have the beta key properly installed and it works with other cores (The Simpsons, TMNT).

I tried reinstalling the JTBUBL core with the beta key and it seems to be installed properly.

Before the recent update, the core seemed to be working fine for me (although I experienced the graphical glitches as per #455) and I was able to insert coin with SELECT.

Since the update, I can't insert coin like I used to with SELECT, even after reinstalling the core.

I can press SELECT + A and the credits appear, so the SELECT button works.

The [FAQ](https://github.com/jotego/jtbin/wiki/Why-I-only-see-the-on-screen-credits%3F) you posted mentioned the following:

> If you can see the game running but cannot insert coins then you are running a closed version of the core

1. What does a "closed version" of the core mean? How is that different from Public, Beta, and Private versions?
2. I didn't see in the FAQ how to resolve the issue. What steps should I take to resolve this?

FYI, it seems to be the same issue as #520

jotego, 2024-01-30:
Either manually copy the files from [JTBIN](https://github.com/jotego/jtbin) or use an updater. Try doing it on a clean SD card. Maybe things are not being copied the way you think. It's probably better to use Discord for troubleshooting this as this is not a _real_ issue with the files. I am sorry it is not working more easily for you this time.

danmacpherson, 2024-01-30:
Somehow managed to get it to work after reinstalling the core again. Not sure how I did it.

Have also joined the discord.

> I am sorry it is not working more easily for you this time.

No stress, dude. You're doing great work!

commit 2541be74 (2024-01-30): shanon: fixes select button not working by default as coin. Fixes #514
closed 2024-01-30 by jotego