jotego/jtcores #515: no SPRITES in latest version on POCKET.

NeilGSPGUK, 2024-01-26:
there are no Car / Road Side Icons in the latest build.

jotego, 2024-01-26:
Are you talking of Out Run? I do get the cars and the signs. But I think that issue may happen occasionally. Try powering off and on completely the Pocket. Let me know if you find some pattern to it. I am not able to replicate this problem at the moment.

NeilGSPGUK, 2024-01-26:
all roms in the OUTRUN Core, I updated today and just get road, in all roms inc Turbo Outrun. didnt backup previous version. is there a way to install previous version? version installed INTERNAL 2024-01-26

jotego, 2024-01-27:
Old versions are kept in [JTBIN](https://github.com/jotego/jtbin/commits/master/pocket/raw/Cores/jotego.jtoutrun/jtoutrun.rbf_r).

This could be a timing issue related to some RAM modules in the Pocket. Would you be willing to test some core versions compiled with different timing to test this hypothesis?

NeilGSPGUK, 2024-01-27:
Yes no problem. Just let me know what I can do to help

NeilGSPGUK, 2024-02-02:
@jotego FYI rolled back to the version 15.0 2023-11-24 and Sprites are all present.

NeilGSPGUK, 2024-02-06:
FYI I just updated, and the new version 659baa4 2024-02-02 again has the missing sprites @jotego .

Dawbo66, 2024-02-09:
As requested from Discord, I'm having the same issue , no graphics on outrun or turbo outrun, (big al) from discord, I can do any test for you @jotego .

Dawbo66, 2024-02-10:
After updating with jtfriday_240209_pocket the game appears to be working as it should, will report back if any update.

jotego, 2024-02-10:
There was no relevant change justifying a _magical fix_ here in the last release.
There is either a sensitivity to the clock phase in the external memory interface or a combinational logic glitch somewhere. Making this feature unstable for some users.

c-hri-s, 2024-02-10:
Confirm the latest released fixed it for me too

SkyNIC, 2024-02-10:
Only I have glitched graphics in turbo outrun?
[image]
[image]
[image]

NeilGSPGUK, 2024-02-10:
> Confirm the latest released fixed it for me too

Fixed for me also thanks!

GoldZabu, 2024-03-03:
They seem to be gone again in the March 1 update.

Dawbo66, 2024-03-03:
> They seem to be gone again in the March 1 update.

I've tried starting Outrun several times on the Pocket and all seems good, a couple of months ago there was an issue, I'm wondering if something's creeping back on, maybe delete and put back on ensuring any saves are also deleted.

JanGunheD, 2024-03-04:
The sprites are also missing on my pocket.

trza49, 2024-03-04:
Yeah had this issue on the Pocket. I just deleted the core and assets and reloaded but still the same.

Dawbo66, 2024-03-04:
Try going back to jtfriday_240209_pocket and see if its resolved..

trza49, 2024-03-05:
> Try going back to jtfriday_240209_pocket and see if its resolved..

Where do I find this file please? I'm new to GitHub.

Dawbo66, 2024-03-05:
> > Try going back to jtfriday_240209_pocket and see if its resolved..
> 
> Where do I find this file please? I'm new to GitHub.

There's here https://www.patreon.com/jotego/posts look for jtfriday_240209_pocket on the 9th feb or https://github.com/jotego/jtbin

JanGunheD, 2024-03-05:
> Try going back to jtfriday_240209_pocket and see if its resolved..

Yes, that fixes it. I used jtfriday_240216_pocket.zip.

Dawbo66, 2024-03-05:
Outrun is a bit temperamental at the moment for some users.

46VR46, 2024-03-12:
Hi everyone.  I still have this issue.  Even after formatting my card and starting again.  If I go back to download jtfriday_240216_pocket.zip version, can I confirm that once I download that, I can go into the core and asset folder and copy and past these outrun core and asset folder into my card (in the same positions) and that's it?  I don't need to change any other files on my card?  And this should work.  Thanks for your help guys.

Dawbo66, 2024-03-13:
Yes just overwrite when asked.

46VR46, 2024-03-13:
> Yes just overwrite when asked.

Hey thanks so much for the fast reply.....  Done but no dice!  So I dropped those two folders from the jtfriday_240216_pocket.zip version and overwrote the folder that were on my SD Card.  Now when I go into the outrun core I get the  message "Error in framework missing 'ROM' ID(1)". Does that mean I need to link something or reset where the AP is no looking for the core.  I know it must just be one step I am missing here.  Thanks.

Dawbo66, 2024-03-13:
Have you deleted the rom? If not the rom isn't correct for the core you replaced, you would need to generate the rom again,. Check if the rom is in assets/ common. Does turbo Outrun work ?

46VR46, 2024-03-13:
> Have you deleted the rom? If not the rom isn't correct for the core you replaced, you would need to generate the rom again,. Check if the rom is in assets/ common. Does turbo Outrun work ? Any issues delete the files including the saved file for Outrun and run updater again.

haha.  Gold.  Inset one of those little slap face emojis here!  Thanks so much.  I knew it was going to be something stupid that I was missing.  100% correct.  Place the rom back in there and now working like a charm.  I think I was just thinking that the rom was in that jtfriday_240216_pocket.zip and didn't think about having to place it in their manually as normally I would use pupdate which would normally download and insert the rom for me.  So good pickup and I appreciate it.  Cheers.  Hopefully this will help someone else have the same issue.

Dawbo66, 2024-03-13:
This issue is known, I had it myself, normally over writing the files wouldn't delete the rom because there isn't one in the download, as you said, but now it's working so all good..

trza49, 2024-03-13:
Still no luck for me. I don't know what's going on. Since the latest FW update Outrun refuses to work properly. I've deleted, updated and replaced everything I could and I still get the same outcome, no sprites.

Dawbo66, 2024-03-13:
Probably best to revert back to an order core for Outrun.

itsTrifle, 2024-03-15:
Just commenting as requested in discord that I have the missing sprites issue too @jotego

jotego, 2024-03-22:
I have prepared a file with better timing constraints. Please test it and report whether it works or not for you: [jtoutrun_4f713435.zip](https://github.com/jotego/jtcores/files/14719878/jtoutrun_4f713435.zip). This version corresponds to git commit 4f713435.

If this is not enough, I will apply the clock shift strategy to the CRAM clock. That is the same approach we use for the SDRAM clock to make it work across all devices.

Dawbo66, 2024-03-22:
@jotego out of the 20 times i loaded this up, it loaded 20 times correctly on freeplay, no graphics missing. after the last update i did have to go back to a previous version. (jtfriday_240216_pocket.zip)

jotego, 2024-03-22:
> @jotego out of the 20 times i loaded this up, it loaded 20 times correctly on freeplay, no graphics missing. after the last update i did have to go back to a previous version. (jtfriday_240216_pocket.zip)

Let me clarify:

- the file [jtoutrun_4f713435.zip](https://github.com/jotego/jtcores/files/14719878/jtoutrun_4f713435.zip) works correctly 
- latest file in JTBIN, does not
- an older version (240216) worked too

Is that correct?

Dawbo66, 2024-03-22:
> Let me clarify:
>
> * the file [jtoutrun_4f713435.zip](https://github.com/jotego/jtcores/files/14719878/jtoutrun_4f713435.zip) works correctly
> * latest file in JTBIN, does not
> * an older version [...]

yes this is correct for me anyway.

itsTrifle, 2024-03-22:
> I have prepared a file with better timing constraints. Please test it and report whether it works or not for you: [jtoutrun_4f713435.zip](https://github.com/jotego/jtcores/files/14719878/jtoutrun_4f713435.zip). This version corresponds to git commit [4f71343](https://github.com/jotego/jtcores/commit/4f7134354b3c33952f032d70e10cc3521ea09295).
> 
> If this is not enough, I will apply the clock shift strategy to the CRAM clock. That is the same approach we use for the SDRAM clock to make it work across all devices.

I can confirm the sprites display correctly for me with this build

Dawbo66, 2024-03-22:
> Let me clarify:
>
> * the file [jtoutrun_4f713435.zip](https://github.com/jotego/jtcores/files/14719878/jtoutrun_4f713435.zip) works correctly
> * latest file in JTBIN, does not
> * an older version [...]

[image]

Just switched the Pocket on and tried outrun and had No sprites and green lines across the screen, Tried switching Pocket off and on, same result, put jtfriday_240216_pocket.zip back on, worked first time, so I'm not sure why it went that way.

JanGunheD, 2024-03-23:
> I have prepared a file with better timing constraints. Please test it and report whether it works or not for you: [...]

Works fine on my Analogue Pocket :-).

manichispanic999, 2024-04-08:
> I have prepared a file with better timing constraints. Please test it and report whether it works or not for you: [...]

This fixed my sprite issue. Thanks!

GoldZabu, 2024-04-27:
The April 26th update seems to have broken sprites again.

hansfbaier, 2024-05-05:
I have the same issue in the MiSTeX port.

jotego, 2024-05-06:
> I have the same issue in the MiSTeX port.

Unless the MiSTeX uses PSDRAM, it is not the same problem.

hansfbaier, 2024-05-06:
> Works fine on my Analogue Pocket :-).

Can you point me to a diff, to see an example of how the changes would look like?

jotego, 2024-05-07:
```
 # setup/hold time requirements for CRAM 0/1
 # CRAM0
-set_output_delay -clock cram0_clk -max 2 \
+# tSP^M
+set_output_delay -clock cram0_clk -max 3 \^M
     [get_ports {cram0_a[*] cram0_dq[*] cram0_adv_n cram0_cre \
                 cram0_oe_n cram0_we_n \
                 cram0_ub_n cram0_lb_n }]
 
+# tCSP^M
 set_output_delay -clock cram0_clk -max 4 \
     [get_ports {cram0_ce0_n cram0_ce1_n  }]
 
-set_output_delay -clock cram0_clk -min -4 \
+# tHD, hold time^M
+set_output_delay -clock cram0_clk -min -2 \^M
     [get_ports {cram0_a[*] cram0_dq[*] cram0_adv_n cram0_cre \
                 cram0_ce0_n cram0_ce1_n cram0_oe_n cram0_we_n \
                 cram0_ub_n cram0_lb_n }]
 
+# tkOH in data sheet, output hold time^M
+set_input_delay -clock cram0_clk -min -2 \^M
+    [get_ports {cram0_dq[*] }]^M
+^M
 # CRAM1
-set_output_delay -clock cram1_clk -max 2 \
+set_output_delay -clock cram1_clk -max 3 \^M
     [get_ports {cram1_a[*] cram1_dq[*] cram1_adv_n cram1_cre \
                 cram1_oe_n cram1_we_n \
                 cram1_ub_n cram1_lb_n }]
@@ -56,6 +65,9 @@ set_output_delay -clock cram1_clk -min -2 \
                 cram1_ce0_n cram1_ce1_n cram1_oe_n cram1_we_n \
                 cram1_ub_n cram1_lb_n }]
 
+# tkOH in data sheet, output hold time^M
+set_input_delay -clock cram1_clk -min -2 \^M
+    [get_ports {cram1_dq[*] }]^M
```

Toryalai1, 2024-05-09:
Turbo outrun has graphical glitches.
Reproduce:
- Start game
- Go into gameplay

SkyNIC, 2024-05-10:
MiSTer port have same issue
[image]

JanGunheD, 2024-06-16:
In the newest version, the problem came back.

jotego, 2024-06-16:
> Turbo outrun has graphical glitches. (Mister fpga) Reproduce:
> 
>     * Start game
> 
>     * Go into gameplay
> 
> 
> ![image](https://private-user-images.githubusercontent.com/82502378/329305599-b7f003ed-0213-4598-8b7b-120d966752b0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg1NTQ2MjQsIm5iZiI6MTcxODU1NDMyNCwicGF0aCI6Ii84MjUwMjM3OC8zMjkzMDU1OTktYjdmMDAzZWQtMDIxMy00NTk4LThiN2ItMTIwZDk2Njc1MmIwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjE2VDE2MTIwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAyZGFkZGYwY2RkOTQ5N2M0NzI5ODAwMDE3NTE5ZDJkNmRiMjllZmY3ZTFkYWZhNjFkZmJmOTBmMTcyYjA5MWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.LsAOTbnPfOhBSycEBybZ9kiqxJWNotm5I7Wwf0gb7JQ)
> 
> ![image](https://private-user-images.githubusercontent.com/82502378/329305631-ad7979ca-72b3-4193-afec-d4594cb6a51f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg1NTQ2MjQsIm5iZiI6MTcxODU1NDMyNCwicGF0aCI6Ii84MjUwMjM3OC8zMjkzMDU2MzEtYWQ3OTc5Y2EtNzJiMy00MTkzLWFmZWMtZDQ1OTRjYjZhNTFmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjE2VDE2MTIwNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ5YTBkMzNmMDIyNTUwOWUxMjNhZDY3Y2M0OGM5OTEzZDc0MTA4OTU5NmYxZDhhMzUwMWNkNTc2NDg2YzYwODMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.G04vd408MHHeouXuPmwoel2vbRflaClRmLn8gd9Fr6o)

That is a different issue from the one in this thread

sir-codealot, 2024-07-20:
Have the same issue with the current version 8533440 . Also it worked fine for me since November 2023 with every release. but with the latest one I just see the road.

Buyukcaglar, 2024-10-08:
Hi, 

I have the exact same issue with the latest version d2f3856 (b.date: 24.10.04). No Car / Road Side sprites etc.

Dawbo66, 2024-10-08:
Use an older version, it's a known issue but I don't think it's being worked on at the moment.

JanGunheD, 2024-10-23:
Yes, the problem is back again. Just saying it because it got fixed every time before and then comes back with the next version.

Dawbo66, 2024-10-23:
It hasn't been worked on for a while, so any "fixes" is probably the timing issues on pocket variants as it works, then stops, then works again that's why I don't update the core as the one I use works all the time.

Buyukcaglar, 2024-11-03:
Hi,

It seems to be, with the core version 090d35d (2024-10-25) (and along with FW ver 2.4) the problem is solved and perfectly playable in my unit.

Thanks.

ctafield, 2024-11-14:
No sprites for me in version 4731644 (2024-11-09). Using a glow in the dark Pocket if that makes any difference.

Buyukcaglar, 2024-11-14:
@ctafield can you please try previous version (which I've attached as zip file, just unpack ZIP to your SD Cards's \Cores folder, and overwrite all files when asked.

[jotego.jtoutrun ver 090d35d 2024-10-25](https://github.com/user-attachments/files/17758087/jotego_jtoutrun_v090d35d_20241025.zip)

> No sprites for me in version [4731644](https://github.com/jotego/jtcores/commit/4731644682824308af4c603786300cb0670d64d6) (2024-11-09). Using a glow in the dark Pocket if that makes any difference.

Kypck, 2025-01-06:
Latest build and the one provided by @Buyukcaglar Buyukcaglar are working on my first edition Pocket.

ctafield, 2025-01-06:
Still missing for me on my GITD, sadly.

jotego, 2025-01-06:
I think I am going to retire the Pocket version of the core until I can find a solution that works for everyone.

Di35el, 2025-01-06:
fwiw whatever you did in jtoutrun_4f713435.zip Jotego has been working for me no problem. After every pupdate I do tend to lose sprites, but I just overwrite it with 4f713435.zip and everything is good again 👍🏻

ctafield, 2025-01-06:
Sorry for the delay in checking it, but can confirm the version that [Buyukcaglar](https://github.com/Buyukcaglar) suggested **does** work on my GITD too.

[17 more replies]