jotego/jtcores #516: Hyper sports speech

Dawbo66, 2024-01-27:
The speech in hyper sports sounds like the commentator is under water and not understandable, sounds very garbled.

jtmiki, 2024-01-30:
Core | game | issue |
----|---|-----------|
roadf | hyper sports | sounds like the commentator is under water and not understandable, sounds very garbled. |

commit: 48980d9

jotego, 2024-01-30:
Related to #136

jotego, 2024-01-30:
I have implemented the software-controlled filters that the board has. Not with the exact cut-off frequency but with a noticeable effect. In the future, when I tackled #402, I will look into implementing the exact filter bandwidth. 
Done in 414e5cf

jotego, 2024-05-24:
The cheering at the end of the swimming competition sounds distorted. There is no clipping indicated by the LED. Commit fcfa7dfe

jotego, 2024-06-22:
seems fine in a1ffbab

jotego, 2024-06-22:
The _audio_ section of the mem.yaml was using `unsigned` instead of `dcrm` for the resistor DAC channel. Hence, the dc component was not removed.

commit cf2d9038 (2024-06-22): roadf/sbaskt/track: RDAC sound fixed. Fixes #516
closed 2024-06-22 by jotego