jotego/jtcores #517: parodius: renombrar o borrar linea BRIT

jotego, 2024-01-27:
La salida **BRIT** en colmix se confunde con la entrada **BRIT[2..0]**. No hay que duplicar el nombre de señales distintas. Los esquemáticos están ahora dentro de `simson/sch/parodius`, porque parece que el juego puede compartir core con _The Simpsons_

Tareas:
- [ ] Renombrar el cable **BRIT** o borrar si no va a ningún sitio.
- [ ] Volver a pasar las reglas de diseño

etriana85, 2024-01-30:
[image]
[image]
[image]
[image]
[image]
[image]

etriana85, 2024-01-30:
A continuación adjunto una imagen del orden del bus de datos que viene de _main_. Es el custom 053248, su diseño está basado en información de la web [https://twitter.com/RndMnkIII/status/1220309920001732608](url) y en la extracción del sch. 
El punto es que no coincide el orden del bus al pasar por la RAM.
[image]

etriana85, 2024-01-30:
[image]
[image]
[image]

jotego, 2024-01-31:
Que no coincida el orden en el custom con la memoria no tiene consecuencias funcionales pero fue mala práctica por parte de los diseñadores.

closed 2024-01-31 by jotego