jotego/jtcores #531: Exed Exes: is second background correct?

jotego, 2024-02-02:
Some tiles in the background do not look right. Compare with PCB in second stage.

[image]
[image]
[image]
[image]
[image]
[image]

etriana85, 2024-02-05:
II have reached level 4, and I never saw the background that appears in the attached images. I checked on youTube a video that seemed to be played from the board (video 1) by the quality of the video, and another one that had the title "mame" (video 2) and I don't see those backgrounds either. In the third video the background does appear. The background appears at minute 1.21, very close to the time of the video I made playing from mame (min 1.16). You start to see the red background (of the images where the error is, and that you see in the video 3), just after the hornet appears. I would think that there is another version or maybe a bootleg. 
[image]
[image]
[image]
[image]
[image]
[image]

YouTube 1
https://youtu.be/V3uHGUwXJJg?si=3rhyktErPJEMkwOQ
YouTube 2 mame
https://youtu.be/MUhQSXOLCeA?si=ugvVBxX6tr4UhOcJ
YouTube 3*
https://youtu.be/tFCtTBNRR_M?si=kSVCyvIDWWtcJ_-z

jotego, 2024-02-09:
The code was designed after a bootleg PCB. @etriana85 thinks that the bootleg had different backgrounds. Maybe the second background tile codes are different?

gyurco, 2024-02-09:
It acts as a completely different layer from stage 2, strange. Maybe issue with the map addressing?

gyurco, 2024-02-09:
A good point to compare:
https://youtu.be/tFCtTBNRR_M?t=133

jotego, 2024-02-09:
In that video the land around the red river looks well shaped, whereas in the core it has sharpen edges as though the tiles were not flipped correctly

gyurco, 2024-02-09:
Seems not just flip, but wrong tile also, see the river on the left middle, around the skull.

gyurco, 2024-02-09:
Well, in MAME, there's not even a river at this place.

gyurco, 2024-02-09:
Maybe rom2_data is not always ready at heff[3:0]==0?
There's no check for rom2_ok.

jotego, 2024-02-10:
> Well, in MAME, there's not even a river at this place.

I could not see the river either in MAME. Our board is not booting up anymore (these old PCBs are not reliable at all) so we cannot test it there.

jotego, 2024-02-10:
The river scene is fixed after the PR. Thanks @gyurco 
[image]
[image]
[image]
[image]
[image]
[image]

closed 2024-02-10 by jotego