jotego/jtcores #535: Most MiSTer jtcores break experimental DV1 support and put MiSTer into a bad state

c0d3h4x0r, 2024-02-07:
This is a reliable repro, starting from a powered-off state, using `MiSTer_unstable_240203_16710d` and the latest stable `menu.rbf` (from 2023).

`MiSTer.ini` has `direct_video=1` and `video_mode=8` in the default section, but it has `direct_video=0` for the `[menu]` section.

**Repro Steps:**
1. Power on.
2. `menu.rbf` loads. RetroTink-4K OSD correctly displays 1920x1080. Good.
3. Load `MegaDrive_unstable_240127_16710d.rbf` and load _Sonic the Hedgehog_. RetroTink-4K OSD correctly displays `240p (MegaDrive)` and correctly auto-decimates and auto-crops the image. Good.
4. RetroTink-4K's OSD diagnostic also displays the SPD Infoframe data. Make a note of it.
5. Open MiSTer OSD >> Core >> select the jotego arcade core "Quartet".

**Actual Result:**
RetroTink-4K does not detect that the video source has changed (its OSD still displays `MiSTer 240p (MegaDrive)` and still displays the same SPD Infoframe data), and it does not correctly auto-crop or auto-decimate the image. It appears that MiSTer is still sending the SPD infoframes for the MegaDrive core, despite it no longer being the running core.

**Expected Result:**
RetroTink-4K should detect a video source change, see a new SPD infoframe, and stop applying the MegaDrive core's `DV1` settings.

**Repro Steps (continued):**

6. Open the OSD >> Reboot (warm, not cold) >> arrive back at `menu.rbf`.

**Actual Result:**
RetroTink-4K detects a video source change and displays `HDMI 1080p` in its OSD, but the image is cropped and scaled incorrectly. RetroTink-4K appears to still be applying the MegaDrive core's `DV1` settings.

**Expected Result**:
RetroTink-4K should stop applying any auto-decimation or auto-cropping because the `menu.rbf` core is explicitly configured to not use DirectVideo.

**Repro Steps (continued):**

7. Power off MiSTer, and power it back on.
8. Repeat repro steps 1-5.
9. Open the OSD >> Reboot (Cold Reboot) >> arrive back at `menu.rbf`.

**Actual Result:**
RetroTink-4K detects a video source change and displays `HDMI 1080p` in its OSD, but the image is cropped and scaled incorrectly. RetroTink-4K appears to still be applying the MegaDrive core's `DV1` settings.

**Expected Result**:
RetroTink-4K should stop applying any auto-decimation or auto-cropping because the `menu.rbf` core is explicitly configured to not use DirectVideo.

**Final observation**:
Most of the jtcores cause MiSTer to end up in a permanently bad state in which it is "stuck" using stale `DV1` data. The only way I've found out of this state (so far) is to hard power cycle MiSTer. I have not (yet) been able to repro this same issue with any non-Jotego cores.

c0d3h4x0r, 2024-02-07:
**Additional notes:**

Does not repro when I follow the same steps but use an official DV1-updated core (ex: "Tropical Angel" arcade) instead of "Quartet".

Does not repro when I follow the same steps but use an official non-DV1-updated core (ex: NeoGeo) instead of "Quartet".

birdybro, 2024-02-07:
This comment from @paulb-nl is likely helpful for jotego:

https://github.com/MiSTer-devel/Main_MiSTer/issues/808#issuecomment-1931978230

birdybro, 2024-02-07:
In sys.tcl the following line is missing from Jotego's fork of the framework:

[image]

jotego, 2024-02-07:
> In sys.tcl the following line is missing from Jotego's fork of the framework:

That explains a lot! Thanks!

c0d3h4x0r, 2024-02-07:
Nice catch!

jotego, 2024-02-07:
I have applied the change in f32092a970399cd86189891d53faa7d84e2e2cfb
I checked the files again and I could not see relevant differences with the template. Again, I am not modifying the core logic at all for this. I am assuming the changes come from the MiSTer subsystem.

c0d3h4x0r, 2024-02-07:
If it will help, I can test a private build against my RetroTink-4K and report back.

mikechi2, 2024-02-07:
Hey @jotego , if you like, I'm more than happy to send you a RT4K. Thanks a lot for taking a look at this.

jotego, 2024-02-07:
Thanks, @mikechi2, I am expecting to borrow one next week. Are you involved in the development of RT4K?

mikechi2, 2024-02-07:
> Thanks, @mikechi2, I am expecting to borrow one next week. Are you involved in the development of RT4K?

Yes you could say that ;)

I just sent you a DM on Twitter since we follow each other.

asturur, 2024-02-08:
Are you involved? Slightly

jotego, 2024-02-09:
All MiSTer files updated today in _JTBIN_. Let me know how it goes.

jotego, 2024-02-09:
[One user reports it working](https://twitter.com/T4YK_68000/status/1755948015666467107)

closed 2024-02-09 by jotego

c0d3h4x0r, 2024-02-09:
Looks great on my end! Thank you!