jotego/jtcores #540: Latest update breaks scaled output on hdmi on MiSTer for several cores

mapfde, 2024-02-10:
Hi,

the following games no longer work properly on my MiSTer stack (analog i/o, de10-nano, usb board). Analog output at 15KHz is fine, but the scaled output on hdmi has a wrong refresh rate. I have seen 100Hz to 180Hz depending on the video_mode setting. It seems like the scaler gets confused.

Affected cores:
Outrun (see attached image)
Street Fighter
Fast Lane
Roc'n Rope
Side Arms
Double Dragon 2

[image]

jotego, 2024-02-10:
Thank you. Yes, I can see the OSD data wrong. Does it prevent the image from appearing on your screen?

mapfde, 2024-02-10:
Yes, my monitor shows "out of range" and shuts down after a few seconds.

BTW: I double checked if the problem came up with the latest update. I downloaded the prior version of the outrun rbf file from this github and the problem is not there. All other settings and files stayed the same.

jotego, 2024-02-10:
7bb0b11 ok
5c84ab6 bad

But the change list does not show anything relevant:

```
g ld 7bb0b112 5c84ab6
cores/bubl/cfg/macros.def
cores/bubl/hdl/jtbubl_game.v
cores/bubl/hdl/jtbubl_sound.v
cores/simson/ver/parodius/sim_inputs.hex
```

mapfde, 2024-02-10:
Yes, this had me confused as well. Maybe you changed something in the framework or the build process?

jotego, 2024-02-10:
Framework and build workflow changes would appear in the change list too...

jotego, 2024-02-10:
Did you find a core from the save version in which it did not happen?

mapfde, 2024-02-10:
I am not sure I understand your question correctly. With the latest version, only the cores mentioned in the issue have the wrong refresh rate. All other cores work fine in the latest version. 

I did not roll back all affected cores to one version earlier, I only tried outrun. I can test the others as well if you like.

jotego, 2024-02-10:
Right. I can see that **1942** is fine for the same commit.

jotego, 2024-02-10:
Disabling all constraint exceptions in _sys_top.sdc_ and leaving just:

 ```
create_generated_clock -name SDRAM_CLK -source \
    [get_pins {emu|pll|pll_inst|altera_pll_i|general[1].gpll~PLL_OUTPUT_COUNTER|divclk}] \
    -divide_by 1 \
    [get_ports SDRAM_CLK]
    # Specify root clocks
create_clock -period "50.0 MHz" [get_ports FPGA_CLK1_50]
create_clock -period "50.0 MHz" [get_ports FPGA_CLK2_50]
create_clock -period "50.0 MHz" [get_ports FPGA_CLK3_50]
create_clock -period "100.0 MHz" [get_pins -compatibility_mode *|h2f_user0_clk]
create_clock -period "100.0 MHz" [get_pins -compatibility_mode spi|sclk_out] -name spi_sck
create_clock -period "10.0 MHz"  [get_pins -compatibility_mode hdmi_i2c|out_clk] -name hdmi_sck

derive_pll_clocks
derive_clock_uncertainty

# Decouple different clock groups (to simplify routing)
set_clock_groups -exclusive \
   -group [get_clocks { *|pll|pll_inst|altera_pll_i|*[*].*|divclk}] \
   -group [get_clocks { pll_hdmi|pll_hdmi_inst|altera_pll_i|*[0].*|divclk}] \
   -group [get_clocks { pll_audio|pll_audio_inst|altera_pll_i|*[0].*|divclk}] \
   -group [get_clocks { spi_sck}] \
   -group [get_clocks { hdmi_sck}] \
   -group [get_clocks { *|h2f_user0_clk}] \
   -group [get_clocks { FPGA_CLK1_50 }] \
   -group [get_clocks { FPGA_CLK2_50 }] \
   -group [get_clocks { FPGA_CLK3_50 }]
```

Still creates a file with no timing warnings but wrong data reported. Compiling with seed 0 instead of seed 30418 results in right data being output. I can only think of this is a Quartus compiler glitch. I will regenerate the files again.

jotego, 2024-02-10:
I have changed the build script so the first seed value tried is zero. The only thing special about seed 0 is that it is the one I use when developing and probably the one used by _official_ MiSTer builds. What Quartus does with the _seed_ is unknown.
I will update the files next Friday. You can grab them from [here](https://github.com/jotego/jtcores/actions/runs/7854676875) before if you want to try. I do not think I can prevent this error from the core side so I will close this issue as _not planned_. Thanks for reporting it.

closed 2024-02-10 by jotego

mapfde, 2024-02-10:
I confirm that the new build from https://github.com/jotego/jtcores/actions/runs/7854676875 fixes the issue.

birdybro, 2024-02-10:
The seed is being used for an initial starting point for the fitter algorithm which is simulated annealing --> https://en.wikipedia.org/wiki/Simulated_annealing

It's just a random point to start from. I don't think I've seen such a high number used for seeding typically. We just always use 0 unless the core is very packed and needs to do a design space exploration for multiple seeds.

jotego, 2024-02-10:
There is nothing special in the seed number. I assign them randomly when compiling until a compilation meets timing contraints. The only special value is zero because it is the default one, so it is the one you see more often. Still, making a change in the design also acts as sort of randomizer. The seed value has only shown a design weakness.