jotego/jtcores #541: simson: crop image by 16 pixels all around

jotego, 2024-02-10:
both _The Simpsons_ and _Parodius_ draw a 16-pixel frame around the image. Move the blanking signals so this border gets cropped

jotego, 2024-05-03:
This border might be related to the image looking stretched in (too tall) in the Analogue Pocket.

jotego, 2024-05-03:
The games draw a black frame in software making the effective screen size smaller:

Game       | Width
-----------|----------
Vendetta   | 304
Simpsons   | 288
Parodius   | 288

The core outputs 320 (see JTFRAME_WIDTH in macros.def). Simpsons/Parodius are off by 32 pixels, Vendetta is off by 16. 

JTFRAME_WIDTH is used to set the line buffer length in MiST(er) but MiSTer scaler is only sensitive to the actual blanking video signals. The Analogue Pocket has the information in one JSON file but it also uses blanking signals. 

Questions:

- Is the black frame symmetric? (8/16 pixels blank left and right of screen) or is it only on one side or different lengths on each side?
- What does the Analogue Pocket scaler do if the width is changed in the JSON files? (without changing the core video signals)

rp-jt, 2024-05-09:
Completed in bc076008

closed 2024-05-09 by rp-jt