jotego/jtcores #547: vendetta: encontrar conexión NCPU

jotego, 2024-02-12:
En la hoja de IO del esquemático de Vendetta hay una señal llamada **NCPU**. Esta señal debe venir de uno de los dos chips de objetos. Localízala y añádela a los esquemáticos en simson/sch/vendetta

[image]

etriana85, 2024-02-14:
[image]
[image]

etriana85, 2024-02-14:
- [ ] Verficiar que realmente CAB2 del G13 acaba en el pin COL7 del J13
Sí, la señal *COL7*, pasa por el buffer *H8* y llega al pin 107 del *J13*

- [ ] Para hacerlo más fácil, llama a las señales que entran a la ROM I16 GFXROMA, sin añadir la B nunca. Para las señales que van de COLx a H8, puedes mantener el nombre COLx que tiene el pin directamente. Así se lee más claro mirando H8. Porque ahora salen de COL llamándose GFXROMA13 y acaban en el pin A15 de la GFX ROM, y eso es confuso.
Así esta bien?
[image]

etriana85, 2024-02-14:
La ROM I1 de vendetta, está bien marcada en el sch. Será que ya la corregiste?. Pero me causa curiosidad que no coincida la OR que tiene al lado, ni el nombre de la señal A0.
[image]

etriana85, 2024-02-14:
Ésta es una fotografía de la PCB de los Simpsons, será que te refieres a ésta?
La RAM de los Simpsons tiene la ubicación 1J
[image]

Lo que si veo es que dejé por defecto de KiCad el orden del nombre de las diferentes ubicaciones,  *letra y número*. En éstas PCB el orden es, *número y letra*. Ya me dices si lo corrijo todo.

jotego, 2024-02-14:
¿cuál es el _driver_ de **DS**?

[image]

etriana85, 2024-02-15:
Página 13 del sch The Simpsons
[image]

closed 2024-02-15 by jotego