jotego/jtcores #550: Pocket Rastan Core Black Screen

bebopjp, 2024-02-14:
Freshly formatted an SD card.

Used Matt Panella's pupdate script to "Update All".

Rastan core loads ROM but is either a black screen or some static trash.

Not sure if it is a pupdate bug or a core bug.

All other JT cores work great.

jotego, 2024-02-16:
Out Run is a different case because it is the only core that use the _cell RAM_ in the Pocket. It is not related to this.

I have just tested today's version and it works fine. I do not use updaters. I think this is probably related to the updaters generating wrong .rom files. What they do about that is beyond me. You can generate the file manually yourself following [these instructions](https://github.com/jotego/jtbin/wiki/Analogue-Pocket-Cores).

[image]

closed 2024-02-16 by jotego

davewongillies, 2024-02-16:
@jotego The weird part is that the ROMs from the Pocket updater archive matches your md5 sums:

```
548:| rastan       | jtrastan   | 8b958b5f7dea436d104db5529fc516a0 | fe,ff        |
549:| rastana      | jtrastan   | 6148baf634a68bea12bff06df7027469 | fe,ff        |
550:| rastanb      | jtrastan   | 11730c745ba58632959253e15d6fbb5a | fe,ff        |
551:| rastanu      | jtrastan   | 13f181de12b379db538ba1fdf995658e | fe,ff        |
552:| rastanua     | jtrastan   | ccdfdb413aaf8cd7ddfedab7bb9ffcb3 | fe,ff        |
553:| rastanub     | jtrastan   | 8e91337be3271303c15d0ac7e5299c4a | fe,ff        |
554:| rastsaga     | jtrastan   | 7fdd1b89b170b25228033a3064f3d6e5 | fe,ff        |
555:| rastsagaa    | jtrastan   | ef4e52889856332024d6bda1789d7a23 | fe,ff        |
556:| rastsagaabl  | jtrastan   | b58fd892743e1bde971615553cf66349 | fe,ff        |
557:| rastsagab    | jtrastan   | f5650d6ddf071dc003936c4d8d71d004 | fe,ff        |
```

```
8b958b5f7dea436d104db5529fc516a0  Assets/jtrastan/common/rastan.rom
6148baf634a68bea12bff06df7027469  Assets/jtrastan/common/rastana.rom
11730c745ba58632959253e15d6fbb5a  Assets/jtrastan/common/rastanb.rom
13f181de12b379db538ba1fdf995658e  Assets/jtrastan/common/rastanu.rom
ccdfdb413aaf8cd7ddfedab7bb9ffcb3  Assets/jtrastan/common/rastanua.rom
8e91337be3271303c15d0ac7e5299c4a  Assets/jtrastan/common/rastanub.rom
7fdd1b89b170b25228033a3064f3d6e5  Assets/jtrastan/common/rastsaga.rom
ef4e52889856332024d6bda1789d7a23  Assets/jtrastan/common/rastsagaa.rom
b58fd892743e1bde971615553cf66349  Assets/jtrastan/common/rastsagaabl.rom
f5650d6ddf071dc003936c4d8d71d004  Assets/jtrastan/common/rastsagab.rom
```

davewongillies, 2024-02-16:
@countmattula @bebopjp actually I just did a full delete of the jtrastan core then installed it again (so *everything* relating to the core from `Assets`, `Cores`, `Settings`, `Presets` & `Saves`) and its now working.

jotego, 2024-02-17:
I'm glad it worked in the end!