jotego/jtcores #551: [MiSTer][NGP] It is not possible to go in game (stuck on the bios)

wwark, 2024-02-15:
Hi,

Since at least two updates of the NGP core, when we load a rom it goes to menu with calendar. I tried also with a reset but the behavior is the same.
It is not possible to load any game.

Thanks for your help.

jotego, 2024-02-16:
You need to _virtually_ power off the NG Pocket, load the cartridge and then power it on again. A reset will not work, the NGP is waiting for that _power on_ signal to occur.

I do not know if I can do anything about this from the hardware side. Maybe this needs a custom firmware to work in a more friendly way.

closed 2024-02-16 by jotego

wwark, 2024-02-16:
I will test thank for the details.
Yes it is not simple !

espiox, 2024-02-17:
This is odd, because a couple of versions ago it _would_ load the cartridge automatically after BIOS setup, without needing to power on/off, which was a much smoother experience.

jotego, 2024-02-17:
But there was a problem accessing the virtual flash memory in the cartridge, which made games crash when trying to save data. That is solved. It may have had this side effect.

espiox, 2024-02-17:
Ah, okay. Thanks 👍🏻