jotego/jtcores #552: Vendetta: Bug Report [vfbb89f0]

Toryalai1, 2024-02-16:
Platform: MiSTer
Core version: vfbb89f0
Video reference: https://www.youtube.com/watch?v=z9skP9BRvDU

General:
- During stage transition, when going from stage 1-1 to 1-2, there are some graphical garbage during the blackout.

Stage 2:
- Floating cranes are appearing randomly:
[image]

Stage 4
- Boat and anchor have sprite tearing
[image]

Also the platform after defeating the boss is following boat movement:
[image]

Stage 5
- Sign gets stuck on player 
[image]

- Two enemies should be out of view:
[image]

michelefornasini, 2024-02-18:
In the second part of the second level, when you enter the elevator that goes down, the 2 enemies that jump from above and try to hit you reappear below.

jotego, 2024-04-15:
The ROM check was implemented as part of #557. It is working in 8028053.

closed 2024-04-17 by jotego

jotego, 2024-04-17:
It looks like it all came down to subtracting the object vertical offset rather than adding it. I have corrected it. Could you check this MiSTer core to verify the fix?

[jtsimson.rbf.zip](https://github.com/jotego/jtcores/files/15006821/jtsimson.rbf.zip)

Please also have a look at _The Simpsons_ and _Parodius Da_ in case this fix caused trouble there.

commit 9314a3c8 (2024-04-17): simson: vertical object offset applied differently. Note that there is a 32-pixel difference for Simpsons and Vendetta. This must be set by the game somehow, but don't know how, so it is hard coded for now. Fixes #552