jotego/jtcores #555: Spinner Cores

SevenlivesDesigns, 2024-02-17:
I tried games like BlockBlock, Arkanoid and Breakout which originally use spinner controls. I think it would be amazing to add mouse support for those who have USB spinners. 

Sorry If I posted this in the wrong place. I did not see other posts with this request. Thank for all the hard work to Jotego and team.

jotego, 2024-04-02:
JT cores do support mouse when spinner is supported, but _Block Block_ support for spinner is not ready. Try _Arkanoid 2_, which should work.

SevenlivesDesigns, 2024-04-06:
Hello Jotego, thanks.  Are we speaking about the Analogue Pocket/Dock? 

I plug in my spinner/mouse and the dock recognizes it  as mouse (player 2).  But Arkanoid Revenge of Doh, doesn't move with the spinner.   Also If it needs to be P1, the mouse doesn't have coin feature to start the game, so I can't get past the title. 

Any help? Perhaps I am doing something wrong.  Thank you in advance for all the hard work as well.

jotego, 2024-09-19:
Why does it recognize as player 2? Do you have something else connected?
You can always insert a coin and start a game using the Pocket buttons, even if docked, right?

SevenlivesDesigns, 2024-09-19:
Hi Jotego, thanks for response. I have an 8bitdo wireless dongle connected to be able to maneuver menus etc. I also have the spinner (mouse) connected. My Spinner is Grs Buttonhole Spinner V2 to be exact, but always behaves just like a mouse.

If I go to the control settings on the Pocket Dock, and rearrange the mouse to be read 1st, so that the game Arkanoid recognizes it as the main input, I can no longer use the 8bitdo 
controller to navigate to the game. So I am stuck at menu without being able to get to the game. If I leave the mouse to be read second, Arkanoid doesn't really respond with the mouse. 

Hope this helps.

I can test whatever if need be. Thank you very much.

jotego, 2024-09-19:
Thank you for the detailed response. How about the buttons on the Pocket itself? Do they stop working if the Pocket is docked? Otherwise you can start the game using them...
A related question: does the spinner have buttons, how many?

SevenlivesDesigns, 2024-09-19:
No problem, it's a very good suggestion, unfortunately the buttons on the AP do not function once docked. 

The dock assigns the mouse to P4. Would it be possible to make core accept mouse support from any of the 4 player slots?

As it is now, In Revenge of Arkanoid, if I move the mouse/ spinner, it jitters back/forth very fast and it isn't controllable. It seems to understand there is some input happening but nothing discernable as proper left/ right.

Hope it helps and thanks for all the hard work with all the cores!!

jotego, 2024-09-20:
I know it sounds like an easy fix but it implies breaking the consistency of the framework. I need to think and test this on our setup to see what can be done that works the user but it is not a kludge.

SevenlivesDesigns, 2024-09-20:
I absolutely understand. Thanks Jotego for looking into this :)