jotego/jtcores #556: cps: cps1 frequency response

jotego, 2024-02-18:
Artemio kindly ported his test suite to CPS1 and made some measurements using real boards. I have some questions about the measurements:

- The name in the files CPS1-number-number-...flac makes reference to the PCB product ID, right? Basically two boards: 89626A-4 and 88617A-7b. The schematics available correspond to model 88617-8 (Forgotten Worlds)
- Each file has a sequence of notes and then a frequency sweep. Which part of the file is FM and which part is PCM?
- Which frequency range did you use for each sound source (FM,PCM)?

Pin 7 of the MSM6295 (PCM chip) selects the sampling frequency. Games do toggle this bit. This will have an effect on the aliasing as the Nyquist filter after the PCM chip is fixed. Thus for high speed setting, the filter will be poorer and there will be more aliasing. I wonder which setting you used for the test.

ArtemioUrbina, 2024-02-18:
My pleasure. Yes, MDFourier has been ported to CPS1 using ccps, the current source code is here: 

https://github.com/ArtemioUrbina/cps1mdfourier

- Yes, the recordings follow that nomenclature, since my personal collection only has only those three types of boards, what's interesting is that they were very consistent between themselves using the same board revisions, as it has been the case with other platforms. We'd need to check how different the older 88617A-7b is from 88617-8, if at all in regards to audio. I can provide photos or check continuity if needed. The 89626A-4 is newer, and has a revised and definitively intentional different filter for ADPCM specifically, this inferred due to testing several boards of that revision that all match between them.
- The first part is FM, the second is ADPCM. They are different colors, with the YM2151 being green and ADPCM being orange:
[image]
- For FM the first note is at 68hz and the highest and last note is at 17689hz, by playing each of the 16 notes from the lowest octave (0) to the highest octave (7). For ADPCM a sweep was used from 20hz to 3780hz that lasts for 5 seconds in the high speed setting (with Pin 7 of the MSM6295 chip being set to 1)

Regarding Pin 7 of the MSM629, I will re-record the whole thing from all my PCBs again to compare having it set to both values, in order to confirm how much it changes and give you a proper comparison. I usually go with the high setting since I assume (maybe incorrectly) that since that covers the highest range and the analog path is the same, the widest range is being compared. I already coded this variant of the test and will be able to record tomorrow if you deem this to be useful.

jotego, 2024-02-18:
There is no explicit low-pass filter for the FM sound on the board:

[image]

But I saw recently schematics for a Yamaha DAC, not sure the model, and I think the output was not buffered. So a resistor chain would be in series with the decoupling capacitors, forming a natural low pass filter. I think it was ikamusume, in Twitter, who published the schematics so I am going to ask him.

That would make this messy because the effective filter frequency could be signal-dependent, and hencen non-linear. I need to see how that resistor divider at the output of YM3012 was done. Some configurations can have a constant output impedance.

jotego, 2024-02-18:
Another artifact that complicates things is that due to the short length of the quarter sine ROM inside the YM2151, for some high frequency settings you get less amplitude at the output. I can see this in simulation where the digital values have no filtering:

[image]

This means that you are measuring two things: the FM ROM table _aliasing_ and the DAC output impedance. It would be nice to separate them. Maybe using a different waveform as the output of the YM2151 could provide a more consistent amplitude level across frequency. A high feedback value for the 1st operator can produce a triangular shape if I remember well. I would need to think about how to overcome this limitation.

jotego, 2024-02-19:
There are schematics available for [YM3014](https://github.com/ika-musume/YMDAC/blob/main/resources_YM3014/YM3014.pdf). From them, the output impedance for S[2:0] codes from 7 to 1 is 0.5R, 0.375R, 0.344R, 0.336R and then 0.333R. It looks like the schematics are missing a resistor, by adding it, the output impedance becomes exactly R/3 for all codes.
I will assume the output impedance is constant on YM3012 too.

ArtemioUrbina, 2024-02-19:
> This means that you are measuring two things: the FM ROM table _aliasing_ and the DAC output impedance. It would be nice to separate them. Maybe using a different waveform as the output of the [...]

I'll be waiting for suggestions on that area. Right now I can only compare the whole thing after the DAC since I am recording from the PCBs, but will follow any instructions to help.

> Pin 7 of the MSM6295 (PCM chip) selects the sampling frequency. Games do toggle this bit. This will have an effect on the aliasing as the Nyquist filter after the PCM chip is fixed. Thus for high [...]

Regarding the Above, I coded and tested a more complete version that runs the sweep in both pin7 settings:

[image]

Yellow is oin 7 in high, and orange is pin 7 in low. Filtering is indeed different in MisterFPGA with both, I'll need to record the other PCBs to check if filtering varies between PCBs between both settings. But it follows the same filtering in hardware and MAME in both settings when comparing a PCB and MAME as you can see below, since both difference lines trace above each other:

[image]

I'll add more results tomorrow after recording from more hardware.

I added the option to test the "Reduction Specification" (amplitude) in ADPCM, and made MDfourier compare that in the time domain only. Here it is with zero reduction, above is the PCB and below is MiSTerFPGA:

[image]

[image]

Here it is at a different Reduction level:

[image]

[image]

I made these since I thought that maybe reduction levels were wrong and that made a perceived difference, but no, they match perfectly in amplitude, but you can clearly see the filtering difference even in the time domain:

jotego, 2024-02-19:
Just pasting results before going through your latest reply.

The bounce back effect seeing in the reference waveform after the filter effect, cannot be seen with and without filter in the core, and using both truncated and non-truncated outputs of JT51. Could the bounce back be aliasing from your signal chain, @ArtemioUrbina ?

**Reference**
[image]

**JT51 with truncated and non-truncated outputs, no filter**
[image]

jotego, 2024-02-19:
Thank you for the new measurements. I think I need to split this in two different issues, one for FM (just continue here) and one for ADPCM (I'll create a new one) as differences will have complete different causes in both of them.
As far as I know, MAME does zero filtering on this driver. I think they do not truncate the FM signal either. This truncation comes from the YM3012, which drops a few lower bits of the signal when the amplitude is large. I can optionally drop them in the core, but I have never really noticed any difference. I leave them as they likely help prevent aliasing due to digital resampling in the FPGA signal path to HDMI.

ArtemioUrbina, 2024-02-19:
I used toslink for MiSTer and the chain has been validated with stricter tests. I attach the new pattern generator in case you want to verify it, it is a new ADPCM tone at 1khz that is played with both Hi and Lo, and the 8 reduction levels to compare.

[sf2ud.zip](https://github.com/jotego/jtcores/files/14328921/sf2ud.zip)

And the flac files that generated these plots for both the PCB and MiSTer:

[NewFullADPCM.zip](https://github.com/jotego/jtcores/files/14328974/NewFullADPCM.zip)

ArtemioUrbina, 2024-02-19:
[image]

[image]

ArtemioUrbina, 2024-02-19:
Just for reference, here is the difference comparison using the toslink capture chain:

[image]

jotego, 2024-02-19:
By the way, @ArtemioUrbina, if you zoom in, you will see that the positive side of the sine wave is distorted in the PCB recording:

[image]

This effect does not show in the core. It may be related to some electrical problem in the board. I do not think it indicates true performance.

ArtemioUrbina, 2024-02-19:
> Thank you. I can see the bouncing indeed. Apparently I had been recording a filtered version. The effect comes from sine wave ROM aliasing. This is a new record from the core. Excuse me for not looking at the spectra just yet. I want to get the easy parts aligned first.
> 
> ![imagen](https://private-user-images.githubusercontent.com/1863036/306040553-836059f1-0412-44cb-82f5-74d091d758da.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgzNjU0MTksIm5iZiI6MTcwODM2NTExOSwicGF0aCI6Ii8xODYzMDM2LzMwNjA0MDU1My04MzYwNTlmMS0wNDEyLTQ0Y2ItODJmNS03NGQwOTFkNzU4ZGEucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIxOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMTlUMTc1MTU5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MGFjMTVhYzM0OGE5OTc3MDgzMGYyZGIyNDRiMTJkZWY2ZDEwM2JiZjZkYTllYTQ3Y2MxYzY3YjM5NWNiMzJkNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.1iabcZVcXIFYGmAgegHLym6Rp1GEwtjLJQkZGfTf-NY)

Ii was afraid I was doing the same and checked just after you mentioned it.

Just recorded the older CPS1 88617A-7b (GNG) with ADPCM and compared it to the newer CPS18 9626A-4 (Sf2) with ADPCM, and the filters in hardware for ADPCM follow the exact same curve in hi and low for pin 7.

[image]

As you can see yellow and orange trace the same difference curve. 

Comparing CPS1 88617A-7b to Mister we can see the bouncing on the frequency domain, and also the difference in filtering for ADPCM:

[image]

ArtemioUrbina, 2024-02-19:
> By the way, @ArtemioUrbina, if you zoom in, you will see that the positive side of the sine wave is distorted in the PCB recording:
> 
> ![imagen](https://private-user-images.githubusercontent.com/1863036/306041297-7f58e96d-a3ad-44da-97f2-322288f23474.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgzNjU2MDYsIm5iZiI6MTcwODM2NTMwNiwicGF0aCI6Ii8xODYzMDM2LzMwNjA0MTI5Ny03ZjU4ZTk2ZC1hM2FkLTQ0ZGEtOTdmMi0zMjIyODhmMjM0NzQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIxOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMTlUMTc1NTA2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NzViZTFjMzdkMmFiMjhmMDQ3YWRmMTY0ZTFkMzg4NjlkOWM3YzhmNGUwZWY5MGJjYjI1ODYxM2U5OWI4MGQ3NyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.T_C_aEPJELQPE2qqkXEOD9W_Ff_Vmoy-MYVJ1HMMMb8)
> 
> This effect does not show in the core. It may be related to some electrical problem in the board. I do not think it indicates true performance.

Yeah, that's true. But it matches across all my boards and across all recordings as you can see in the FM comparisons and in all the recordings you have. 

[image]

And there is no missing or extra frequency information when you compare them. The match in FM is basically perfect across PCBs from different games and revisions:

[image]
[image]

ArtemioUrbina, 2024-02-19:
Here is the CPS1-88617A-7b-GNG-ADPCMFull recording:
[CPS1-88617A-7b-GNG-ADPCMFull.zip](https://github.com/jotego/jtcores/files/14335608/CPS1-88617A-7b-GNG-ADPCMFull.zip)

jotego, 2024-02-19:
> Comparing CPS1 88617A-7b to Mister we can see the bouncing on the frequency domain, and also the difference in filtering for ADPCM:
> 
> ![DA__ALL_AVG_CPS1-88617A-7b-GNG-ADPCM_vs_MisterFPGA-CPS1-ADPCM-Fu_0000](https://private-user-images.githubusercontent.com/6305986/306041512-8d3b49fe-3ec4-4ce4-b327-f7735e827b69.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgzNjU2NjcsIm5iZiI6MTcwODM2NTM2NywicGF0aCI6Ii82MzA1OTg2LzMwNjA0MTUxMi04ZDNiNDlmZS0zZWM0LTRjZTQtYjMyNy1mNzczNWU4MjdiNjkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIxOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMTlUMTc1NjA3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTcyMjA0MGQ1ZGZiNTFlMTgxZmNhN2RlMTE0NmViOGIzNmVjNDI0NjIyZjViZGIwNTk2NDkxMmUwZDIwZmQyNSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.ScJ55D68q00KP24lG5qA5_nW_5DnvkDmvdjrd1btc_U)

If I understand this correctly, this plot is the spectrum difference between the MiSTer core and the board. If the dB are negative, who is small? the core or the board? Should I read it as FM is the same in the 100-1000Hz band? But then PCM is about 3-6dB louder in the core around 200Hz? Then the core has a low-pass filter so the board eventually gets 6dB louder PCM than the core at 3kHz? 

The board being 18dB+ louder than the core at 2.3kHz sounds unreasonable. The difference is too large and at a very hearable frequency. It is also a frequency at which the YM2151 will produce nice sine waves. It does not make senseunless the wave distortion is creating those HF harmonics.

Speaking of the distortion, supply noise is a likely culprit. Yes, capacitors will likely help. But we can just forget about the _broken_ boards and use the _Strider_ one as reference, if that one produces a clean sine wave.

I have recompiled the core without any filter on the ADPCM side. [jtcps1.rbf.zip](https://github.com/jotego/jtcores/files/14335792/jtcps1.rbf.zip). Could you produce the spectrum difference again of this core vs Strider?

By the way, if you could produce a longer ADPCM burst, it would be easier to compare the waveforms visually. The spectrum is very useful but plain transient waveforms are also kind to the eye. The current frequency sweep is too fast for that and the results look really different.

Thank for all your help

ArtemioUrbina, 2024-02-19:
> If I understand this correctly, this plot is the spectrum difference between the MiSTer core and the board. If the dB are negative, who is small? the core or the board? 

Yes it is the spectrum difference between the Board (as reference) and the MiSTer (as comparision). MdFourier uses the reference as the "truth" in a relative per comparison exercise. Thus, the plot is of spectral differences of the comparision signal (MiSTer in this case), in other words if the plot is negative it shows that MiSTer has that difference againt the PCB.

> Should I read it as FM is the same in the 100-1000Hz band? But then PCM is about 3-6dB louder in the core around 200Hz? Then the core has a low-pass filter so the board eventually gets 6dB louder PCM than the core at 3kHz?

So yes, all the above are correct.

> The board being 18dB+ louder than the core at 2.3kHz sounds unreasonable. The difference is too large and at a very hearable frequency. It is also a frequency at which the YM2151 will produce nice sine waves. It does not make sense unless the wave distortion is creating those HF harmonics.

I agree, but that's what the numbers say. All the process is open so we can audit any part of it if you like.

> Speaking of the distortion, supply noise is a likely culprit. 

Just as I read you I was about to tell you I had found the cause, and yes it is the power supply. I was using a new one, ofr those recordings, I went back to an era appropriate arcade power supply and the issue is gone.

> Yes, capacitors will likely help. But we can just forget about the _broken_ boards and use the _Strider_ one as reference, if that one produces a clean sine wave.

With the power supply out of the equation, we now have clean form all of them, but I went with the Magic Sword one since it is a newer revision (88617A-b7) closer to the schematics we have. 

> I have recompiled the core without any filter on the ADPCM side. [jtcps1.rbf.zip](https://github.com/jotego/jtcores/files/14335792/jtcps1.rbf.zip). Could you produce the spectrum difference again of this core vs Strider?

I will and will post results soon here, with the current build of MDfourier.
> 
> By the way, if you could produce a longer ADPCM burst, it would be easier to compare the waveforms visually. The spectrum is very useful but plain transient waveforms are also kind to the eye. The current frequency sweep is too fast for that and the results look really different.

I can and will to the largest possible size on the OKI, thank you. MDfourier can automatically produce the time domain plots of each "note" as the above comparing aligned segments (with the -u option), let me know if that helps you.

> Thank for all your help

Thank you for caring.

ArtemioUrbina, 2024-02-19:
> I have recompiled the core without any filter on the ADPCM side. [jtcps1.rbf.zip](https://github.com/jotego/jtcores/files/14335792/jtcps1.rbf.zip). Could you produce the spectrum difference again of this core vs Strider?

Here are the results comparing the unfiltered ADPCM:

[image]

[image]

[image]

> The board being 18dB+ louder than the core at 2.3kHz sounds unreasonable. The difference is too large and at a very hearable frequency. It is also a frequency at which the YM2151 will produce nice sine waves. It does not make sense unless the wave distortion is creating those HF harmonics.

That is due to the results in the PCB, check these out:

PCB:
[image]

MiSTer:

[image]

Since the PCB is lower by that much at that frequency, MiSTer is that much higher when compared relative to it.

ArtemioUrbina, 2024-02-19:
Here is the Unfiltered ADPCM MiSTer FPGA recording:
[MisterFPGA-CPS1-MDF-unfilteredADPCM.zip](https://github.com/jotego/jtcores/files/14336113/MisterFPGA-CPS1-MDF-unfilteredADPCM.zip)

ArtemioUrbina, 2024-02-20:
> By the way, if you could produce a longer ADPCM burst, it would be easier to compare the waveforms visually. The spectrum is very useful but plain transient waveforms are also kind to the eye. The current frequency sweep is too fast for that and the results look really different.

remnade a version that plays a 40 second ADPCM sweep, here are the recordings;

Magic Sword CPS1 88617A-7b
[CPS1-88617A-7b-MagicSword-ADPCM-40s.zip](https://github.com/jotego/jtcores/files/14338629/CPS1-88617A-7b-MagicSword-ADPCM-40s.zip)

MiSTerFPGA non filtered code via toslink:

[MisterFPGA-CPS1-MDF-unfilteredADPCM-40s.zip](https://github.com/jotego/jtcores/files/14338631/MisterFPGA-CPS1-MDF-unfilteredADPCM-40s.zip)

And here's MDFourier difference plot:

[image]

Spectrograms:

PCB:
[image]

MiSTer:

[image]

And the sweeps:

PCB:
[image]

MiSTer:

[image]

jotego, 2024-02-20:
Thank you very much for all the new measurements and plots. Let me comment on the FM side. The summary is:

- The comb shape is present in the core as well. It is visible in the time domain and frequency domains
- The PCB recording has more filtering in the 0-10Hz band than my signal chain (the core does not filter that band)
- The PCB recording has less filtering above 18kHz than my signal chain
- The Magic Sword recording still shows distortion for large positive amplitudes

My takeaway is that the FM channel in MiSTer is good as it is. Differences  below 10Hz and 18kHz are hard to control because each user has a different signal chain and filters around those frequencies will always be present. In the mid frequencies, I do not see differences large enough to justify going after them, at least for now because the PCM channel does have a large mismatch.
 
About the distortion seen in the PCB recordings, one way around it would be to use a lower amplitude setting in the YM2151. We do not know what an out of the box PCB would be like but I would assume that the distortion was not present in fresh boards. Debugging the problem in the PCB can be quite a quest of its own as the number of possible culprits is quite large (power amplifier, opamps, DAC, digital noise coupling, capacitors, power supply...)
 
This spread sheet contains the frequency comparison data I have used for the above judgement.
[comp.ods.zip](https://github.com/jotego/jtcores/files/14340953/comp.ods.zip)

jotego, 2024-02-20:
> I must add that that filtering at 2.3khz is inherent to CPS1, the X68000 doesn't show it with its YM2151:
> 
> ![SP_A_CZ-600C-Artemio-Headphon_01FM](https://private-user-images.githubusercontent.com/6305986/306054151-86c90fa2-5109-4238-9ff6-9dfccadc75ad.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg0MTU2MTEsIm5iZiI6MTcwODQxNTMxMSwicGF0aCI6Ii82MzA1OTg2LzMwNjA1NDE1MS04NmM5MGZhMi01MTA5LTQyMzgtOWZmNi05ZGZjY2FkYzc1YWQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIyMCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMjBUMDc0ODMxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjUxMmI0MjA5NDU5ZjhiZTEwOTE1Y2NlZTFiMjEyN2FhNDdjZGU4MzI5MDhkODBkMzc2YjkxMjliNThlNDNjYyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.KiB-nKyHKFxPOchglB2rHBxvneMCluZCAqj9bSDp59M)

Let me insist on this one: I think this is due to aliasing in the quarter-sine ROM table. It should always be present. For high frequency settings only a handful of values are read from the table, and this may skip the largest amplitude ones, so you get the effect of having a different gain. 

For this to be repetitive, the phase counter must be reset at the beginning of each note. The key on command should reset it. It will also be sensitive to anything affecting the phase, like DT and LFO settings.

If you want to test this, you can set a very slow note, like a 10Hz tone and then just set the frequency setting above 2kHz at different positions of the output waveform. That will make the operator read the ROM table starting from a different position. Depending on when you trigger the frequency change relative to the current output, you will get a different amplitude for the HF sine wave. But, the important take away is that this does not represent real filtering due to external devices outside the YM2151 chip.

jotego, 2024-02-20:
@ArtemioUrbina it looks like the latest mdf zip does not run the long ADPCM sweep but the short 11 second version. I do see the volume steps (nice addition!) Could you post again the .zip with the modified files for mdf.zip? There's no need to include other files that do not change.

ArtemioUrbina, 2024-02-20:
> @ArtemioUrbina it looks like the latest mdf zip does not run the long ADPCM sweep but the short 11 second version. I do see the volume steps (nice addition!) Could you post again the .zip with the modified files for mdf.zip? There's no need to include other files that do not change.

Sure thing, sorry about that. 

[mdf_40sADPCM.zip](https://github.com/jotego/jtcores/files/14349657/mdf_40sADPCM.zip)

jotego, 2024-02-20:
First spectrum comparison for ADPCM:

- the ac coupling in the PCB has a high pass effect visible in the spectrum, with 10Hz signals more attenuated than in the core. Not critical as these are not really audible
- the low-pass filter on the board has a clear effect and it is needed for sound accuracy (compare red curve with blue one)
- the current low-pass filter in the MiSTer core is too steep and it comes too late (see yellow curve)

I will work on a new low-pass filter to match the PCB spectrum better. I will look at the relative FM/ADPCM balance after that.

[image]

jotego, 2024-02-21:
ADPCM issue moved to #559 for clarity. I will continue here after solving that one.

jotego, 2024-02-21:
If you are using key on and key off to control the notes during the FM test, the phase counter should get reset and you should get the same waveforms consistently. So the aliasing caused by the large sampling of the quarter sine ROM should be repetitive.
But, if other register values such as the LFO, AM, PM or DT1/2 happen to be set, you will get a different sampling effect.
The X68000 could have other values preloaded? What does MAME produce for X68000?

jotego, 2024-02-22:
@ArtemioUrbina, could you please give me a file that only generates four tones of the same amplitude on the ADPCM chip and that plays it in an infinite loop? That way I can fine tune the filter in real time. The tones would be at:

- 20 Hz
- 200 Hz
- 1000 Hz
- 2000 Hz

If possible, could the Z80 just work on its own without waiting for the M68k to trigger it? That will allow for Z80-only simulations, which are faster. It will also be more portable to other arcade systems.

Excuse me for asking for so many things, but, if you do not mind, reply in #559 so the ADPCM conversation is there.

jotego, 2024-03-03:
Fixed in 22b08c09

closed 2024-03-03 by jotego