jotego/jtcores #557: Bad Rom Checks - Konami

Toryalai1, 2024-02-18:
Hi,
I've gone through all Konami games and the following list are games that reports bad roms checks in service menu:

- Aliens
- Crazy cop
- Crime fighters
- Super contra
- Parodius Da
- The Simpsons 
- Vendetta
- Punk shot 
- TMNT

jotego, 2024-04-02:
Technical description:

The GFX chips in these games have a mechanism to let the ROM data pass through to the CPU. The CPU writes the desired ROM address in some internal registers and the chip serves the data. A similar mechanism is implemented in the K053260 for the PCM ROM.

Although this is implemented and it seems fine in simulation, something seems to go wrong along the way and the check sums are not correct.

gyurco, 2024-04-02:
It looks like, there are missing implementation pieces here and there, like in jtaliens_obj, rom_addr is adjusted when romrd=1, but rom_cs is not activated, neither rom_ok is propagated to dtack.

gyurco, 2024-04-02:
Activating rom_cs and anding dtack with rom_ok makes it read something, but still fails (at least not with a 00000 checksum). Probably the addressing also needs to be revisited.
Question about JTKCPU: is it possible to distinguish between a read and write cycle? As every write begins with a read (same address, but we is down), it's impossible to not activate rom_cs in a write cycle.

jotego, 2024-04-02:
The CPU design uses mostly microcode. Not as much as the JT680X series, but enough to make changes quite easy to do. 
The tool to compile the microcode is in the same repository and trivial to build. 

The microcode is basically words that have a 1:1 translation to signals being active while the word is present in the code. There is a little bit more to it than that, but not much. 

How would the distinction help this issue? Or is it more of a general thing?

gyurco, 2024-04-02:
Just generally it would be advisable to not activate rom_cs in a write cycle. Nothing to do with the issue anyway, hopefully it doesn't insert unnecessary delays with stressing the sdram more.

gyurco, 2024-04-02:
I don't know if the JTKCPU is designed to be bus compatible with the original one, if the real CPU works this way, then I must accept it.

gyurco, 2024-04-03:
Aliens is done, but there are 2(?) more cores to fix, it'll take some more time:
- simson
- tmnt

gyurco, 2024-04-03:
Hmm, even in Aliens, Gang Busters and Super Contra still fail.

jotego, 2024-04-03:
> Hmm, even in Aliens, Gang Busters and Super Contra still fail.

It can be tricky as the check takes so long! But it's a very good sign that you got one working. I wonder if we need to hold the CPU if the rom_ok signal is missing. Maybe not.

gyurco, 2024-04-03:
DTACKs are needed, and I added them:
https://github.com/jotego/jtcores/commit/6f2091eebde01e27e6407f1ce4b4b51292b457bc#diff-f01b7a0bfeabe9032e923c98d8933e04418ba892d7aec51c0506790a544f1f67R95

Unfortunately these two games doesn't write out the wrong checksum, only 'BAD'. In Aliens, it gave some hints what went wrong (the checksum is adding up all the bytes). However at least in Gang Busters, the tilemap ROMs are passing, only the sprite ROMs are 'BAD'. I suspect something with ROM addressing (something is shuffled around).

gyurco, 2024-04-03:
It was byte shuffling.
For Super Contra and Gang Busters, the order here:
https://github.com/jotego/jtcores/blob/ad6168392495c0c54ecd058e15367d8aeccad73c/cores/aliens/hdl/jtaliens_obj.v#L94
must be 3,2,1,0.
For the others, the correct one is 2,3,0,1.
Strange that the byte order is different, but using the 32-bit word in the rendering is the same?

gyurco, 2024-04-06:
In Simpsons, where the CS logic comes from?
In the core:
    objsys_cs  = A[15:13]==1 && WOC1 && init;

In MAME:
    map(0x1fc8, 0x1fc9).r(m_k053246, FUNC(k053247_device::k053246_r));

Now the problem is that reading 0x1fc8 doesn't activate objsys_cs. WOC1 is low.

jotego, 2024-04-07:
> In Simpsons, where the CS logic comes from? In the core: objsys_cs = A[15:13]==1 && WOC1 && init;
> 
> In MAME: map(0x1fc8, 0x1fc9).r(m_k053246, FUNC(k053247_device::k053246_r));
> 
> Now the problem is that reading 0x1fc8 doesn't activate objsys_cs. WOC1 is low.

It should be coming from the schematics and PAL dumps. It looks like I did not commit the PAL equations for _Simpsons_ in the pal folder this time. Compatible PALs can be found [here](https://wiki.pldarchive.co.uk/index.php?title=The_Simpsons). The PAL devices are shown in the **CPU** page of the schematics. Although they are marked with Konami part numbers. MAME's `jedutil -view` will dump the equations.

I don't know if you've looked at this type of equations before. It can be confusing until you get used to it. I've seen things that don't make quite sense in some PAL dumps in the past, so a grain of salt may be needed. Nonetheless, PAL dumps may unveil quirks in the decoding logic. Particularly, these Konami games have a boot-up procedure that is not well captured in MAME and that relies in a bit, typically called _init_ in the schematics.

gyurco, 2024-04-08:
What's objsys_cs in the core is something else...but must be used (palette or tilemap RAM?), otherwise other checks fail.
The real objsys_cs is CRCS, which marked with '?'. At least that question is answered now.
And there's a twist: Simsons check the audio sample ROMs, too.

jotego, 2024-04-09:
There are two _chip select_ signals for the object chips. _objsys_cs_ selects the RAM and _objreg_cs_ the internal registers, which were spread across the two object chips.

The PCM ROM reads are also implemented, but again, something is falling through the cracks.

gyurco, 2024-04-09:
I wonder where CRCS is connected then, as it's surely the select for the ROM read back.

jotego, 2024-04-09:
I know what you mean know. There was a _ROMRD_ signal in the other related cores. Let me check the schematics.

jotego, 2024-04-09:
_The Simpsons_ uses a different chipset for objects. Let's limit this issue to the chipset used in Aliens/TMNT, which is better known. We can tackle the other chipset separately after that.

gyurco, 2024-04-09:
Finally it works without the knowledge: (objsys_cs | crcs) ? objsys_dout  in the cpu_din mux does the job well.

gyurco, 2024-04-09:
The problem with sound ROM check is this:
https://github.com/jotego/jtcores/blob/d036b7738925e13775ca13dbd50db363773b9850/cores/simson/hdl/jt053260.v#L439
Why skip the first byte? At least the ROM check fails because of that.

jotego, 2024-04-09:
Apparently the first byte is not ADPCM data, so starting the calculation from it will break the sound. But that line is gated by _tst_en_, isn't that the signal for reading the ROMs?

gyurco, 2024-04-09:
tst_en is used when the ROM data is read, and the address has to be incremented. For the initial setup, it must start from 0.

gyurco, 2024-04-09:
Ah, ok, tst_en is constantly on in ROM read mode. But it still has to be set to 0 before the test starts, I think it's done before test mode is set.

gyurco, 2024-04-09:
BTW, I don't hear sound glitches when the +1 is removed.
There are however object glitches, I wonder if you noticed them? E.g. when you start a game, the diamond threw up, a yellow 'dirt' follows it on the right side.

gyurco, 2024-04-09:
> Ah, ok, tst_en is constantly on in ROM read mode. But it still has to be set to 0 before the test starts, I think it's done before test mode is set.

Disregard this, see below.

> Apparently the first byte is not ADPCM data, so starting the calculation from it will break the sound. But that line is gated by _tst_en_, isn't that the signal for reading the ROMs?

The condition is (!tst_en || we). The cpu writes to FC0E which should reset the address to 0. But it resets to 1, so the test will be off-by-one.

jotego, 2024-04-09:
So we need to reset the counter upon entering the test mode, I suppose

gyurco, 2024-04-09:
Write to FC0E happens _after_ entered test mode. Simply the write should reset to 0, not 1. Maybe discard the +1 if test mode is active. Seems a bit hacky for me.

gyurco, 2024-04-09:
'''
                    rom_addr <= start+!tst_en; // skip first byte for ADPCM
'''
works actually, even if not very elegant.

jotego, 2024-04-09:
> ''' rom_addr <= start+!tst_en; // skip first byte for ADPCM ''' works actually, even if not very elegant.

I see. I think there are schematics for this. Is that +1 in the schematics?
BTW, for verilator to accept this the sum will have to be `start+{20'd0,~test_en}`

gyurco, 2024-04-09:
+1 is in your code currently :)
On the schematics, there's a separate start and position counter registers, and the counter increases, not rom_address. rom_address is always start+counter.
The counter is a result of some level of multiplexers, I don't know if it starts from 0 or 1. Or maybe in one case, it starts from 0, in other case it starts from 1?

gyurco, 2024-04-09:
The multiplexer is muxing ch0...ch3 counters. Counters have a reset input, which resets it to 0 on some conditions, but nothing resets it to 1. Is it really needed to skip the first byte? There must be a reason why it's there. They put a dummy byte before every sample just for fun?

jotego, 2024-04-09:
Let's remove the +1. I must haven taken it from MAME's ADPCM algorithm.

gyurco, 2024-04-09:
Ok. I wonder what happens if the sample is not ADPCM? Then it just discards that sample? I'm sure nobody will hear that, but probably it was not the intention of the chip makers.

gyurco, 2024-04-09:
Now only Parodius has an issue with obj ROM test: objcha_n is always 1, and no CRCS?

gyurco, 2024-04-09:
So Parodius has another chip (k053244) with different rom read semantics...

jotego, 2024-04-09:
> Ok. I wonder what happens if the sample is not ADPCM? Then it just discards that sample? I'm sure nobody will hear that, but probably it was not the intention of the chip makers.

If the first byte is unrelated data, it may ruin the whole sound from beginning to end as ADPCM has very poor tolerance to bad data

jotego, 2024-04-09:
> Now only Parodius has an issue with obj ROM test: objcha_n is always 1, and no CRCS?

We have the schematics in JTBIN (paroda.pdf). I could not identify the signal for it in the schematics today

gyurco, 2024-04-09:
Updated the PR. Parodius still don't work (it uses a different chip, objchan_n and cr_cs like functions are inside the chip, and does the test ROM address setup differently - all from MMRs, doesn't use the CPU AB).
TMNT is ok, however Punk Shot reports the obj ROMs swapped, which is strange, as they're basically the same machine.

jotego, 2024-04-09:
Thank you.
Konami used to swap wires each time they connected two chips together. Does the report actually say "swapped", that would be very clever!
I guess we can reverse the connection in the video module based on the selected game. If that's all what's left, I can do that bit off this issue.

gyurco, 2024-04-09:
It doesn't report 'swapped', just the checksums are exchanged. However if I change the order, then the check in TMNT breaks. If we swap the ROMs in the MRA, then probably the rendering will be bad. Strange issue.
Adjusting the order from 3-2-1-0 to 1-0-3-2 just for Punk Shot works, but why?

jotego, 2024-04-09:
The Thunder Cross check also fails. I was expecting that one to pass as the hardware is pretty the same as Gang Busters. Maybe there's still a weakness in the implementation. Could you check it too, please? Compare the run with MAME to see the difference.

**JTALIENS**
- [X] Aliens
- [X] Crime Fighters
- [X] Gang Busters
- [X] Super Contra
- [ ] Thunder Cross. **Gets stuck**

**JTTMNT**
- [X] TMNT
- [X] MIA
- [X] Punk Shot (fixed in 0a09669)

**JTSIMSON**
- [X] The Simpsons
- [X] Vendetta
- [ ] Parodius da! **K8/K2 obj ROM fails**

_Parodius da!_ can be dealt with on a later core, when we get more exposure to that chipset.

gyurco, 2024-04-09:
Damn, I would bet it was working at some stage.

gyurco, 2024-04-09:
Hmm, the problem is that as soon the ROM test starts, a FIRQ comes, and breaks it. Disabling the FIRQ makes the test passes, but I doubt it's the correct way to solve.

gyurco, 2024-04-09:
The code in question
'''
LDA $1f98
ORA #$01 ; turn on romrd
STA $1f98
'''
As no readback is implemented for $1f98, and not even 'we' considered, the first LDA basically "writes" $FF which turns on everything, including starting PMC...

gyurco, 2024-04-09:
This makes it work:
```
diff --git a/cores/aliens/hdl/jtaliens_main.v b/cores/aliens/hdl/jtaliens_main.v
index 0c608480..9e2f59bf 100644
--- a/cores/aliens/hdl/jtaliens_main.v
+++ b/cores/aliens/hdl/jtaliens_main.v
@@ -282,7 +282,7 @@ always @(posedge clk, posedge rst) begin
                         endcase
                     end
                     5: port_in <= A[0] ? dipsw[15:8] : dipsw[7:0];
-                    6: { pmc_start, pmc_bk, rmrd } <= cpu_dout[2:0];
+                    6: if (cpu_we) { pmc_start, pmc_bk, rmrd } <= cpu_dout[2:0]; else port_in <= {5'd0, pmc_start, pmc_bk, rmrd};
                     default:;
                 endcase
             end
```

In Aliens, 'we' separation is more clear, in the others, 'we' is discarded. I don't know which one is more correct, but in this case, 'we' definitely needed.

gyurco, 2024-04-09:
Oh, I see you found out the case with Punk Shot!

jotego, 2024-04-10:
> This makes it work:
>
> ```
> diff --git a/cores/aliens/hdl/jtaliens_main.v b/cores/aliens/hdl/jtaliens_main.v
> index 0c608480..9e2f59bf 100644
> --- a/cores/aliens/hdl/jtaliens_main.v
> +++ [...]

Excellent hint. Many systems do not use the CPU write signal in the decoding logic. According to the original schematics, Thunder Cross did not either. I just added it and it actually did not work. I checked your code and you are also assigning some data to port_in. That change was not consistent with schematics because the CPU cannot read _rmrd_, etc. Nonetheless, it is a great hint! This commit fixes it in a consistent way:

```
index 0c608480..1fe60bcd 100644
--- a/cores/aliens/hdl/jtaliens_main.v
+++ b/cores/aliens/hdl/jtaliens_main.v
@@ -282,7 +282,10 @@ always @(posedge clk, posedge rst) begin
                         endcase
                     end
                     5: port_in <= A[0] ? dipsw[15:8] : dipsw[7:0];
-                    6: { pmc_start, pmc_bk, rmrd } <= cpu_dout[2:0];
+                    6: begin
+                        { pmc_start, pmc_bk, rmrd } <= cpu_dout[2:0];
+                        port_in <= cpu_dout; // required by Thunder Cross ROM test
+                    end
                     default:;
                 endcase
             end
```
I should take an action to apply routinely this approach, just in case. The logic behind this is that the data bus will hold the last value on it for a while and the next read instruction can see its value if it does not activate any other logic.
That could actually be the the real culprit. There must be some read instruction that might not be activating the right CS signal. But arcade software tends to have test leftovers, so it might be a legit mistake in the original code.

Anyway, I think this is done. Thanks a lot @gyurco. @jtmiki this issue is solved.

closed 2024-04-10 by jotego

gyurco, 2024-04-10:
Yes, the read back "something" is important here. Even reading back 0 will do the job here. I think it's a bug in the code, the devs didn't know they cannot read back the actual values of the regs.

jotego, 2024-04-10:
> The code in question 
```
LDA $1f98 
ORA #$01 ; turn on romrd 
STA $1f98
``` 
> As no readback is implemented for $1f98, and not even 'we' considered, the first LDA basically "writes" $FF which turns on everything, including starting PMC...

I had missed the first LDA when reading over the messages. It's funny that the software developer is following _good practice_ by trying to modify only the required bit, but his code is actually bad for the target hardware and works by mere chance.