jotego/jtcores #558: Outrun vertical white line in middle of screen

theenemy-thd, 2024-02-20:
This vertical white line appears down the middle of the screen at the start of the race when the car spins in and the guy waves the flag .. it does disappear quickly but its very annoying
Rgds
Alex

jotego, 2024-02-20:
Indeed. There is probably a missing _video off_ signal that must be connected. I'll look into it.

jotego, 2024-02-24:
The video enable signals were disconnected for debugging purposes. After connecting them back in the JTSHANON and JTOUTRUN cores, the video works as expected.
@theenemy-thd thanks for spotting this. It is fixed in 38014369198bcd6161a60b9267b5d3a9f29fcb8b and will be released in the next JTFRIDAY update

theenemy-thd, 2024-05-04:
Seems while the in game white vert line is fixed it does the same thing after you play a game and after scores are shown

jotego, 2024-05-04:
I cannot see that. Could you please check the version you are running a provide a screen shot? Thanks

theenemy-thd, 2024-05-06:
[image]
[image]
[image]

Here happens after you die and enter your name then it goes back to the outrun screen

jotego, 2024-05-06:
Thank you for the screen shots. I cannot reproduce this issue, nor have I seen it during development. It might be your ROM set or some old configuration file. Try deleting configuration files related to Out Run. Also try with a different ROM dump.

theenemy-thd, 2024-05-10:
Deleted everything to do with outrun and ran update all ... still has that white line after score screen shown.

Are you sure tou are starting a game and then dying ... allow it just to go through it screens and after it shows the score table (don't press anything) it show the line as per pics

jotego, 2024-05-12:
I was able to reproduce it. I think this is genuine. The hardware does have a bit that disables the video output and the game normally uses it so there is no video produced while the game makes a screen that takes extra time. It seems like that transition needs a long time but the software does not enable the bit.

The behavior is the same in MAME. Maybe older versions of the game have it fixed.

I do not think it is a core issue. Thank for reporting it.

closed 2024-05-12 by jotego

theenemy-thd, 2024-05-12:
Ahhh ok, well if it happens on real hardware then all fine then

Thanks for looking into it so quickly

jotego, 2024-05-13:
Thank you for looking for footage of the real board.