jotego/jtcores #559: cps1: adpcm filter

jotego, 2024-02-21:
First spectrum comparison for ADPCM, on commit fbb89f0

- the ac coupling in the PCB has a high pass effect visible in the spectrum, with 10Hz signals more attenuated than in the core. Not critical as these are not really audible
- the low-pass filter on the board has a clear effect and it is needed for sound accuracy (compare red curve with blue one)
- the current low-pass filter in the MiSTer core is too steep and it comes too late (see yellow curve)

[image]

jotego, 2024-02-21:
The steep attenuation comes from the interpolation filter inside jt6295. Using the _soft_ filter with `INTERPOLATOR=2`, results in the same profile (error in the coefficients?)

paulb-nl, 2024-02-22:
Which pcb are you targeting for ADPCM? Artemio showed that there big differences with older (GNG) vs newer pcb's (SF2).

jotego, 2024-02-22:
> Which pcb are you targeting for ADPCM? Artemio showed that there big differences with older (GNG) vs newer pcb's (SF2).

I need to collect more information. Ideally, I will match each game to its PCB. If anyone has a list of PCB ID code vs games, please share it.

ArtemioUrbina, 2024-02-22:
Hello @jotego, here is the Z80 code that loops between:
    20 Hz
    200 Hz
    1000 Hz
    2000 Hz

[CPS1-Z80loop.zip](https://github.com/jotego/jtcores/files/14371981/CPS1-Z80loop.zip)

Take care

jotego, 2024-02-23:
It looks like the data in that file is not correctly encoded as there are some HF oscillations in it.  I did my own ADPCM rom and plugged it with your Z80 code. [patches.bin.zip](https://github.com/jotego/jtcores/files/14388091/patches.bin.zip). But I still got waveforms that didn't look great.

The problem is that looking at the ADPCM alone, with no filtering, brings so much aliasing that signals look very distorted. When I add a 1st order low-pass filter, I can start seeing the sine waveforms appears as I pull the cut-off frequency lower and lower.

This makes the comparisons a trickier than expected.

I did find a regression error I introduced recently and could fix that. I also understand now that the interpolator I was using was removing aliasing too well in comparison with the real PCB. So we're getting results, but we're not quite there yet.

ArtemioUrbina, 2024-02-24:
> In case you want to try, using this file [jtcps1.rbf.zip](https://github.com/jotego/jtcores/files/14388210/jtcps1.rbf.zip) you can vary the ADPCM gain and the filter cut-off frequency.
> 
> Connecting a keyboard to the MiSTer, you can control the numbers visible at the bottom. By pressing shift and the numbers from 1 to 8, you will be able to alter each one of the bits making up that number.
> 
>     * Top four bits affect the filter cut-off frequency. The higher the number, the stronger the filtering
> 
>     * Lower four bits set the gain. Zero is no ADPCM, 1111 is maximum gain
> 
> 
> Thank you

Here are my results with the test core, excellent idea. Sorry I didn't have these sooner, but I was fixing some issues I had with the SDK. I made a test ROM that can run my regular MDFourier and an ADPCM only version, here are the results:

5F:
[image]

6F:
[image]

7F:
[image]

8F:
[image]

9F:
[image]

AF:
[image]

BF:
[image]

CF:
[image]

DF:
[image]

EF:
[image]

FF:
[image]

I ignored the amplitude parameter since it made no difference (here's FF vs F7):

[image]

Here's the recordings and profile:  [ADPCM_Only.zip](https://github.com/jotego/jtcores/files/14391441/ADPCM_Only.zip)

To my eyes AF and BF look the best from these, so here's how the spectrograms look for AF

PCB:
[image]

AF:
[image]

PCB:
[image]

AF:
[image]

Please let me know what you think

ArtemioUrbina, 2024-02-24:
PCB:
[image]
8F:
[image]

jotego, 2024-02-24:
Thank you very much. We are still not getting the right shape. I reviewed the signal chain again and made two changes:

- The low-pass filter must run at the full sampling rate (48kHz) instead of the ADPCM sampling rate in order to have the same effect as the PCB
- I changed the core so all 8-bits of the debug value go to the filter. The expected correct value is **E7**, which should give you a Fc of **770 Hz**, matching _Forgotten Worlds_ schematics. Use this file [jtcps1.rbf.zip](https://github.com/jotego/jtcores/files/14391995/jtcps1.rbf.zip). Amplitude is fixed.

The algorithm I used is [here](https://github.com/jotego/jt6295/blob/34b095bf5c0d988ced0ee09accde40296fffd06b/src/cmd/sine.go#L135). This comes originally from an old telephony book (page 275). You can browse it [here](https://archive.org/details/pctelephony00bobe/page/274/mode/2up?view=theater)

ArtemioUrbina, 2024-02-24:
I believe you'll like this 

> * I changed the core so all 8-bits of the debug value go to the filter. The expected correct value is **E7**, which should give you a Fc of **770 Hz**, matching _Forgotten Worlds_ schematics. Use [...]

Here are E6, E7 and E8. I didn't record others further on either side since it is way too late here, I can do that tomorrow for completeness. I download them to the same folder and compare them quickly changing between them with an image viewer

E6:
[image]

E7:
[image]

E8:
[image]

I like E8 better, but they are very similar:

PCB:

PCB:
[image]
E6:
[image]

PCB:
[image]
E7:
[image]

PCB:
[image]
E8:
[image]

> The algorithm I used is [here](https://github.com/jotego/jt6295/blob/34b095bf5c0d988ced0ee09accde40296fffd06b/src/cmd/sine.go#L135). This comes originally from an old telephony book (page 275). You can browse it [here](https://archive.org/details/pctelephony00bobe/page/274/mode/2up?view=theater)

Thank you

jotego, 2024-02-24:
Thank you very much for staying late to take the measurements. It looks like we're almost there. There is no need to take other measurements.

I think at this stage we can focus at a single SS value, to make things clearer, as we understand its effect well.

I am fine with the discrepancies at low frequencies as any ac coupling capacitor in the signal flow -and there will be a few of them- will alter the spectrum there without our control. But I am surprised to see so much _random_ variation in the error plots. It call my attention that you name it "amplitude". We should be comparing power and not amplitudes, otherwise the signal phase will cause funny effects. Maybe you're already comparing power and it is just an improper name in the difference plots.

ArtemioUrbina, 2024-02-24:
> I think at this stage we can focus at a single SS value, to make things clearer, as we understand its effect well.

Understood

> I am surprised to see so much _random_ variation in the error plots. It call my attention that you name it "amplitude". We should be comparing power and not amplitudes, otherwise the signal phase [...]

I am using sqrt(r1*r1 + i1*i1)) to calculate the values for each bin from the DFT. I am indeed a software engineer implementing these, out of my element. Any corrections are more than welcome to improve the work. So please feel free to comment, I won't be offended.

Regarding the random variation I attribute it to the aliasing and distortion caused by the ADPCM compression.

jotego, 2024-02-28:
It looks like the IIR filter causes some ringing. I will reimplement the filter as a FIR one. I am doing some preparatory works in other cores at the moment so I can apply a global solution to all cores. That is why I am quiet. I will report back in a few days

jotego, 2024-03-01:
I moved to a FIR implementation. Could you please measure this one [jtcps1.rbf.zip](https://github.com/jotego/jtcores/files/14460070/jtcps1.rbf.zip) ?

This is the spectrum. It should match the analog filter at 7700Hz, although at 770Hz the analog one has more attenuation. The analog one will have more attenuation at 14kHz too. I used this [online tool](https://fiiir.com/) to design it.

[image]

ArtemioUrbina, 2024-03-01:
> I moved to a FIR implementation. Could you please measure this one [jtcps1.rbf.zip](https://github.com/jotego/jtcores/files/14460070/jtcps1.rbf.zip) ?

Here are the results against the CPS1 88617A-7b (Magic Sword) PCB we were targeting:

[image]

It is interesting to note the comparison between 88617A-7b (ref, older and 89626A-4 (Comp, newer):

[image]

So, here is the 89626A-4 (Ref/SF2) and the new Core:

[image]

For reference here are the recordings:

[resultsFIR.zip](https://github.com/jotego/jtcores/files/14466262/resultsFIR.zip)

And the ROMs used (use button 3, HP for ADPCM only)

[sf2ud.zip](https://github.com/jotego/jtcores/files/14466265/sf2ud.zip)

And these:

[image]

[image]

Thanks for all your great work!

jotego, 2024-03-02:
Thank you. I have been trying several filter options. I think I am going to settle for:

- 2kHz 1st order filter for 88617A style boards
- 4kHz 2nd order filter for 89626A style boards

These two numbers make sense from a signal processing point of view and the spectra I get are reasonable. See plots below. The only thing is that the core seems to dip a bit in power around 1kHz, even without filtering. This is more noticeable in the plot using the 4kHz filter. Maybe the OKI chip did a different decoding for the ADPCM signal or applied some filtering of its own or it might be the effect of the VDD-related distortion we saw in the the board samples.

[image]
all spectra

[image]
2kHz filter

[image]
4kHz filter

What do you think, @ArtemioUrbina ?

jotego, 2024-03-02:
what is the frequency range you are sweeping? I wonder if you are going above the Nyquist frequency for the sampling rate and that is making the ADPCM algorithm do funny things... The sweep should go from 20Hz to 2 or 3 kHz. We will still get signal for higher frequencies because of quantization noise, aliasing and ADPCM artifacts.

ArtemioUrbina, 2024-03-02:
20 to 3787hz since the sampling is 7575 to my knowledge

closed 2024-03-03 by jotego

jotego, 2024-03-03:
I have finally settled for the single pole at 770Hz, as schematics suggest for the 88617A board, and a 2nd order Butterworth filter at 4kHz for the 89626A. I have made it selectable via the DIP switches menu. I have enabled the FX volume setting in the OSD for user flexibility.

This is the core file: [jtcps1.rbf.zip](https://github.com/jotego/jtcores/files/14474459/jtcps1.rbf.zip)

The _ideal_ filter responses are below. The single pole works quite well in the core but the Butterworth has worse performance in the fixed point math version (i.e. the core). This is often the case when going from float to fixed point, but it is a bit worse than expected.

The FM/PCM sound balance might not be exact. I find it difficult to derive from the recordings, to be honest as too many things are moving. That's one of the reasons I left open the user option to adjust it.

Let me know what you think and thanks again for all the support!

[image]
[image]
[image]

ArtemioUrbina, 2024-03-05:
> I have finally settled for the single pole at 770Hz, as schematics suggest for the 88617A board, and a 2nd order Butterworth filter at 4kHz for the 89626A. I have made it selectable via the DIP switches menu. I have enabled the FX volume setting in the OSD for user flexibility.

Selectable is the perfect optio0n in my opinion. I couldn't find it in the file linked above yet, but did find the FX volume setting.
 
> The FM/PCM sound balance might not be exact. I find it difficult to derive from the recordings, to be honest as too many things are moving. That's one of the reasons I left open the user option to adjust it.

I used MDfourier to measure it against both PCB setups. And high (the default) seems to be the most appropriate

**88617A-7b vs Highest:**

[image]

**88617A-7b vs High:**

[image]

**88617A-7b vs Low:**

[image]

**89626A-4 vs Highest**

[image]

**89626A-4 vs High**

[image]

**89626A-4 vs Low**

[image]

Source recordings:

**PCBs:**

[CPS1-89626A-4-SF2Board-ADPCM-Full.zip](https://github.com/jotego/jtcores/files/14488603/CPS1-89626A-4-SF2Board-ADPCM-Full.zip)

[CPS1-88617A-7b-MagicSword-ADPCM-Full.zip](https://github.com/jotego/jtcores/files/14488615/CPS1-88617A-7b-MagicSword-ADPCM-Full.zip)

**Mister:**
[MisterFPGA-CPS1-04032024-hi-lo.zip](https://github.com/jotego/jtcores/files/14488593/MisterFPGA-CPS1-04032024-hi-lo.zip)

 
> Let me know what you think and thanks again for all the support!

No thank you for caring about audio and your great work! Let me know if I can be of further assistance

commit 22b08c09 (2024-03-03): two options for PCM filter in CPS1. The single pole filter fits quite well the available data. The 2nd order IIR, does not fit it so well, probably because of rounding errors in the coefficients. Although the exact ones do not give perfect matching either... Fixes #559