jotego/jtcores #560: btiger: Screen not blanking during transitions

BMF54123, 2024-02-23:
During certain screen transitions, the core is not completely blanking the screen as it should. For example, when transitioning from the high score screen to the gameplay demo, the "Black Tiger" logo remains onscreen; on both my bootleg PCB and in MAME, the entire screen goes black until the demo starts (also confirmed on a friend's authentic Capcom PCB).

jotego, 2024-02-23:
There might be a _video enable_ signal missing. I will check it.

jotego, 2024-02-24:
All three control bits for video signals are set low, but the char layer remains on screen

jotego, 2024-02-24:
Thans for spotting this, @BMF54123. This is fixed now and I will release it in the next JTFRIDAY update.

commit d5656e0d (2024-02-24): btiger: video enable bits connected to individual GFX modules rather than colormix. Fixes #560
closed 2024-02-24 by jotego