jotego/jtcores #562: identificar

jotego, 2024-03-01:
Escribir en el fichero cfg/mem.yaml una sección audio describiendo el filtro RC paso bajo del chip de FM. Por ejemplo, para aliens sería:

```
audio:
  - name: fm
    stereo: true
  - name: pcm
    data_width: 12
    rout: 100k
    res: 20k
    cap: 470p
```
Donde **rout** es la resistencia de salida del DAC, res la resistencia serie y **cap** el condensador a masa.

- [ ] castle
- [ ] comsc
- [ ] cop
- [ ] dd
- [ ] dd2
- [ ] flane
- [ ] fround
- [ ] kchamp
- [ ] kicker
- [ ] kiwi
- [ ] kunio
- [ ] midres
- [ ] mx5k
- [ ] ninja
- [ ] outrun
- [ ] pang
- [ ] rastan
- [ ] roadf
- [ ] s16
- [ ] s16b
- [ ] sbaskt
- [ ] sf
- [ ] shanon
- [ ] shouse
- [ ] simson
- [ ] tmnt
- [ ] tora
- [ ] track
- [ ] trojan
- [ ] twin16
- [ ] vigil
- [ ] yiear

jotego, 2024-03-01:
# aliens

rout: R2R, 100 kOhm
res: 20k
C: 270p
fcut: 2.8 kHz

[image]

etriana85, 2024-03-01:
# TMNT
rout: R2R, 100 kOhm
res: 20k
C: 270p
fcut: 2.8 kHz
[image]

etriana85, 2024-03-01:
# castle
rout: R2R, 100 kOhm
res: 20k
C: 470p
fcut: 2.82 kHz
[image]

etriana85, 2024-03-01:
# fround
rout: R2R, 100 kOhm
res: 20k
C: 470p
fcut: 2.82 kHz 
[image]

etriana85, 2024-03-01:
# simson
rout: 1000 Ω
C: 33pF
fcut: 4.82 kHz

[image]

etriana85, 2024-03-01:
# cop
rout: uk
res: 30k
C: 10nF
fcut: 5.31 kHz
[image]

etriana85, 2024-03-01:
# kicker
Primera y segunda etapa
rout: uk
res: 1k
C: 470pF
fcut: 3.39 kHz

[image]

etriana85, 2024-03-01:
# dd
Chip de sonido 5205
Aparentemente este chip tiene el  o los filtros integrados.
[image]

# dd2
No hay sch para consultar

etriana85, 2024-03-01:
# kchamp
Chip de sonido 5205
Aparentemente este chip tiene el o los filtros integrados.
[image]

etriana85, 2024-03-01:
# kunio
Chip de sonido 5205
Aparentemente este chip tiene el filtro integrado
[image]
Enlace a hoja de datos del chip 5205
https://wiki.console5.com/tw/images/f/f8/MSM5205.pdf

etriana85, 2024-03-01:
# shouse
Chip de sonido cus30
Aparentemente este chip tiene el filtro integrado
[image]

etriana85, 2024-03-04:
# midres
No hay sch para consultar

etriana85, 2024-03-05:
# mx5k
rout: R2R, 100 kOhm
res: 20k
C: 470p
fcut: 2.8 kHz
[image]

etriana85, 2024-03-05:
# ninja
rout: uk
res: 10k
C: 10nF
fcut: 1.59 kHz

2da etapa
res: 20k
C: 150pF
fcut: 5.31 kHz
[image]

etriana85, 2024-03-05:
# outrun
rout: uk
res: 10k
C: 0.1nF
fcut: 1.59 kHz
[image]
 
Por verificar
[image]

etriana85, 2024-03-05:
# pang
jt6295
rout: uk
res: 20k
C: 3.3nF
fcut: 2.41 kHz

jt2413
rout: uk
res: 4.72k
C: 15nF
fcut: 2.25 kHz

2da etapa
rout: uk
res: 4.72k
C: 15nF
fcut: 1.42 kHz ? 2.84 kHz?
[image]

etriana85, 2024-03-05:
# rastan
ym3012?
rout: uk
res: 18k
C: 10nF
fcut: 8.84 kHz
[image]

jt5205
rout: uk
res: 8.6k
C: 10nF
fcut: 1.85 kHz 

2da etapa
rout: uk
res: 10.2k
C: 68nF
fcut: 2.29 kHz 
[image]

etriana85, 2024-03-05:
# roadf
rout: uk
res: 10k
C: 47pF
fcut: 3.39 kHz

2da etapa
rout: uk
res: 10k
C: 330pF
fcut: 4.82
[image]
 kHz

etriana85, 2024-03-06:
# s16
rout: 
res: 
C: 
fcut: 

# s16b
rout: uk
res: 15.07k
C: 2.2n
fcut: 4.8kHz

2da etapa
rout: uk
res: 7.5k
C: 10n
fcut: 2.12kHz
[image]

etriana85, 2024-03-06:
# sbaskt
rout: uk
res: 10k
C: 47n
fcut: 3.39kHz

2da etapa
rout: uk
res: 10k
C: 330p
fcut: 4.82kHz
[image]

etriana85, 2024-03-06:
# sf (alternativa tora?)
No hay sch para consultar. Descargué toda la información que encontré en la web de sf 
rout: 
res: 
C: 
fcut:

#  shanon (mismo caso de outrun)
rout: uk
res: 66k
C: 100n
fcut: 2.41kHz

[image]

https://pdf1.alldatasheet.com/datasheet-pdf/download/167323/AD/AD7520.html

etriana85, 2024-03-06:
# tora
rout: uk
res: 15k
C: 470p
fcut: 2.26kHz

2da etapa
rout: uk
res: 10k
C: 22n
fcut: 7.23kHz

3ra etapa
rout: uk
res: 10k
C: 6.8n
fcut: 2.34kHz

[image]
[image]

etriana85, 2024-03-06:
# track
rout: uk
res: 10k
C: 47n
fcut: 3.39kHz

2da etapa
rout: uk
res: 108.2k
C: 330p
fcut: 4.46kHz
[image]

etriana85, 2024-03-06:
# trojan
No hay sch para consultar
rout: uk
res: 
C: 
fcut: 

# twin16  Dark Adventure?
rout: uk
res: 1k
C: 33n
fcut: 4.82kHz
[image]

upd7759
rout: uk
res: 5.7k
C: 10n
fcut: 2.79kHz
[image]

etriana85, 2024-03-06:
# vigil
jt51
rout: uk
res: 4.7k
C: 2.2n
fcut: 1.54kHz

2da etapa
rout: uk
res: 20k
C: 1n
fcut: 7.96kHz

[image]

ym3014
rout: uk
res: 4.7k
C: 10n
fcut: 3.39kHz

2da etapa
rout: uk
res: 20k
C: 4.7n
fcut: 1.69kHz
[image]

etriana85, 2024-03-06:
# yiear
rout: uk
res: 139k?
C: 47n
fcut: 2.44kHz

2da etapa
rout: uk
res: 10k
C: 330p
fcut: 4.82kHz
[image]

76489 AN
res: 1K
C: 47n
fcut: 3.39kHz
https://map.grauw.nl/resources/sound/texas_instruments_sn76489an.pdf

skutis, 2024-03-07:
uPC324 will have an output impedance of 50..200ohms on the high side. My guess it is ~100ohm. When it is HP filtered afterwards, the effect will only be half of it. So the practical output impedance is around 50ohm. If we have for instance a series resistor of 1k after the op amp it will add 5%. 1/sqrt(1.05) = 0.975 -> it will decrease the LP cutoff with 2.5%.

skutis, 2024-03-07:
Dark Adventure

[image]

The LP cutoff for the PCM channels are
Rout: 100k
Res: 20k
C: 0.47n
fcut: 2.82kHz

jotego, 2024-03-08:
> ## kicker
> 
> rout: uk res: 1k C: 470pF fcut: 3.39 kHz

Debe estar mal. 1k y 470pF da **338 kHz**, no 3.39kHz. Es demasiado. Comprueba los valores.

closed 2024-03-11 by etriana85