jotego/jtcores #564: jts16/Shinobi - player is 1 pixel too high.

jonxs1976, 2024-03-02:
Player is 1 pixel too high, more noticable when on the upper level, this only happens with the following .mra's, the other .mra in the _Alternative folder seem correct.

**_Arcade**
Shinobi (Set 6, World, S16A) [No Protection].mra

**_Alternative**
Shinobi (Set 1, World, S16A) [FD1094 317-0050].mra

[image]

jotego, 2024-03-02:
Can you provide a non-emulated image with the right position?

jotego, 2024-03-06:
Objects are off by one pixel in S16 core with respect to S16B apparently.

jotego, 2024-04-07:
The object layer misalignment is clearer in the S16A version of WB3

[image]

jotego, 2024-04-07:
Fixed. I'll release the update soon.

commit 4b1da664 (2024-04-07): s16a: obj alignment corrected. Fixes #564
closed 2024-04-07 by jotego