jotego/jtcores #576: tora: comparar circuito en esquemático con placa F1-Dream

jotego, 2024-03-09:
El circuito del MSM5205 podría ser diferente en la placa de F1-Dream. Hacer una comparación rápida.

etriana85, 2024-03-13:
R13 está ausente o fue cortada en la PCB F1 
[image]
[image]

etriana85, 2024-03-13:
- [ ] Comprobar si el pin 5 del amplificador 13B conecta al punto M realmente (pág 9)  **Sí**
- [ ] Verificar el valor de R30 y R31 (¿58k?) (pág 8)  **56k**
- [ ]  Verificar el valor de C45 (¿1nF?) (pág 8) **Sí**
- [ ] En la página 8, hay un _resistor array de 1kOhm, RA4?, y otro. ¿cuál es el valor y el esquema de las conexiones en el     resistor array de la derecha? **56k**

[image]

jotego, 2024-03-13:
Gracias. Sobre el _resistor array_ me refería a este bloque de puntos:

[image]

Necesito saber qué estructura de resistencias hay dentro.

closed 2024-03-22 by jotego