jotego/jtcores #578: kicker: condensador demasiado grande

jotego, 2024-03-10:
Los condensadores CC10 y CC12 están marcados como 104M, lo que indica 100nF±20%. 100nF es demasiado. Saldría una frecuencia de corte para el filtro de 200Hz ([simulacion](https://www.circuitlab.com/editor/#?id=s82443x933gq)). Revísalo en la placa. Comprueba también las resistencias y el otro condensador que forman el filtro.

Resimula con los nuevos valores. Esperaría que la frecuencia de corte estuviera en el rango 2-20kHz. Esto afecta también al Ping Pong. Actualiza los esquemáticos de Ping Pong y Kicker con los nuevos datos.

[image]

etriana85, 2024-03-13:
Condensadores:
CC10, CC12 104K  (100nF±10%)
R21, R24 10 

Los filtros:
R20, R23 1K
Cp 47nF 
[image]

etriana85, 2024-03-13:
[image]

etriana85, 2024-03-13:
He puesto el mismo valor de kicker,  a los condensadores de pin pong.aa
[image]

etriana85, 2024-03-13:
[image]
[image]

closed 2024-03-22 by jotego