jotego/jtcores #584: CPS1: Street Fighter 2 Hyper Fighting runs too fast

icypottle586, 2024-03-17:
Hi,

Street Fighter 2 Hyper Fighting runs way too fast.  For reference: https://www.youtube.com/watch?v=UUTSosokEDk

jtcps1.rbf of 3/17/2024 takes roughly that 47s range to finish just like the MAME-rr in that video.

I grabbed a random older version from last year: https://github.com/jotego/jtbin/blob/b1f9fe2f0f5269e7106aca49eea67364fbc2e1b2/mister/cps1/releases/jtcps1.rbf and it takes roughly 51s or so, still too fast compared to real PCB but not as much as more recent jtcps1.rbf

and looks like it's still broken in MAME?  https://mametesters.org/view.php?id=00408

HabbleFox, 2024-04-21:
If it helps as an additional reference, 2 years ago I made a comparison video for the speedrun community based on the core versions we had on MiSTer at the time. https://youtu.be/DvIe35lBntU

icypottle586, 2024-05-06:
Hi @jotego , I see you tagged it as question, do you need more information from me about this?

jotego, 2024-05-07:
Thank you but it is a question for me. I do not mark items as bugs until I have checked them.

real-amano, 2024-05-09:
I think that SF2 Turbo was always a conversion of other SF2 boards. Might the speed possibly vary depending on the board hardware that was converted?

jotego, 2024-05-11:
Using the test Grego2d built a long time ago, the speed we are getting is:
 
Commit  | speed (hex)  | Date
--------|--------------|-----
4f7da4b | 3596         | May '24
ea01437 | 3596         | Nov '23

The result should be 2C00, as measured in the PCB.

The whole core, not just SF2HF, has been running too fast for the last six months. This is probably related to changes in the jtframe_68kdtack module.

jotego, 2024-05-12:
This patch restores the speed:

```
--- a/modules/jtframe/hdl/cpu/jtframe_68kdtack_cen.v
+++ b/modules/jtframe/hdl/cpu/jtframe_68kdtack_cen.v
@@ -77,12 +77,12 @@ module jtframe_68kdtack_cen
 localparam CW=W+WD;
 
 reg [CW-1:0] cencnt=0;
-reg  [1:0]   waitsh;
+reg  [2:0]   waitsh;
 wire         halt;
 wire [W-1:0] num2 = { num, 1'b0 }; // num x 2
 wire over = cencnt>den-num2;
 reg  [CW:0] cencnt_nx=0;
-reg         risefall=0, wait1;
+reg         risefall=0;
 
 `ifdef SIMULATION
     // This is needed to prevent X's at the start of simulation
@@ -93,25 +93,22 @@ reg         risefall=0, wait1;
     wire rstl=0;
 `endif
 
-assign halt = !rstl && RECOVERY==1 && !ASn && {waitsh,wait1}==0 && (bus_cs && bus_busy && !bus_legit);
+assign halt = !rstl && RECOVERY==1 && !ASn && waitsh==0 && (bus_cs && bus_busy && !bus_legit);
 
 
 always @(posedge clk) begin : dtack_gen
     if( rst ) begin
         DTACKn <= 1;
         waitsh <= 0;
-        wait1  <= 0;
     end else begin
         if( ASn | &DSn ) begin // DSn is needed for read-modify-write cycles
                // performed on the SDRAM. Just checking the DSn rising edge
                // is not enough on Rastan
             DTACKn <= 1;
-            wait1  <= 1; // gives a clock cycle to bus_busy to toggle
-            waitsh <= {wait3,wait2};
-        end else if( !ASn ) begin
-            wait1 <= 0;
-            if( cpu_cen ) waitsh <= waitsh>>1;
-            if( waitsh==0 && !wait1 ) begin
+            waitsh <= {wait3,wait2,1'b1};
+        end else if( !ASn && cpu_cen ) begin
+            waitsh <= waitsh>>1;
+            if( waitsh==0 ) begin
                 DTACKn <= DTACKn && bus_cs && bus_busy;
             end
         end
```

Needs to check that it does not break Dynamite Dux, Shinobi or Chelnov

closed 2024-05-12 by jotego

jotego, 2024-05-12:
Fixed without altering other M68000 cores. CPS1 needs to always wait for a positive CPU cen, whereas System 16 does not. Schematics are always a bit tricky to follow around the DTACK signal and because the dtack module is common to all M68000 cores, it requires a lot of attention to configuring it correctly.

The files will get out on the next JTFRIDAY.

icypottle586, 2024-05-13:
awesome news!  thank you so much @jotego !!!

icypottle586, 2024-05-24:
Hi @jotego I have tried the latest 94d0874 jtcps1.rbf, however I see no difference between that vs the earlier 4f7da4b version.  Both of those still takes roughly 47s to finish the 99 seconds round, unlike the 55s range from https://www.youtube.com/watch?v=2nJiMnxXYVA.

icypottle586, 2024-05-24:
@jotego I have tried the latest 22e0205 and it's the same.  I have tried all regions of Street Fighter II': Hyper Fighting (World 921209) and there has been only one revision.

```
sf2hf.zip
sf2hfj.zip
sf2hfu.zip
```

All the romsets are correct as both the core and MAME load them with no complaint.  And I have not changed any DIPs, i.e., nothing under `/media/fat/config/dips`.

Please let me know what else can I do to help.  As I mentioned the 210131 build was perfect (https://www.youtube.com/watch?v=2nJiMnxXYVA) and the discrepancies of wall-clock time (about 8 seconds between that build and now) is very easy to tell. 

Is that `jtcps1_210131.rbf` build still obtainable somehow?

jotego, 2024-05-27:
The above screen shot is a comprehensive test where the CPU bus timing is tested against the different hardware elements that can affect it. It had gotten 20% too fast after a change made for a different core. That was corrected.

It does not make sense that this version has the CPU running 20% slower than the previous one and you still get the same result. 

If you still get a different speed, the only element that can be different is the frame rate that your board oscillator produces. The frame rate controls the game speed as long as the CPU speed is not limiting it. 

The core operates at 59.63Hz. Your board may be operating at a different speed because components may have drifted off the original specifications.

jotego, 2025-12-12:
There were some comments about this outside of GitHub. My view of this issue:

- There are two elements that affect the speed: the DMA implementation and the CPU speed
- The DMA system is accurate as we have a custom firmware to test it on a real board
- The CPU used on an original SF2HF board is reported to be 12MHz on MAME documentation. But users may have board conversions with regular 10MHz CPUs. Or maybe the 12MHz figure is wrong
- Maybe the board has more differences to regular CPS1 games that have not been reported. As the number of board conversions being sold is high, you have to take reports with a grain of salt

commit a3a96e7a (2024-05-12): cps1: dtack always wait 1 cpu_cen. Fixes #584, restoring the right core speed