jotego/jtcores #585: Outrun: Blue Line on Left Side of Screen on Music Select + Rear Tail Lights Flicker and Vanish

Jlmbles, 2024-03-22:
Flickering Tail Lights which eventually turn grey. Also a thin blue line on the left side of the screen when selecting a music track.
[image]

Dawbo66, 2024-03-22:
@jotego when you press the break button the back lights at the top go see through and its the road your seeing through the lights, so the flickering is actually the white lines on the road, the back lights should be solid, when the break lights come on at the bottom the top lights go see through.. So to recreate drive over the white lines and break ,you'll see the lines where the top lights should be..

Dawbo66, 2024-03-23:
@jotego when you press the break button the back lights at the top go see through and its the road your seeing through the lights, so the flickering is actually the white lines on the road, the back lights should be solid, when the break lights come on at the bottom the top lights go see through.. So to recreate drive over the white lines and break ,you'll see the lines where the top lights should be..Even when going fast on a bend and press the break you can see the white road lines as you go over them. sorry about the crude drawing..😊
[image]

[image]
[image]

Jlmbles, 2024-03-29:
Also this seems to trigger at the first checkpoint.

jotego, 2024-05-06:
Thank you for the detail screen shots. I can see the problem with the rear lights when the brakes are on. I cannot see the blue line mentioned in the issue title, though.

jotego, 2024-05-09:
Fixed the rear lights by removing a piece of logic in colmix derived from the PAL dumps that did not make quite sense. See d7cdc1b1

jotego, 2024-05-09:
Blue line fixed in f3c11490 by adjusting the tilemap position.

closed 2024-05-09 by jotego