jotego/jtcores #588: Pang (World) - Graphical glitch on the ladders

Urotsukidoji, 2024-03-24:
Hi there!

  Several people, included me, have seen what it seems to be a graphical glitch on the game Pang (World). 
  On the screens where a ladder/s appear, in the left ladder appear a little strange green object.

  Cheers.

jotego, 2024-03-25:
Please attach a screenshot if you have one handy

Urotsukidoji, 2024-03-25:
Sure, 

    Here's an image.

 Cheers (S
[image]
aludos)

vastago, 2024-04-04:
Además de las escaleras está el tema del efecto de sonido cuando rompes las bolas, se escucha demasiado alto con respecto a la musica. Esto ocurre en Mister, en cambio en mist parece que está todo correcto (al menos con el core antiguo de mist que estoy corriendo)

jotego, 2024-04-08:
stairs: regression introduced in a7c531f

jotego, 2024-04-08:
Stairs are visible in simulation: `jtsim -inputs -video 510 -w 490` in 20824ee6
[sim_inputs.zip](https://github.com/jotego/jtcores/files/14906787/sim_inputs.zip)

[image]

jotego, 2024-04-08:
fixed

commit d036b773 (2024-04-08): pang: fixes missing stair sprite. Fixes #588
closed 2024-04-08 by jotego