jotego/jtcores #589: shouse: Marchen Maze (mmaze) does not boot

jotego, 2024-03-24:
@gyurco it looks like this one does not boot. It gets halted at a screen showing `TESTING C64`. Could you check it, please?

gyurco, 2024-03-24:
I assumes it needs a preloaded NVRAM. I don't know if it is implemented already or totally missing? Can you give me a pointer where can I find it in jtshouse_game? Or where should I add support if it isn't there yet?

jotego, 2024-03-24:
I see. It is possible to add default data for the NVRAM in the MRA. The file `doc/jtframe-mra.md` shows how to do it (look for nvram). The same help contents are shown with the command `jtframe mra --help`

gyurco, 2024-03-24:
I tried actually with giving it an MMAZE.RAM file (copied from mmaze.zip - it should be loaded on SiDi/MiST at boot), but didn't help. However I don't see where the NVRAM part is implemented in the core.

jotego, 2024-03-24:
It is in the mem.yaml file, at the bottom:

```
  # EEROM
  - name: eerom
    addr_width: 11
    data_width: 8
    din: mcu_dout
    rw: true
    ioctl: { save: true, restore: true, order: 0 }
```

As it is maked as `restore: true`, it should be able to accept data.

jotego, 2024-03-24:
These mem.yaml files are parsed to generate other files. I suppose you have already found them in the sidi128 folder as they placed there for synthesis.

gyurco, 2024-03-24:
The issue is: mmaze.nv is already at the MRA, however prog_we remains activated when it's reached, effectively overwriting the MCU prog rom by the supposed-to-be nvram. Removing it fixes the issue, however I'm not sure how to add it to only programming the EEPROM with this content.

gyurco, 2024-03-24:
There's a
JTFRAME_PROM_START=0x690000
but should be a
JTFRAME_PROM_END=0x690FFF
or something equal

gyurco, 2024-03-25:
```
diff --git a/cores/shouse/hdl/jtshouse_game.v b/cores/shouse/hdl/jtshouse_game.v
index 23560ab7..d3a8c761 100644
--- a/cores/shouse/hdl/jtshouse_game.v
+++ b/cores/shouse/hdl/jtshouse_game.v
@@ -175,9 +175,9 @@ jtshouse_mcu u_mcu(
     .dip_test   ( dip_test  ),
 
     // PROM programming
-    .prog_addr  (prog_addr[11:0]),
+    .prog_addr  ( prog_addr[11:0] ),
     .prog_data  ( prog_data ),
-    .prog_we    ( prom_we   ),
+    .prog_we    ( prom_we & !prog_addr[12] ),
 
     // EEROM
     .mcu_addr   ( eerom_addr),
```

Solves the overwirte problem.

I'm not sure about the eeram preloading. Adding a second port to eerom helps:
```
    .clk1    ( clk ),
    .addr1   ( prog_addr[10:0] ),
    .data1   ( prog_data ),
    .we1     ( prom_we & prog_addr[12] ),
    .q1      ( )
```
But it's in jtshouse_game_sdram.v, which is auto-generated.

These two changes makes the game start, just need to integrate the second part with the framework.

jotego, 2024-03-25:
I see. Thank you for the clear explanation. PROM data does not have an end marker because it should be at the end. The problem is the NVRAM region in the .rom showing up after it.

The framework deals with NVRAM through a different ROM index, so it cannot be in the same file.

I have added support for explicit NVRAM files in the MAME ROM set. This is straightforward for MiSTer as you can specify different files for different download indexes in the MRA. For the Analogue Pocket/MiST, it requires users to create one more file, which is troublesome. Can SiDi128 read .mra files directly? If it has an MCU strong enough, we should go for that and ditch the .rom files as they are cumbersome to users.

Although the commit looks like a lot of changes, a lot of it is simple refactoring.

closed 2024-03-25 by jotego

gyurco, 2024-03-25:
It might be possible to add an MRA parser, just need an XML parser without big mallocs (while the MCU is more powerful, it still has only 384 KB SRAM), and an unzipper, thus .ROM/ARC files are still needed for a while.

jotego, 2024-03-25:
I have used parsers that only read one element at a time in the past, they were light and fast. The problem comes when they try to read the whole file and create an object model. That adds a lot of overweight which is not needed for our use case.

gyurco, 2024-03-25:
Yes, a streaming parser what is needed. Maybe even one (full) element a time is not a good thing, as it can contain explicit big ROM data in <part></part>.

BTW, does the resulting MRA works on MiSTer? Because I tried with an mmaze.ram file (which does the same), and it still needs one reset after loading. I think the core reset is released before the nvram is uploaded, so the MCU stucks.

jotego, 2024-03-25:
It works correctly in MiSTer. I think NVRAM reads/writes do not reset the core -particularly true for dumping to SD- but maybe MiSTer is just faster and gets the NVRAM written before the MCU gets to that point.

commit 3891fe11 (2024-03-25): nvram region files added to MRA automatically. Fixes #589