jotego/jtcores #590: analogue pocket  cps core  issues

sonicteam77, 2024-03-26:
When loading the Japanese version of CPS ROM, the DIP SWITCH option disappears in the core settings menu, and Allow to continue is turned off.

jotego, 2024-03-26:
Could you give more details? Which game are you trying to load?

sonicteam77, 2024-03-26:
Thanks for your reply. When I load the Japanese version of the CPS game US Navy, I can't see any CPS setting options in the core options, such as difficulty adjustment, screen flip and other DIP SWITCH settings. But these options are available when loading the world version CPS game. And the game's allow continue is also turned off.

sonicteam77, 2024-03-26:
Another problem is that the Japanese version of the CPS vertical shooter 1941 and varth screens are not rotated for the analogue pocket, but the screen orientation of the us version of the game is correct.

sonicteam77, 2024-03-27:
Thank you very much for your reply. I have solved this problem. Just copy the files in the alternative folder in PRESETS.

jotego, 2024-03-27:
I'm glad you figured it out

closed 2024-03-27 by jotego