jotego/jtcores #593: Stretch CRT width doesn’t work on Pang/Robocop/Rastan cores analogue pocket

ruffianes, 2024-03-30:
Stretch CRT width function doesn’t work on Pang/Robocop/Rastan cores analogue pocket.

jotego, 2024-03-31:
Does it work in any core at all? It might be incompatible with the Pocket dock.

jotego, 2024-03-31:
How are you exercising the stretch? It is not an option of the core...

ruffianes, 2024-03-31:
Hello, the option is available on all cores together with the integre +. Before the last cores update, this option worked.

ruffianes, 2024-03-31:
In fact this function is available and works with all your cores, except the ones I mentioned. By the way, amazing work you do.

jotego, 2024-03-31:
The issue title says _Analogue Pocket_. Are you sure you don't mean _MiSTer_?

ruffianes, 2024-03-31:
Sure. The issue is on analogue pocket.

espiox, 2024-03-31:
“Stretch CRT Width” is a new option added in Analogue OS 2.1 (in January). It stretches (or contracts) the left/right edges of the CRT Trinitron Display Mode to fit the screen.

jotego, 2024-05-05:
@rp-jt, please check out how this _stretch CRT width_ option works on the Analogue Pocket test the case described in the issue.

jotego, 2024-10-09:
It seems like the option works by stretching the image but without cropping it. On the mention cores, stretching it so the image takes the whole screen would imply cutting off some of the left and right edges, so the Analogue Pocket ignores the setting.

closed 2024-10-09 by jotego