jotego/jtcores #596: tnzs: dac circuit

jotego, 2024-04-01:
Hi @skutis,

I think the device values in the **HA17408P** DAC signal path are wrong as the gain-frequency looks very awkward. I am a bit surprised at the number of decoupling caps (C13, C15 and C11), but that is probably correct.

Drawing R23 vertically and upwards, and R29 vertically downwards would be better for schematic reading too.

I do not know the output impedance of the HA17408. The data sheet is very poor. I have ordered some to make measurements. So the gain in this plot may be wrong. That should not affect the corner frequencies, though.

Thanks

[image]

skutis, 2024-04-07:
C14 was 30pF, I got it wrong. This sets the second LP filter cutoff at 354kHz. The values are probably different because they don't want the DAC to sound that much.I don't know what kind of sounds are supposed to be played there?

jotego, 2024-04-08:
If those are the values, I think the filter was not designed correctly. It wouldn't be the first board like that!
Thanks for checking it

closed 2024-04-08 by jotego

skutis, 2024-04-08:
I don't know the bandwidth of the op-amps, the old ones were typically 1MHz. So it might be to just limit the bandwidth second LP filter.