jotego/jtcores #608: sidi128: bad STA

jotego, 2024-04-10:
The following cores fail STA when compiled for SIDI128. Just checked 1942 and the problem was in the BA lines to the SDRAM at the 85ºC corner. If it was possible to limit STA to low temperatures, I would be fine for that, but the next lowest temperature is 0ºC, which is too low. The FPGA will be a bit hotter than room temperature at least.

It might get solved by fixing the PLL assignments, or trying to make the SDRAM controller placement further up, closer to the SDRAM pins. The PnR seems to put everything around the center so it gets far from the pins.

- [ ] 1942
- [ ] 1943
- [ ] aliens
- [ ] biocom
- [ ] btiger
- [ ] bubl
- [ ] castle
- [ ] commnd
- [ ] cop
- [ ] cps15
- [ ] dd
- [ ] dd2
- [ ] exed
- [ ] fround
- [ ] gng
- [ ] gunsmk
- [ ] karnov
- [ ] kchamp
- [ ] kicker
- [ ] kiwi
- [ ] kunio
- [ ] midres
- [ ] ngp
- [ ] ninja
- [ ] outrun
- [ ] pang
- [ ] rastan
- [ ] roadf
- [ ] rumble
- [ ] s16
- [ ] s16b
- [ ] sarms
- [ ] sectnz
- [ ] sf
- [ ] shanon
- [ ] shouse
- [ ] simson
- [ ] slyspy
- [ ] tmnt
- [ ] toki
- [ ] tora
- [ ] track
- [ ] trojan
- [ ] vigil
- [ ] yiear

gyurco, 2024-04-10:
As I see most problematic are the pins which couldn't be put to fast output regs.
Also I wonder if a new set of PLLs should be created for the device. The SDRAM_CLK (and SDRAM2_CLK) pins are dedicated PLL outputs, however in hardware, only c0 is going to the dedicated output, and now c2 routed to the ram clocks. This actually applies to MiST/SiDi(32) too.

gyurco, 2024-04-10:
For the record, PLL locations in Cyclone 10:
https://www.intel.com/content/www/us/en/docs/programmable/683777/current/gclk-network-clock-source-generation.html

External clock outputs:
https://www.intel.com/content/www/us/en/docs/programmable/683777/current/external-clock-outputs.html

The SiDi128 board is designed in a way that PLL2_CLKOUTp pin is SDRAM_CLK and PLL4_CLKOUTp is SDRAM2_CLK.

jotego, 2024-04-11:
That is a good point. Using dedicated hardware outputs is the best practice. I changed that in branch sidi128_sta, just for the 48MHz PLL, which covers all cores except JTCPSx. The timing errors are gone but there are warnings related to which port of the PLL is used for compensation. The cores are not stable either and crash or have graphic glitches after a couple of minutes on screen (only tested JT1942 though).

The positive side is that the reported timing is clean, so the change is beneficial. I have tried to set the compensated clock in a different sdc file for sidi128, but that did not fix the warnings. There might be something else. sidi128 may also need a different delay, rather than the 5ns I set.

gyurco, 2024-04-11:
My experience if the dedicated PLL outputs are used, there's no need for any phase shift below 100 MHz. 0.58ns was needed for SNES at 128MHz. I'll investigate more thoroughly. Actually the original code, while there were some timing errors, was stable.
Warning to compensation to source clock is not really relevant, e.g. nothing depends on between the phase of the PLL's source clock and output clocks.

gyurco, 2024-04-11:
Shouldn't be 'ifndef?
https://github.com/jotego/jtcores/blob/312446f6eb5d48375f0efb5ae4db8542aa7e5923/modules/jtframe/target/mist/jtframe_mist_clocks.v#L86

jotego, 2024-04-11:
> My experience if the dedicated PLL outputs are used, there's no need for any phase shift below 100 MHz. 

That would be the best. I think there's a good chance it will work.

> Actually the original code, while there were some timing errors, was stable. 

It was stable because probably the timing was good at that temperature. A 48MHz clock using a small fraction of the FPGA resources should not heat the device much. The problem there is that we do not have timing models between 0 and 85ºC so we cannot guarantee stability at say 50ºC.

> Warning to compensation to source clock is not really relevant, e.g. nothing depends on between the phase of the PLL's source clock and output clocks.

I'd say so too. I may just waive those warnings in jtcore.

I wonder if we can just swap the output clocks in the original PLL, rather than creating a new one as I did in the branch. I think the other FPGAs using the mist clocks module do not have a direct PLL to FPGA pin route. Maybe the Pocket does. That will be the one to check in the .qsf file

jotego, 2024-04-11:
> Shouldn't be 'ifndef?
> 
> https://github.com/jotego/jtcores/blob/312446f6eb5d48375f0efb5ae4db8542aa7e5923/modules/jtframe/target/mist/jtframe_mist_clocks.v#L86

That would explain a lot! LOL
Yes, that's a bad typo

gyurco, 2024-04-11:
> > My experience if the dedicated PLL outputs are used, there's no need for any phase shift below 100 MHz.
> 
> That would be the best. I think there's a good chance it will work.
> 
It works in 100% on other cores :) (Even on MiST, the dedicated output is also SDRAM_CLK there). STA never fails on SDRAM pins with clock below 100 MHz and no shift (and the input-output registers are in use).

> It was stable because probably the timing was good at that temperature. A 48MHz clock using a small fraction of the FPGA resources should not heat the device much. The problem there is that we do [...]

Maybe. But I updated Outrun to 96MHz, and still worked. Ok, I don't have many samples of the FPGA, only one...
The main problem could be the pins which couldn't be put to the input-output registers, then the signals must travel inside the FPGA for a while, being on the mercy of the placement.

> 
> > Warning to compensation to source clock is not really relevant, e.g. nothing depends on between the phase of the PLL's source clock and output clocks.
> 
> I'd say so too. I may just waive those warnings in jtcore.
> 

CLOCK27 (and CLOCK50) are dedicated input pin to PLL2. But you're cascading, so it'll never be an input to PLL1 or 3, where the first stage will placed, that change the possibilities about source synchronizing. But at the end, it's not needed at all (compensation can be turned off, I think).

> I wonder if we can just swap the output clocks in the original PLL, rather than creating a new one as I did in the branch. I think the other FPGAs using the mist clocks module do not have a direct PLL to FPGA pin route. Maybe the Pocket does. That will be the one to check in the .qsf file

I think it's worth to create new PLL megafunctions, as the originals are for Cyclone III (however I have a feeling that Cyclone 10 LP is mostly a rebranded Cyclone III - the features are too similar).

gyurco, 2024-04-11:
> > Shouldn't be 'ifndef?
> > https://github.com/jotego/jtcores/blob/312446f6eb5d48375f0efb5ae4db8542aa7e5923/modules/jtframe/target/mist/jtframe_mist_clocks.v#L86
> 
> That would explain a lot! LOL Yes, that's a bad typo

Another problem if SDRAM_CLK is defined in sidi128.sdc, then it must come before mist_io.sdc, otherwise it won't be picked up by other lines in mist_io.sdc. So the passing STA is a bit misleading. Instead of success, it's:
Unconstrained Output Ports	73	73

gyurco, 2024-04-11:
Unfortunately exchanging the order in sidi128/target.yaml doesn't work. How to put sidi128.sdc first?

jotego, 2024-04-11:
Making update IPs for Cyclone 10 makes sense indeed.
In the branch I created a separated sdc for MiST so we can split definitions:
https://github.com/jotego/jtcores/blob/312446f6eb5d48375f0efb5ae4db8542aa7e5923/modules/jtframe/target/mist/common.yaml#L18

I think that by using this we should be able to get the port definition well.

We do not have control  over pin registers other than asking for them and hoping for the best. The tool takes the final decision. In large projects, there is a person assigned to physical design that will spend weeks in fine tuning constraints and adding specific placements, etc. We do not have that luxury in our small projects. But at least we can try to get a clean STA report.

gyurco, 2024-04-11:
That'll do the job, thanks!

Usually if it's really a register, then it'll be placed to the fast IO (e.g. register on the pad). The infamous SDRAM_BA=SDRAM_A[12:11] won't work of course. But I think it won't be an issue to generate a clean STA, especially at 48 MHz.

jotego, 2024-04-11:
> The infamous SDRAM_BA=SDRAM_A[12:11] won't work of course. But I think it won't be an issue to generate a clean STA, especially at 48 MHz.

That line should only occur when compiling for MiSTer. Do you see it on sidi128?

jotego, 2024-04-11:
I hear that PLL2 and PLL4 are connected to the input clock and the CLK_SDRAM. We want those to be used for the SDRAM PLL, not for the 27 to 48MHz conversion PLL. I think we need a placement constraint for the gamepll so it uses the right PLL

gyurco, 2024-04-11:
PLL placement already there:
https://github.com/jotego/jtcores/blob/312446f6eb5d48375f0efb5ae4db8542aa7e5923/modules/jtframe/target/sidi128/sidi128.qsf#L331

gyurco, 2024-04-11:
> That'll do the job, thanks!
> 
> Usually if it's really a register, then it'll be placed to the fast IO (e.g. register on the pad). The infamous SDRAM_BA=SDRAM_A[12:11] won't work of course. But I think it won't be an issue to generate a clean STA, especially at 48 MHz.

Not checked actually. I see in most cases, the SDRAM pins are not in registers because they're constant GND or VCC, thus it's not an error.

jotego, 2024-04-11:
> `set_location_assignment PLL_2 -to "u_clocks|u_pll_game|altpll_component|auto_generated|pll1"`

does this incomplete path to pll1 work? I am not sure how the SDC tool is checking this because you are only matching the end of the hierarchy path.

gyurco, 2024-04-11:
You don't need to match more (the individual ports). It's working, if you check the PLL section in the fitter report.

gyurco, 2024-04-11:
There's an "SDC pin name" field there, it's used.

gyurco, 2024-04-11:
Pushed some more changes to the branch. I'm not sure how to do I2S pins, there the clock is I2S_BCK (or HDMI_BCK).

gyurco, 2024-04-11:
No, but they're unconstrained right now. Maybe set_false_path is enough for them.

gyurco, 2024-04-11:
There's no required setup or hold time data for LRCK and SDATA related to BCK in PCM510x datasheet. Also the HDMI chip datasheet doesn't mention anything about audio pins.

jotego, 2024-04-11:
can you share a link to the datasheet?

gyurco, 2024-04-11:
https://github.com/mist-devel/mist-firmware/blob/master/it6613/IT6613-ITETECH.pdf

gyurco, 2024-04-11:
For the PCM chip:
https://www.hestore.hu/prod_getfile.php?id=14183

jotego, 2024-04-11:
Thanks for sharing the data sheets.
Indeed, the PCM chip has a clock and data recovery circuit, so you can set it as a false path. The HDMI chip does not need the clock for the I2S bus, so MCLK can be dropped altogether. Indeed, you have to enable its use if you provide it. I wonder whether it is enabled.
This makes our lives easier.

gyurco, 2024-04-11:
Currently MCLK is set to 0 for both chips (for the PCM chip, it's not even connected to the FPGA), so the internal PLLs are used.

gyurco, 2024-04-12:
I found some Quartus versions are crashing in the PLL dialog. The current one seems to be OK (however it has a problem with editor syntax highlighting in `ifdef...`endif blocks). It's 22.1. Version 23 is a crasher. Looks like you're using 20, I didn't try it.

jotego, 2024-04-12:
I tried the latest one too in linux and windows. Are you saying 22.1 works?
I like retro but not as much as to use Quartus built-in editor :-P
I edit files with Sublime text. Indeed, I rarely boot Quartus.

gyurco, 2024-04-12:
I used to it :)
22.1 works the best so far for me.

skutis, 2024-04-12:
I'm setting up the jtframe environment today as well, 22.1 is the best. I guess 13.1 is not needed for sidi128.

gyurco, 2024-04-12:
13.1 is only for MiST (Cyclone III FPGA).

jotego, 2024-04-13:
Thank you for help Gyurco. This one is finished, @jtmiki.

closed 2024-04-13 by jotego