jotego/jtcores #610: Glitched graphics in X-Men vs Street Fighter

elmisterache7, 2024-04-12:
[image]
[image]
I noticed there is a bar of glitched graphics above the VS RANKING text during the demo gameplay.

jotego, 2024-05-05:
@jtmiki please verify that this is happening in the current version. Compare it with original footage.

black-francis, 2024-05-29:
https://github.com/jotego/jtcores/issues/155

Duplicate of this closed report.

jotego, 2025-12-29:
Doing a long sim, this is visible around frame 13965~13998

[image]

 also around frame 41945

[image]

Probably the row DMA is happening at the wrong time related to the raster interrupt