jotego/jtcores #611: kicker: not booting on sidi128

jotego, 2024-04-12:
There's something going on that prevents it from booting on sidi128, while the same commit works fine on MiSTer/SiDi. Please consider that the problem could also be outside the core (firmware?).

gyurco, 2024-04-12:
Strange, a core from a week ago works, what I compiled yesterday is not (only zeroes on the screen).

jotego, 2024-04-12:
I tried the version just before fixing the STA errors and it did not work either. JTROADF, JTGNG are not working either. Maybe they are all broken?

gyurco, 2024-04-12:
I'll try, the cores I compiled in 31th March are working.

jotego, 2024-04-12:
Try running `jtgood`. This is from the help:

```
Use jtgood <corename>

jtgood is useful when combined with git bisect command
In order to find the commit that broke a core do:

git bisect start
git bisect bad
git bisect good good_rev_id
git bisect run jtgood <corename> [-q]
```

For other platforms, I started keeping daily builds for all cores. I'll share the folder with you. In this case, you cannot use that folder though because it still does not contain sidi128 builds.

gyurco, 2024-04-12:
In the SDRAM controller, there' s this: DST     = READ  + (SHIFTED ? 1 : 2) ,
If I remove the SHIFTED case, e.g. DST = READ + 2, it's working.
Capturing the data right after the read command is registered into the SDRAM needs very tight timing (officially there's no CL1 timing).

gyurco, 2024-04-12:
An analogy popped in my mind: if you fly westwards fast enough, you can arrive sooner than you departed :)

gyurco, 2024-04-12:
https://github.com/jotego/jtcores/blob/f10e310067f691ed0f4bae70cac9273f92976d71/modules/jtframe/hdl/jtframe_board.v#L724
Aren't these cases exchanged? When the SDRAM clock is 180 degree ahead of the FPGA clock, then it's possible that the FPGA can capture the data at the same cycle as the RAM sent it (but it makes the RAM's job harder to catch FPGA's commands and data). That's why SHIFTED(1) probably never worked at 96MHz.

gyurco, 2024-04-12:
A 3.17ns _positive_ shift of the SDRAM_CLK works actually, but it's still a serious abuse of the SDRAM, and I don't think it's guaranteed to work on all boards. True that SHIFTED(0) is introducing an additional one cycle delay, but on SiDi128 (and MiST/SiDi) it would be better to use 96 MHz SDRAM clock then, it's rock stable without any hacks.

jotego, 2024-04-12:
There is nothing _hacky_ about shifting the clock. It just makes it easier for the logic synthesizer to meet timing. Note that the only one who knows the real clock shift is the synthesizer. What you do by adding shift to the PLL is creating a fixed shift that is temperature and device independent whereas the rest of the shift, created by the signal path, will depend on temperature and the specific speed of the actual FPGA. That is all it is: it guarantees an amount of reliable shift. Without it, the synthesizer will try to add the shift using buffers and routing, which is highly variable.

JTFRAME_180SHIFT is deprecated and never used. This comes from a past misunderstanding. I'll delete it.

We should not need to add one clock cycle, just helping the synthesizer with the shift is enough.

If SiDi128 used 96MHz for the SDRAM while the other platforms use 48MHz, it would not serve as a development platform as quirks may arise from that innocent change. All cores in all platforms keep the same IO signals except for the frame buffer used in JTOUTRUN, which is platform-specific.

What is the culprit for the cores not working then?

jotego, 2024-04-12:
Sorry, about JTFRAME_180SHIFT, it is used in JTSF and JTCPSx. It creates a 180º shift at the pad. We used to think that this would mean that the SDRAM_CLK would be 180º, but again, it depends on all the additional shift added by the synthesizer. I'll check if I can delete it in those cores.

gyurco, 2024-04-12:
With the current code, you expect the SDRAM data one cycle sooner than it should. It's only possible because the clock to the SDRAM arrives much sooner, so when it sends the data, the FPGA still can capture it in the same cycle (from the POV of the FPGA).
Let me demonstrate. With CL2, the SDRAM should work as:
cycle 0 - issue the READ command
cycle 1 - the SDRAM gets the READ command
cycle 2 - SDRAM sends the data
cycle 3 - FPGA can read the data

However you expect the data in cycle 2. It's only possible because the rising edge of the clock sensed by the SDRAM arrives somewhere at the middle of the cycles, so when the SDRAM sends back the data (from the POV of the FPGA, it's around at cycle 1.5), the FPGA can still catch it at cycle 2.

Of course it doesn't work at higher clocks, so as I see, at 96 MHz, SHIFTED is always 0, obeying the extra cycle needed to catch the data. BTW, if the SDRAM is used normally, there shouldn't be a need to change where the data is captured, it's well-defined for CL2, no matter if it's 48 or 96 MHz. You can have a little shortcut at 48 MHz, but it's risky.

gyurco, 2024-04-12:
You basically do this now:
cycle 0 - send READ command
cycle 0.5 - SDRAM gets the read command
cycle 1.5 - SDRAM sends the read data
cycle 2 - FPGA captures the data

However it's not really healthy operation. Basically you already doubled the clock.

jotego, 2024-04-12:
Where do you get the 0.5 from?

gyurco, 2024-04-12:
Observation. It's an approximation based on what's happening right now. The original MiST's PLL creates a huge phase shifted SDRAM clock (5 ns = quarter period), while most game cores don't need more than 0.7ns even at 128 MHz!
The phase shift is needed to help the SDRAM to send back its data in the time window which is good for the FPGA (e.g. to meet FPGA's setup timing). But not to send it a full cycle earlier!

I think we can agree that the SDRAM timings (digitally, in cycles) is the same at 48 and 96 MHz (CL2). Yet the data is captured 1 cycle earlier at 48 MHz (DST = READ + 1) than at 96 MHz (DST = READ + 2) right now. Why?

jotego, 2024-04-12:
It is a long time since I wrote that controller but it was well simulated using the full MiST setup in full verilog simulators that represent Z and X correctly. 

I think the misunderstanding comes from taking the phases at face value, pum not intended. Even if you have a bit of phase delay in the PLL, that does not mean that the SDRAM is going to get data from _the future_. Because precisely what the shift is trying to do is to make the SDRAM wait for _the present_ data. The clock phase needs to follow the data phase, and what we do with the PLL delay is just bringing a bit of certainty so the synthesizer can make its job better.

Again, adding shift to the clock does not mean that you get that shift at the pin. The synthesizer will do whatever it takes to match clock and data edges.

Notice that Kicker is reading the correct data, just not all the time. Showing zeros at the beginning is what it should do. But it shows a lot of graphical errors. We are not getting the data reliably. Probably adding a bit of shift will fix it.

If the constraints are well set, ideally, the SDRAM should work correctly if you get a PASS in synthesis. Maybe we are missing a constraint to link the SDRAM_CLK with clk_rom?

jotego, 2024-04-12:
> I think we can agree that the SDRAM timings (digitally, in cycles) is the same at 48 and 96 MHz (CL2). Yet the data is captured 1 cycle earlier at 48 MHz (DST = READ + 1) than at 96 MHz (DST = READ + 2) right now. Why?

No, that is different. The CL is configured differently for 48 and 96MHz.

gyurco, 2024-04-12:
> > I think we can agree that the SDRAM timings (digitally, in cycles) is the same at 48 and 96 MHz (CL2). Yet the data is captured 1 cycle earlier at 48 MHz (DST = READ + 1) than at 96 MHz (DST = READ + 2) right now. Why?
> 
> No, that is different. The CL is configured differently for 48 and 96MHz.

How? There's CL2 or CL3. It's CL2 in every case.

gyurco, 2024-04-12:
Hardcoded to CL2, not even selectable:
https://github.com/jotego/jtcores/blob/d6eb8cec1285d3406f1e1c1cc214634f15ab1e8e/modules/jtframe/hdl/sdram/jtframe_sdram64_init.v#L83

The number of cycles between read cmd and data arrive must be the same. The trick with the huge phase shift makes it 'DDR-like', but not in a healthy way.

jotego, 2024-04-12:
I just checked the code. You're right it is not changing. Let me check the datasheet again.

jotego, 2024-04-12:
I see your point. It is not meant to be a trick to have that effect though. Let's try to make sense on the logic side first:

For 48MHz:

- change CL to 1
- change shift to 0

For 96 MHz:

- keep CL as 2
- keep phase shift as it is

this would apply to all systems, not just sidi128. Do you agree?

gyurco, 2024-04-12:
Except that there's no such thing as CL1 :)
CL2 basically when the SDRAM registers the read command, and issues the data at next cycle (see https://github.com/jotego/jtcores/issues/611#issuecomment-2051814852)

However we can win back one cycle with active to read/write, as I see it's 3/2 now, it can be 2/1 without violating the specs.

jotego, 2024-04-12:
Well CL=1 does not work indeed. Apparently some models, like the one used in the [Analogue Pocket](https://www.alliancememory.com/wp-content/uploads/pdf/mobile_sdram/20180115_AllianceMemory_512M_LPSDRAM_AS4C32M16MSA-6BIN(TR)_rev1.0_Dec2017.pdf), do have CL=1.

> However we can win back one cycle with active to read/write, as I see it's 3/2 now, it can be 2/1 without violating the specs.

Just tried it. It does not work :-(

gyurco, 2024-04-12:
> Well CL=1 does not work indeed. Apparently some models, like the one used in the [Analogue Pocket](https://www.alliancememory.com/wp-content/uploads/pdf/mobile_sdram/20180115_AllianceMemory_512M_LPSDRAM_AS4C32M16MSA-6BIN(TR)_rev1.0_Dec2017.pdf), do have CL=1.
> 
> > However we can win back one cycle with active to read/write, as I see it's 3/2 now, it can be 2/1 without violating the specs.
> 
> Just tried it. It does not work :-(
> 
> ```
> index 1146cc0f..b545fb63 100644
> --- a/modules/jtframe/hdl/sdram/jtframe_sdram64_bank.v
> +++ b/modules/jtframe/hdl/sdram/jtframe_sdram64_bank.v
> @@ -81,7 +81,7 @@ localparam IDLE    = 0,
>             // AUTOPRECH 1+2(1)
>             PRE_ACT = HF ? 3:2,
>             ACT     = PRE_ACT+1,
> -           PRE_RD  = PRE_ACT + (HF ? 3:2),
> +           PRE_RD  = PRE_ACT + (HF ? 2:1),
> ```

For SiDi128? First you have to make sure SHIFTED is 0.

jotego, 2024-04-12:
This works:

```
index 1146cc0f..8d5bb0b0 100644
--- a/modules/jtframe/hdl/sdram/jtframe_sdram64_bank.v
+++ b/modules/jtframe/hdl/sdram/jtframe_sdram64_bank.v
@@ -81,9 +81,9 @@ localparam IDLE    = 0,
            // AUTOPRECH 1+2(1)
            PRE_ACT = HF ? 3:2,
            ACT     = PRE_ACT+1,
-           PRE_RD  = PRE_ACT + (HF ? 3:2),
+           PRE_RD  = PRE_ACT + (HF ? 2:1),
            READ    = PRE_RD+1,
-           DST     = READ + (SHIFTED ? 1 : 2) ,
+           DST     = READ + 2,
```

gyurco, 2024-04-12:
Cool. That way actually we even won one cycle at 96 MHz.

gyurco, 2024-04-12:
Now I guess the PLLs has to be revisited, the phase shift at 48 MHz will be too large for most of them.

jotego, 2024-04-12:
I am going to try with 0 for 48MHz. Then I'll revisit the 96MHz.

jotego, 2024-04-12:
0 works in the Pocket too. That's good. See the _sdram_ branch.

gyurco, 2024-04-12:
> Well CL=1 does not work indeed. Apparently some models, like the one used in the [Analogue Pocket](https://www.alliancememory.com/wp-content/uploads/pdf/mobile_sdram/20180115_AllianceMemory_512M_LPSDRAM_AS4C32M16MSA-6BIN(TR)_rev1.0_Dec2017.pdf), do have CL=1.

WOW, CL1 SDRAM, never seen one until now.

jotego, 2024-04-12:
> I checked the old Micron datasheet, and tPR (Precharge command period) is 20ns (1 cycle@50 MHz), thus
> 
> ```
> PRE_ACT = HF ? 2:1
> ```
> 
> also should be safe.

It works on SiDi128, at least on Kicker.

Sadly, applying these changes to MiSTer breaks it. The SDRAM modules in MiSTer are not very reliable. I might get it working for me and then get reports from 100 users saying it fails for them.

I may have to leave the MiSTer setup as it is. I agree that the changes you have suggested make sense and I would like to have them for the platforms where the SDRAM is soldered to the board.

gyurco, 2024-04-12:
> Sadly, applying these changes to MiSTer breaks it. The SDRAM modules in MiSTer are not very reliable. I might get it working for me and then get reports from 100 users saying it fails for them.

That's sad. Then some parameters should be used for the better behaving boards. Probably there's a specific SDRAM clock which would make it work, but would need a lot of testing.

jotego, 2024-04-13:
Everything seems good in eed27572 (except for a few cores with other STA problems in sidi128). Minimum changes done:

- SDRAM parameter SHIFT always set to 0 in non-mister platforms
- Saved 2 clock cycles in SDRAM pipeline
- SDRAM clock shift removed for non-mister platforms
- Tested all clock variations in Pocket, sidi128 and MiSTer

```
diff --git a/modules/jtframe/hdl/jtframe_board.v b/modules/jtframe/hdl/jtframe_board.v
index 3ab2ca7d..c89c0b35 100644
--- a/modules/jtframe/hdl/jtframe_board.v
+++ b/modules/jtframe/hdl/jtframe_board.v
@@ -717,14 +717,15 @@ jtframe_sdram64 #(
     .PROG_LEN     ( PROG_LEN      ),
     .MISTER       ( MISTER        ),
 `ifdef JTFRAME_SDRAM96
-    .HF(1),
-    .SHIFTED(0)
+    .HF(1)
 `else
-    .HF(0),
+    .HF(0)
+    `ifdef MISTER
     `ifdef JTFRAME_180SHIFT
-        .SHIFTED(0)
+        ,.SHIFTED(0)
     `else
-        .SHIFTED(1)
+        ,.SHIFTED(1)
+    `endif
     `endif
 `endif
 ) u_sdram(
diff --git a/modules/jtframe/hdl/sdram/jtframe_sdram64_bank.v b/modules/jtframe/hdl/sdram/jtframe_sdram64_bank.v
index 1146cc0f..ec232881 100644
--- a/modules/jtframe/hdl/sdram/jtframe_sdram64_bank.v
+++ b/modules/jtframe/hdl/sdram/jtframe_sdram64_bank.v
@@ -79,9 +79,9 @@ localparam ROW=13,
 // states
 localparam IDLE    = 0,
            // AUTOPRECH 1+2(1)
-           PRE_ACT = HF ? 3:2,
+           PRE_ACT = HF ? 2:1,
            ACT     = PRE_ACT+1,
-           PRE_RD  = PRE_ACT + (HF ? 3:2),
+           PRE_RD  = PRE_ACT + (HF ? 2:1),
            READ    = PRE_RD+1,
            DST     = READ + (SHIFTED ? 1 : 2) ,
            DTICKS  = BURSTLEN==64 ? 4 : (BURSTLEN==32?2:1),
diff --git a/modules/jtframe/target/mist/pllgame/jtframe_pllgame.v b/modules/jtframe/target/mist/pllgame/jtframe_pllgame.v
index e3d4eaba..6557b475 100644
--- a/modules/jtframe/target/mist/pllgame/jtframe_pllgame.v
+++ b/modules/jtframe/target/mist/pllgame/jtframe_pllgame.v
@@ -121,7 +121,7 @@ module jtframe_pllgame (
 		altpll_component.clk2_divide_by = 1,
 		altpll_component.clk2_duty_cycle = 50,
 		altpll_component.clk2_multiply_by = 1,
-		altpll_component.clk2_phase_shift = "5000",
+		altpll_component.clk2_phase_shift = "0",
 		altpll_component.clk3_divide_by = 2,
 		altpll_component.clk3_duty_cycle = 50,
 		altpll_component.clk3_multiply_by = 1,
diff --git a/modules/jtframe/target/mist/pllgame96/jtframe_pllgame96.v b/modules/jtframe/target/mist/pllgame96/jtframe_pllgame96.v
index ca866e9f..89b8e273 100644
--- a/modules/jtframe/target/mist/pllgame96/jtframe_pllgame96.v
+++ b/modules/jtframe/target/mist/pllgame96/jtframe_pllgame96.v
@@ -121,7 +121,7 @@ module jtframe_pllgame96 (
 		altpll_component.clk2_divide_by = 1,
 		altpll_component.clk2_duty_cycle = 50,
 		altpll_component.clk2_multiply_by = 2,
-		altpll_component.clk2_phase_shift = "-1000",
+		altpll_component.clk2_phase_shift = "0",
 		altpll_component.clk3_divide_by = 2,
 		altpll_component.clk3_duty_cycle = 50,
 		altpll_component.clk3_multiply_by = 1,
```

closed 2024-04-13 by jotego

jotego, 2024-04-13:
The old sidi does not meet timing with 0º shift for all cores. I am restoring the old scheme for that platform.

gyurco, 2024-04-13:
I think it would be wise to switch to c0 for SiDi and MiST, too to use the dedicated PLL output.

jotego, 2024-04-13:
I did not know sidi and mist also had the PLL correctly hooked. I've tested it but it is not enough to get rid of the errors.

I'll do both then, keep the shift for sidi and use c0 instead of c2. For mist, I still need the shift for 96MHz games, but in that case it is a small negative shift.

gyurco, 2024-04-13:
At 96 MHz, maybe, but a very small amount should be enough. However probably the delay values in the SDC are really really worst case values, and the actual chips used have better performance.
I doubt at 48 MHz they need any shift at all.