jotego/jtcores #614: Support for ROM hacks

jotego, 2024-04-15:
There is a request to incorporate the MRAs from [Arcade Offset](https://github.com/atrac17/Arcade_Offset) to JTCORES. Considerations:

- Patch support in TOML files only takes in-line data
- Patches are applied to the main MRA files that provide compatibility with original arcade games without altering them substantially
- All MRA files in JTBIN are derived from mame.xml and the core TOML file by using the command `jtframe mra`

Incorporating these files would require:

- Verifying that the ROM hack does not contain ROM chunks from the original developer, to prevent copyright infringements
- Deriving separate MRA files for patched MRAs
- Taking patch binary data from an external file (jtframe can then convert it to text to embed it into the MRA)
- Clarifying the author of each ROM hack
- Identifying, maybe via a different OSD color, that the system running is an alteration of third party software
- Adding URL links to romhacking.net for referencing the hack source
- Hack binary files won't be stored in jtcores

jotego, 2024-04-16:
I think an easier solution to these problems would be to add IPS file format support to MiSTer directly. That way the MRA file can refer to the desired IPS patch in the attributes of the `<rom>` element. MiSTer can find the patch in a local folder and apply it to the original files before building the data stream for the core.

MRA file generation would only need to be modified to include that IPS file reference in a new MRA with a different name from the original.

This solves the copyright concerns, simplifies MRA generation and enables users to apply IPS patches easily.

What do you think, @deepthaw?

jotego, 2024-04-17:
> I think an appropriate first step would be to submit the IPS patching code to mister-dev (I think it'd go there. I'm still learning all the codebases) and also add in the ability for MRAs to reference it there. Once it's available in that manner, we can see how people feel about the best way to expose it within console cores.

Yes, let's do this first. It does not imply any change to the OSD menu if you only think of arcade patches. Everything would be in the MRA.

This will also imply the existance of a standard folder in MiSTer for patches.

> Would we still want to generate pre-patched MRAs that have popular, non-controversial hacks as part of the release process? We'd probably also want to generate the other MRAs so that end users don't have to do it themselves, but somehow indicate that they need to supply the IPS patches themselves - this would be a change to current behavior and may confuse users.

I think that having the patch data in the MRA itself is too risky legally. Supporting IPS patches in MiSTer should be fine and it translates the judgement on whether a file can be used or not to the final user, who can make a choice a one by one basics.

I appreciate your interest and I think IPS support can be a good addition to MiSTer. I do not see immediate actions for this in JTCORES, though. If you are fine with it, maybe we can close this issue for now and look at it again when MiSTer firmware is ready.

asturur, 2024-04-17:
Could the cheat code logic offer some leverage?
Cheatcodes as memory patches while applied in a different moment and at runtime share some similarities with what described.

- they are in a special folder per system
- they can be matched to the game that is loaded by something ( not sure if is rom checksum or what else )
- they do not clutter your view until you go in the cheat menu

Something to take in account anyway when thinking of a patching system outside arcade is that an applied patch should invalidate ( for safety ) the cheat for the unpatched rom.

deepthaw, 2024-04-17:
@jotego sounds good to me. I'll start working on something that may have a better chance of getting accepted than previous attempts at IPS patches and see how it goes. If nothing else, I'd like to add the ability for relocation of binary data and RLE so we can reduce the size of some of the MRAs and possibly reduce the chance of copyrighted data being in it. Go ahead and close this and we'll circle back in a while. I'll continue making contributions to Arcade_Offsets with notes that those aren't affiliated with you or your projects in any way.

@asturur That's definitely an approach I would want to take with any MiSTer-wide application. There are differences such that I imagine you'd want the patch applied before the core tries to boot the game, or at the least force a reset once it's been applied - then what happens to the SRAM if you apply a patch and reset? etc. And those seem like they could introduce unintended behaviors or at least be something to look out for.

I also think the cheat/game matching is being done by the people who assembled the cheat collections - IPS files (the most common, to the best of my knowledge) have zero info about the game they're patching. They're literally a series of offsets and data and nothing else. I was talking on the MiSTer discord and the possibility of tying IPS patches to MGLs also came up, that could be an approach that doesn't require fiddling within the OSD/Menus and would let us patch the ROM before it launches - avoiding some of the runtime weirdness.

jotego, 2024-04-17:
Perfect. We'll leave it here for now. Thank you.

closed 2024-04-17 by jotego