jotego/jtcores #615: ngp: support for color model

jotego, 2024-04-15:


gyurco, 2024-04-30:
I can work on it in May.
First step - able to compile is solved.
Second step - no video on VGA (or HDMI) - the screen resolution is too low. Probably adding a scan-quadrupler (instead of a scandoubler only), like in Gameboy can solve it. Otherwise it's hard to make video related changes without seeing the actual video :)

jotego, 2024-04-30:
It is a pity you need to spend time on doing the screen scaler. Maybe it is worth getting a MiSTer or a Pocket after all as the core works in these systems without any special consideration.

The screen resolution is very odd and it does not match the expected timing for a CRT at all. Making it visible on CRT/VGA is not a trivial task.

jotego, 2024-05-03:
Maybe in the future, other systems will gain more users, but for now I do not see value on supporting anything other than MiSTer or Pocket for this core. So I wouldn't spend time on trying to fix the video for SiDi128.

I have some spare MiSTer memories I can send if you get one of the nano clones. Maybe the developer can send you one of the prototypes.

asturur, 2024-05-03:
if you live in europe i can send you a spare analogue board and some spare memories i think also a spare hub.

asturur, 2024-05-05:
@gyurco send me an email to andreabogazzi79@gmail.com, i prefer to ship you things now that i m out of italy and post work better for sure. Those things come with no strings attached and are taking space in a drawer in a static bag. They come from aliexpress are not the best version but they worked ok and i didn't trash just because i really wanted to recycle them in a useful way

gyurco, 2024-05-05:
@asturur  Thanks, when I get a board, I'll send you a mail.

jotego, 2024-08-15:
I implemented support for color on ngp as a new **ngpc** core. That closes this issue.
Currently, there is a boot up problem (#769) to debug.

closed 2024-08-15 by jotego