jotego/jtcores #616: shouse: support for dragon spirit

jotego, 2024-04-15:


jotego, 2024-04-15:
Flip is implemented, maybe not completely but at least to a good extend. Apparently the flip is written to the object MMR (see the mmr.yaml file). Information about this in MAME is not consistent across games. They think that a certain bit there is for flip but then they do per-game hacks to either flip horizontally or vertically too.

The scroll going the other way around here, whereas in Pac-Mania goes correctly, is because of that. There are comments in MAME referring to this but their implementation is not good. They also consider the MMR to be shorter than it actually is. So I think there might be specific bits for scroll H flip and scroll V flip, rather than a single scroll (screen) flip bit.

You may already know, but just in case. To run a scene in simulation, do:

```
swcore shouse ver/dspirit
jtsim -setname dspirit  # just run it once to get the sdram files generated
sim.sh -s 1
```

That will simulate scene 1. Wherever you see a sim.sh in recent cores, it means there is something extra needed on top of jtsim. In this case, it needs to copy a file to a place where it will get loaded as BRAM.

gyurco, 2024-04-15:
It's the flip flag comes from the sprite MMR. In shouse_scr for example, it does nothing (only a commented out block refers to that). I'm almost successful with it, gives good results.

gyurco, 2024-04-15:
BTW, I think either blanks or tilemap offsets are off-by-one, look at Splatterhouse, there's an extra line at the bottom and during the game, an extra line at the right edge.

jotego, 2024-04-15:
Yes, the scroll offset might off by one pixel, at least horizontally as visible in scene 1 for splatter (ver/game folder)

[image]

The tricky thing here is that whatever change is done, it must not break the other three working games: splatter, pacmania, galaga88.

gyurco, 2024-04-15:
Uploaded the scroll layer change, the obj one looks like more tricky, as large sprites are very broken.

gyurco, 2024-04-16:
- I created a DSPIRIT.RAM file with 'Save NVRAM' (during paused core)
- copied to shouse/ver/dspirit/scenes/1/dump.bin
- jtsim -setname dspirit
- ./sim.sh -s 1

But the frame_0001.jpg doesn't really look like as the one in the screen. Actually totally different. What am I done wrong?

jotego, 2024-04-16:
That actually looks correct. Try running one of the scenes in the other folders just to check that steps 3, 4 work for you

jotego, 2024-04-16:
[This](https://wiki.mate-desktop.org/mate-desktop/applications/eom/) is eom. It displays the image after the sim. I use Ubuntu Mate, so that's the default image viewer. I think you can install it in any ubuntu flavour.

gyurco, 2024-04-16:
Ah, ok. But then the issue is not the missing eom.

jotego, 2024-04-16:
I guess you already went through the jtsim on screen log carefully. The dump.bin is broken into pieces for the different BRAM blocks. You should see messages on screen reporting that each video RAM loaded some default data from disk. You even get a warning if the loaded data do not fill the RAM completely.

Sometimes going back to a simulation after months may find it broken, but in this case I just used it yesterday and it was fine.

gyurco, 2024-04-16:
jtsim:
```
[cores/shouse/ver/pacmania (shouse-flip)]$ jtsim -setname pacmania
=========== JTSIM starts for mist ===========
Using shouse as the system name
Pac-Mania.mra
Pac-Mania.mra
ROM load through SPI enabled
Verilator sim starts
Multibank SDRAM enabled
Read 6885384 bytes from rom.bin
Simulation clock period set to 20344 ps (49.154542 MHz)
DIP sw set to FF
WARNING: TOP.game_test.u_game.u_brom_mask.genblk1.u_ram cannot open file: mask.bin
Blank tmap_lo.bin created
Blank tmap_hi.bin created
Blank oram_lo.bin created
Blank oram_hi.bin created
Blank red.bin created
Blank green.bin created
Blank blue.bin created
WARNING: TOP.game_test.u_game.u_bram_eerom cannot open file: eerom.bin
jtframe_pxlcen: using 4 as clock divider
SDRAM burst mode changed to 2 mask 0xFFFFFFFE -> burst per bank = { 2, 2, 2, 2 }
0123456789ABCDEF0123456789ABCDEF0123456789ABCDEF0123456789ABCDEF -   64
0123456789ABCDEF0123456789ABCDEF0123456789ABCDEF0123456789ABCDEF -  128
01234567
ROM file transfered (frame 136)
        sdram_bank0.bin dumped
        sdram_bank1.bin dumped
        sdram_bank2.bin dumped
        sdram_bank3.bin dumped
KEY: 0 => 12
KEY: 0 <- 02
KEY: 1 <- 14
KEY: 2 <- 2c
KEY: 3 <- 78
KEY: 0 => 00
KEY: 0 => d4
KEY: 0 => 00
KEY: 0 => 15
KEY: 0 <- 01
KEY: 1 <- 37
KEY: 2 <- b5
KEY: 3 <- b9
KEY: 0 => 00
KEY: 0 => b6
KEY: 0 => 00
KEY: 0 => 95
KEY: 0 <- 00
KEY: 1 <- 00
KEY: 2 <- 00
KEY: 3 <- 00
KEY: 0 => 00
KEY: 0 => 00
KEY: 0 => ff
KEY: 0 => ff
KEY: 0 <- ff
KEY: 1 <- ff
KEY: 2 <- 00
KEY: 3 <- 01
KEY: 0 => 00
KEY: 0 => 01
KEY: 0 => 00
KEY: 0 => 00
KEY: 0 <- 7d
KEY: 1 <- 6f
KEY: 2 <- cc
KEY: 3 <- f5
KEY: 0 => 4f
KEY: 0 => 86
KEY: 0 => 00
KEY: 0 => 01
KEY: 0 <- 00
KEY: 1 <- 01
KEY: 2 <- f2
KEY: 3 <- c9
KEY: 0 => 00
KEY: 0 => 00
KEY: 0 => f2
KEY: 0 => c9
KEY: 0 <- 00
KEY: 1 <- 02
KEY: 2 <- 08
KEY: 3 <- 67
KEY: 0 => 00
KEY: 0 => 01
KEY: 0 => 04
KEY: 0 => 33
KEY: 0 <- 00
KEY: 1 <- 04
KEY: 2 <- 22
KEY: 3 <- b8
KEY: 0 => 00
KEY: 0 => 00
KEY: 0 => 08
KEY: 0 => ae
KEY: 0 <- 00
KEY: 1 <- 2d
KEY: 2 <- 4e
KEY: 3 <- 21
KEY: 0 => 00
KEY: 0 => 15
KEY: 0 => 01
KEY: 0 => bc
KEY: 0 <- 00
KEY: 1 <- 6a
KEY: 2 <- 4c
KEY: 3 <- e4
KEY: 0 => 00
KEY: 0 => 4a
KEY: 0 => 00
KEY: 0 => b9

====== JTSIM finished in 2'16" ======
```

sim.sh:
```
[cores/shouse/ver/pacmania (shouse-flip)]$ ./sim.sh -s 1
256+0 records in
256+0 records out
131072 bytes (131 kB, 128 KiB) copied, 0.000493236 s, 266 MB/s
=========== JTSIM starts for mist ===========
Using shouse as the system name
Pac-Mania.mra
Using DIP switches from rom/pacmania.dip
Verilator sim starts
Multibank SDRAM enabled
Read 800000 from sdram_bank0.bin
Read 800000 from sdram_bank1.bin
Read 800000 from sdram_bank2.bin
Read 800000 from sdram_bank3.bin
Read 6885384 bytes from rom.bin
Simulation clock period set to 20344 ps (49.154542 MHz)
Verilator will dump to test.vcd
DIP sw set to FF
MMR TOP.game_test.u_game.u_game.u_video.u_scroll.u_mmr - read 32 bytes from offset 0
        hscr = 0000000000000000
        vscr = 0000000000000000
        enb = 00
        prio = 00000
        pal = 00000
MMR TOP.game_test.u_game.u_game.u_video.u_obj.u_mmr - read 16 bytes from offset 32
        xoffset = 000
        flip = 0
        yoffset = 00
MMR TOP.game_test.u_game.u_game.u_video.u_colmix.u_mmr - read 16 bytes from offset 48
        left = 000
        right = 000
        top = 000
        bottom = 000
        hirq = 0000
        virq = 0000
INFO: PROM download: 691008 bytes loaded from file (rom.bin)
INFO: fast PROM download from 00690008 to 00691008
INFO: Read       mask.bin (131072 bytes) for TOP.game_test.u_game.u_brom_mask.genblk1.u_ram
-tmap_lo.bin (   0 bytes) TOP.game_test.u_game.u_bram_tmap.u_lo.u_ram
-tmap_hi.bin (   0 bytes) TOP.game_test.u_game.u_bram_tmap.u_hi.u_ram
-oram_lo.bin (   0 bytes) TOP.game_test.u_game.u_bram_oram.u_lo.u_ram
-oram_hi.bin (   0 bytes) TOP.game_test.u_game.u_bram_oram.u_hi.u_ram
-red.bin    (8192 bytes) TOP.game_test.u_game.u_bram_red.u_ram
-green.bin  (8192 bytes) TOP.game_test.u_game.u_bram_green.u_ram
-blue.bin   (8192 bytes) TOP.game_test.u_game.u_bram_blue.u_ram
INFO: Read      eerom.bin (2048 bytes) for TOP.game_test.u_game.u_bram_eerom
jtframe_pxlcen: using 4 as clock divider
SDRAM burst mode changed to 2 mask 0xFFFFFFFE -> burst per bank = { 2, 2, 2, 2 }
ROM download shortened to 32 bytes.

ROM file transfered (frame 0)
0123
====== JTSIM finished in 0'7" ======
```

I don't see any failure.

jotego, 2024-04-16:
Exactly. The sim.sh is the one that generates the mask.bin file. I can see it created in the log from sim.sh. But then the tile maps and object RAM are zero-length files from what I see. Those files are created by the dump2bin.sh in the ver/game folder, which is generated automatically when you run a sim.

I cannot test it now. But the splatter house image in this thread is scene 1 from the game folder, if I remember correctly. It was working yesterday.

gyurco, 2024-04-16:
There's a dump2bin.sh which was never run.

jotego, 2024-04-16:
jtsim runs it

jotego, 2024-04-16:
`jtsim -setname pacmania; sim.sh -s 1` in commit 8afcc921

produces this:

[image]

I cannot figure out what might be going wrong in your end.

By the way, the same way you can use keyboard's F7-F10 to disable layers in debug builds, you can pass the **JTFRAME_SIM_GFXEN** macro to the simulation for the same effect. This will only show the objects: `sim.sh -s 1 -d JTFRAME_SIM_GFXEN=8`

Note that the simulation will also rotate the image for you in the right direction.

[image]

gyurco, 2024-04-16:
```
../game/dump2bin.sh scenes/1/dump.bin
8+0 records in
8+0 records out
2048 bytes (2.0 kB, 2.0 KiB) copied, 9.1492e-05 s, 22.4 MB/s
128+0 records in
128+0 records out
32768 bytes (33 kB, 32 KiB) copied, 0.000256882 s, 128 MB/s
../game/dump2bin.sh: line 13: drop1: command not found
../game/dump2bin.sh: line 14: drop1: command not found
32+0 records in
32+0 records out
8192 bytes (8.2 kB, 8.0 KiB) copied, 0.000113302 s, 72.3 MB/s
32+0 records in
32+0 records out
8192 bytes (8.2 kB, 8.0 KiB) copied, 9.4688e-05 s, 86.5 MB/s
32+0 records in
32+0 records out
8192 bytes (8.2 kB, 8.0 KiB) copied, 0.000114375 s, 71.6 MB/s
16+0 records in
16+0 records out
4096 bytes (4.1 kB, 4.0 KiB) copied, 9.2934e-05 s, 44.1 MB/s
../game/dump2bin.sh: line 27: drop1: command not found
../game/dump2bin.sh: line 28: drop1: command not found
0+1 records in
0+1 records out
64 bytes copied, 5.4573e-05 s, 1.2 MB/s
```

Ok, seems another command I never heard about :)
What's this drop1? Googling it resulted in a function in R, but I hope it's not needed.

BTW, wouldn't be useful to write the shebang as #!/bin/bash -e, to exit on error?

gyurco, 2024-04-16:
I see now what drop1 wants to do - separate the .bin to lo and hi bytes. Maybe is it an utility you wrote and forgot to commit?
(Another missing script is jtmacros.awk)

jotego, 2024-04-17:
> Ok, seems another command I never heard about :) What's this drop1? Googling it resulted in a function in R, but I hope it's not needed.

This is an utility I wrote to drop every second byte of a file. It converts 16-bit memory dumps to two 8-bit files (in two calls). I have included it as a `jtutil` command in e7f196e0 and updated all references to it. Like `jtframe`, the first time you call `jtutil` it will get build for you.

Sorry I didn't catch this. I used to have all the utilities outside of the repository. I started adding them a while ago in an organized manner as `jtutil` commands a while ago. This one had fallen through the cracks.

> BTW, wouldn't be useful to write the shebang as #!/bin/bash -e, to exit on error?

Yes, I didn't know I could have that in the shebang. Some scripts have a `set -e` but having it at the top is more compact.

gyurco, 2024-04-17:
Thanks, it works now!

Just a final question: how to do a .vcd dump with sim.sh?

gyurco, 2024-04-17:
About the bottom line garbage: I don't understand why the VBL is delayed by 2 lines, e.g. after the overflow (511->248), even 249 line is visible. That means the last line to the line buffer is rendered when vrender=248 -> overflown already, so the garbage. The last line should be rendered when vrender=511, I think.

gyurco, 2024-04-17:
And I think lin_row also should be reset to 1 one line before vblank de-asserted. Now it's reset to 1 together with vblank disabled, so the first row is empty.

jotego, 2024-04-17:
> Just a final question: how to do a .vcd dump with sim.sh?

I guess you do not care about the file format of the dump, do you? Scene simulations will generate a .fst dump that you can view with **gtkwave**. In general, `jtsim -w` will create dumps. You can use `fst2vcd`, which comes with gtkwave to produce a vcd after the sim.

The .fst file is created during the simulation by transforming the .vcd that verilator generates through an Unix pipe. FST files are smaller and faster but are a **gtkwave/iverilog** format. Are you planning of using a different waveform viewer?

> About the bottom line garbage: I don't understand why the VBL is delayed by 2 lines, e.g. after the overflow (511->248), even 249 line is visible. That means the last line to the line buffer is [...]

I would need to look at the counters around VBL. But, yes, in general, the H/V transitions are tricky. An additional complication is that I often disable video processing during blanking to ease SDRAM traffic. Also, depending on the game, the vertical counter input to video modules may be one or two lines ahead of the visible line to allow for line buffers. Some games do this in software (by placing sprites in a different line) and some do it in hardware (by inputting shifted counters to the line comparator)

jotego, 2024-04-17:
> And I think lin_row also should be reset to 1 one line before vblank de-asserted. Now it's reset to 1 together with vblank disabled, so the first row is empty.

where is lin_row?

gyurco, 2024-04-17:
gtkwave is OK for me (not a perfect viewer, but free at least).

lin_row is in jtframe_scr. It counts fixed layer position. The problem with rendering at wrap-around is not at the sprite layer, but in the scroll. You can see even in the splatterhouse frame that the first line is empty (and at the starting logo, there's a garbage line at the bottom).

jotego, 2024-04-17:
> Probably moving everything one line up would solve this (reset lin_row one line sooner for the fix layer, and adjust voffset by -1 for the scroll ones). Then adjust sprites, too.

Sounds good.

gyurco, 2024-04-17:
Hmm, the top line emptiness is because the colmix_mmr's top value doesn't agree with vbl. I wonder if in the real HW, the mmr sets up VBL, and no fixed blanking.

jotego, 2024-04-17:
I think so. The JAMMA connector does not output blanking signals, so they are used on the board to gate the color signals mainly. colmix applies the blanking in our case but it does not alter the signals depending on the configuration.

The core also has to output stable blanking signals because Pocket and MiSTer require them to scale the video. If we were to vary dynamically the length of LVBL it would likely have funny effects on those platforms.

By the way, the LVBL/LHBL names are inherited from old CAPCOM schematics. The first L stands for active low.

gyurco, 2024-04-17:
Probably they're not messing around too much with those register, so doesn't matter if the fixed blanks are used.

Finished the scroll layers, fighting with sprites.

gyurco, 2024-04-17:
Uploaded the latest changes, everything works well - almost.
The dragon's head is flashing when you move it upwards. Must be a very stupid reason for it.

jotego, 2024-04-17:
I am testing the changes now. It is quite impressive how a few edits can unleash so many games. That is the shocking nature of digital design. Thank you very much for spending the time on it.

One game that still looks broken is **blastoff** although it might be for completely different reasons. I would look in the _colmix_ first for that one, priorities may play a role too.

The namco zoom in animation in **dangseed* also breaks around the corners in a funny way. I wonder how they are generating that effect. By the way, when the game is paused, **pressing the 9 key will advance just by one frame**. In some games it will also insert a credit as it is the _service_ button, though...

Overall it looks amazing. Please give it that final push to the dragon head!

gyurco, 2024-04-17:
I'm scratching my head on that head...
As I see, one tile is 32x16 in the ROM, and 'vos' selects which half of that is displayed when the requested tile is 16x16. The problem is that moving up displays the wrong half. No idea why, as ypos [4:3] == '11', thus it should be the right half?
The tile number is 38e, you can see the right part is required, but the left part is displayed.

gyurco, 2024-04-17:
I think I got it now!
It's a bit disappointing to see the patch is only 196 lines long after 1 day of debugging.

jotego, 2024-04-18:
I have pushed to this branch a commit with two scenes for **blastoff** and **youkaidk1**. They may be affected by the same or very similar bug:

**blastoff**
[image]

**youkaidk1**
[image]

Please try to fix them too

gyurco, 2024-04-18:
Strange, Shadowland works actually. Maybe they're using a Japanese color chip :)

I'll look at it later on, maybe the fix is only 2 lines, but need half a day to figure out what two lines...

jotego, 2024-04-18:
> I'll look at it later on, maybe the fix is only 2 lines, but need half a day to figure out what two lines...

I think that will be the case. It might even be one line in this case.

gyurco, 2024-04-18:
In the case of Blast Off, scr_pix input to colmix is either 0 or 1ff, thus it doesn't look like palette error. Something fetched wrongly from tile ROMs.

gyurco, 2024-04-18:
It's not a core issue, but MRA issue. The tilemap ROMs in MAME:
```
<------>ROM_REGION( 0x100000, "c123tmap", 0 ) /* characters */
<------>ROM_LOAD( "bo_chr-0.bin",       0x00000, 0x20000, CRC(bdc0afb5) SHA1(09d01640bd0ff8d9d96282411ec9176e5fc57cae) )
<------>ROM_LOAD( "bo_chr-1.bin",       0x20000, 0x20000, CRC(963d2639) SHA1(1bb936631c17e7695e087c95a72f1871fabb6647) )
<------>ROM_LOAD( "bo_chr-2.bin",       0x40000, 0x20000, CRC(acdb6894) SHA1(a0d712e4883cda7068aa678d698abbe9c910363b) )
<------>ROM_LOAD( "bo_chr-3.bin",       0x60000, 0x20000, CRC(214ec47f) SHA1(0f314740bdb2f3cd0ed8ab9a98a47eb8f8b1480b) )
<------>ROM_LOAD( "bo_chr-4.bin",       0x80000, 0x20000, CRC(08397583) SHA1(e2c0ce46982a1f9035f0a3c4f08dbf3a5e11a871) )
<------>ROM_LOAD( "bo_chr-5.bin",       0xa0000, 0x20000, CRC(20402429) SHA1(a84760f693ec15f8f8238653d1761a34e2499da9) )
<------>ROM_LOAD( "bo_chr-7.bin",       0xe0000, 0x20000, CRC(4c5c4603) SHA1(aa0c4f651ae38513f0a2b685dcc62c1897d0f310) )
```

Not the hole between -5.bin and -7.bin

The MRA however doesn't have a hole here, but added a padding 0 after -7.bin. Exchange the padding and -7, then voila'!

gyurco, 2024-04-18:
I considered the TOML method as manual :)

jotego, 2024-04-18:
@jtmiki this issue is completed

PR #623 (2024-04-18): Shouse flip
Fix #616
closed 2024-04-18 by jotego