jotego/jtcores #619: jtframe mra: keep author name as specified in TOML

jotego, 2024-04-16:
Could it be possible to add support for overwriting the author name, website, twitter (the about xml fields) from  mame2mra.toml fields ? (like it's done with 'jtframe mra --author) so when you generate the release files those information are not lost and author are credited ?

_Originally posted by @vertrex in https://github.com/jotego/jtcores/issues/618#issuecomment-2058658273_

jotego, 2024-04-17:
@vertrex 

I have implemented some changes to `jtframe mra`. This is how it looks now:

``` XML
<misterromdescription>
    <about author="Solal Jacob" 
           webpage="https://ko-fi.com/solalretrogaming" 
           twitter="@s_on_vg" 
           source="https://github.com/jotego/jtcores"/>
    <name>Toki (World, set 1)</name>
    <setname>toki</setname>
```

Let me know if you need anything else.

closed 2024-04-17 by jotego

vertrex, 2024-04-17:
Perfect thank you !