jotego/jtcores #620: recursive clone of repository fails

hansfbaier, 2024-04-16:
That is what I get when cloning this with `--recursive`:
```
$ git clone --recursive https://github.com/jotego/jtcores.git
Cloning into 'jtcores'...
remote: Enumerating objects: 54112, done.
remote: Counting objects: 100% (2210/2210), done.
remote: Compressing objects: 100% (719/719), done.
remote: Total 54112 (delta 1405), reused 2162 (delta 1391), pack-reused 51902
Receiving objects: 100% (54112/54112), 264.86 MiB | 5.17 MiB/s, done.
Resolving deltas: 100% (37897/37897), done.
Submodule 'modules/fx68k' (https://github.com/jtfpga/fx68k) registered for path 'modules/fx68k'
Submodule 'modules/j12' (https://github.com/jotego/jt12) registered for path 'modules/jt12'
Submodule 'modules/jt49' (https://github.com/jotego/jt49.git) registered for path 'modules/jt49'
Submodule 'modules/jt51' (https://github.com/jotego/jt51) registered for path 'modules/jt51'
Submodule 'modules/jt5205' (https://github.com/jotego/jt5205) registered for path 'modules/jt5205'
Submodule 'modules/jt6295' (https://github.com/jotego/jt6295.git) registered for path 'modules/jt6295'
Submodule 'modules/jt7759' (https://github.com/jotego/jt7759.git) registered for path 'modules/jt7759'
Submodule 'modules/jt8255' (https://www.github.com/jotego/jt8255) registered for path 'modules/jt8255'
Submodule 'modules/jt89' (https://github.com/jotego/jt89.git) registered for path 'modules/jt89'
Submodule 'modules/jt900h' (git@github.com:jotego/jt900h.git) registered for path 'modules/jt900h'
Submodule 'modules/jtdsp16' (https://github.com/jotego/jtdsp16.git) registered for path 'modules/jtdsp16'
Submodule 'modules/jteeprom' (https://github.com/jotego/jteeprom.git) registered for path 'modules/jteeprom'
Submodule 'modules/jtframe/target/pocket' (git@github.com:jotego/pocket.git) registered for path 'modules/jtframe/target/pocket'
Submodule 'modules/jtkcpu' (https://github.com/JTFPGA/jtkcpu.git) registered for path 'modules/jtkcpu'
Submodule 'modules/jtopl' (https://github.com/jotego/jtopl.git) registered for path 'modules/jtopl'
Submodule 'modules/jttms' (https://github.com/jotego/jttms.git) registered for path 'modules/jttms'
Cloning into '/MiSTeX/jtcores/modules/fx68k'...
remote: Enumerating objects: 440, done.        
remote: Counting objects: 100% (178/178), done.        
remote: Compressing objects: 100% (16/16), done.        
remote: Total 440 (delta 168), reused 171 (delta 162), pack-reused 262        
Receiving objects: 100% (440/440), 166.69 KiB | 699.00 KiB/s, done.
Resolving deltas: 100% (281/281), done.
Cloning into '/MiSTeX/jtcores/modules/jt12'...
remote: Enumerating objects: 4429, done.        
remote: Counting objects: 100% (708/708), done.        
remote: Compressing objects: 100% (98/98), done.        
remote: Total 4429 (delta 646), reused 645 (delta 610), pack-reused 3721        
Receiving objects: 100% (4429/4429), 116.44 MiB | 5.20 MiB/s, done.
Resolving deltas: 100% (3169/3169), done.
Cloning into '/MiSTeX/jtcores/modules/jt49'...
remote: Enumerating objects: 538, done.        
remote: Counting objects: 100% (220/220), done.        
remote: Compressing objects: 100% (155/155), done.        
remote: Total 538 (delta 103), reused 161 (delta 48), pack-reused 318        
Receiving objects: 100% (538/538), 2.58 MiB | 4.57 MiB/s, done.
Resolving deltas: 100% (300/300), done.
Cloning into '/MiSTeX/jtcores/modules/jt51'...
remote: Enumerating objects: 1315, done.        
remote: Counting objects: 100% (39/39), done.        
remote: Compressing objects: 100% (27/27), done.        
remote: Total 1315 (delta 21), reused 19 (delta 10), pack-reused 1276        
Receiving objects: 100% (1315/1315), 2.22 MiB | 4.27 MiB/s, done.
Resolving deltas: 100% (834/834), done.
Cloning into '/MiSTeX/jtcores/modules/jt5205'...
remote: Enumerating objects: 258, done.        
remote: Counting objects: 100% (17/17), done.        
remote: Compressing objects: 100% (12/12), done.        
remote: Total 258 (delta 6), reused 12 (delta 4), pack-reused 241        
Receiving objects: 100% (258/258), 457.15 KiB | 2.75 MiB/s, done.
Resolving deltas: 100% (132/132), done.
Cloning into '/MiSTeX/jtcores/modules/jt6295'...
remote: Enumerating objects: 431, done.        
remote: Counting objects: 100% (113/113), done.        
remote: Compressing objects: 100% (81/81), done.        
remote: Total 431 (delta 49), reused 88 (delta 31), pack-reused 318        
Receiving objects: 100% (431/431), 380.62 KiB | 2.95 MiB/s, done.
Resolving deltas: 100% (254/254), done.
Cloning into '/MiSTeX/jtcores/modules/jt7759'...
remote: Enumerating objects: 354, done.        
remote: Counting objects: 100% (65/65), done.        
remote: Compressing objects: 100% (40/40), done.        
remote: Total 354 (delta 36), reused 48 (delta 24), pack-reused 289        
Receiving objects: 100% (354/354), 917.50 KiB | 3.23 MiB/s, done.
Resolving deltas: 100% (212/212), done.
Cloning into '/MiSTeX/jtcores/modules/jt8255'...
warning: redirecting to https://github.com/jotego/jt8255.git/
remote: Enumerating objects: 78, done.        
remote: Counting objects: 100% (78/78), done.        
remote: Compressing objects: 100% (51/51), done.        
remote: Total 78 (delta 27), reused 69 (delta 18), pack-reused 0        
Receiving objects: 100% (78/78), 816.67 KiB | 3.14 MiB/s, done.
Resolving deltas: 100% (27/27), done.
Cloning into '/MiSTeX/jtcores/modules/jt89'...
remote: Enumerating objects: 379, done.        
remote: Counting objects: 100% (63/63), done.        
remote: Compressing objects: 100% (43/43), done.        
remote: Total 379 (delta 30), reused 46 (delta 16), pack-reused 316        
Receiving objects: 100% (379/379), 668.38 KiB | 1.34 MiB/s, done.
Resolving deltas: 100% (203/203), done.
Cloning into '/MiSTeX/jtcores/modules/jt900h'...
ERROR: Repository not found.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:jotego/jt900h.git' into submodule path '/MiSTeX/jtcores/modules/jt900h' failed
Failed to clone 'modules/jt900h'. Retry scheduled
Cloning into '/MiSTeX/jtcores/modules/jtdsp16'...
remote: Enumerating objects: 1628, done.        
remote: Counting objects: 100% (14/14), done.        
remote: Compressing objects: 100% (9/9), done.        
remote: Total 1628 (delta 4), reused 10 (delta 4), pack-reused 1614        
Receiving objects: 100% (1628/1628), 22.79 MiB | 4.88 MiB/s, done.
Resolving deltas: 100% (1084/1084), done.
Cloning into '/MiSTeX/jtcores/modules/jteeprom'...
remote: Enumerating objects: 180, done.        
remote: Counting objects: 100% (180/180), done.        
remote: Compressing objects: 100% (104/104), done.        
remote: Total 180 (delta 83), reused 141 (delta 48), pack-reused 0        
Receiving objects: 100% (180/180), 1.41 MiB | 4.49 MiB/s, done.
Resolving deltas: 100% (83/83), done.
Cloning into '/MiSTeX/jtcores/modules/jtframe/target/pocket'...
ERROR: Repository not found.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:jotego/pocket.git' into submodule path '/MiSTeX/jtcores/modules/jtframe/target/pocket' failed
Failed to clone 'modules/jtframe/target/pocket'. Retry scheduled
Cloning into '/MiSTeX/jtcores/modules/jtkcpu'...
remote: Enumerating objects: 2393, done.        
remote: Counting objects: 100% (795/795), done.        
remote: Compressing objects: 100% (311/311), done.        
remote: Total 2393 (delta 602), reused 673 (delta 484), pack-reused 1598        
Receiving objects: 100% (2393/2393), 1.97 MiB | 4.72 MiB/s, done.
Resolving deltas: 100% (1797/1797), done.
Cloning into '/MiSTeX/jtcores/modules/jtopl'...
remote: Enumerating objects: 890, done.        
remote: Counting objects: 100% (17/17), done.        
remote: Compressing objects: 100% (12/12), done.        
remote: Total 890 (delta 9), reused 11 (delta 5), pack-reused 873        
Receiving objects: 100% (890/890), 27.97 MiB | 5.23 MiB/s, done.
Resolving deltas: 100% (555/555), done.
Cloning into '/MiSTeX/jtcores/modules/jttms'...
remote: Enumerating objects: 34, done.        
remote: Counting objects: 100% (34/34), done.        
remote: Compressing objects: 100% (26/26), done.        
remote: Total 34 (delta 5), reused 30 (delta 5), pack-reused 0        
Receiving objects: 100% (34/34), 11.91 MiB | 5.15 MiB/s, done.
Resolving deltas: 100% (5/5), done.
Cloning into '/MiSTeX/jtcores/modules/jt900h'...
ERROR: Repository not found.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
fatal: clone of 'git@github.com:jotego/jt900h.git' into submodule path '/MiSTeX/jtcores/modules/jt900h' failed
Failed to clone 'modules/jt900h' a second time, aborting
```

hansfbaier, 2024-04-16:
And the reason is: The jt900h  repository does not exist
[image]

jotego, 2024-04-17:
I know. That repository is hidden for now but it will come public soon. You can manually populate the submodules and everything but the JTNGP core will work correctly.

closed 2024-04-17 by jotego