jotego/jtcores #622: SiDi128: Volume of games

jtmiki, 2024-04-18:
I checked sound volumes of all games and this is the list.

**Sounds loud**
Game | Minimum volume | Maximum volume| Note
-- | -- | -- | --
1941 | 4 | 5 |  
1943KAI | 5 | 6 |  
1943 | 7 | 8 |  
1943MII | 6 | 7 |  
AFIGHTER | 4 | 6 |  
CWORLD2J | 5 | 6 |  
ALEXKIDD | 5 | 7 |  
ALIENS | 3 | 4 |  
BIONICC | 4 | 5 |  
BLKTIGER | 5 | 7 |  
BODYSLAM | 5 | 7 |  
BOXYBOY | 3 | 4 |  
CAWING | 3 | 5 |  
COMBATSC | 5 | 8 |  
COMMANDO | 4 | 6 |  
CONTRA | 4 | 5 |  
DDRAGON | 4 | 6 |  
DSPIRIT | 3 | 5 |  
DYNWAR | 6 | 7 |  
EXEDEXES | 3 | 7 |  
F1DREAM | 5 | 7 |  
FACEOFF | 3 | 5 |  
FANTZONE | 5 | 8 |  
FASTLANE | 4 | 6 |  
FFIGHT | 4 | 6 |  
GNG | 3 | 4 |  
GUNSMOKE | 4 | 6 |  
HCASTLE | 4 | 6 |  
KCHAMP | 5 | 8 |  
KARNOV | 4 | 6 |  
LWINGS | 4 | 7 |  
MTWINS | 4 | 5 |  
MERCS | 4 | 6 |  
MIKIE | 3 | 5 |  
NEMO | 3 | 5 |  
PANG3 | 3 | 5 |  
PARODIUS | 3 | 4 |  
PASSSHT | 3 | 4 |  
HIGEMARU | 1 | 3 |  
PNICKJ | 3 | 4 |  
POKONYAN | 3 | 5 |  
PUNKSHOT | 3 | 5 |  
QUARTET | 4 | 6 |  
QUARTET2A | 4 | 7 |  
QUESTER | 3 | 5 |  
QAD | 3 | 5 |  
QTONO2J | 3 | 5 |  
RASTAN | 3 | 4 |  
ROADF | 3 | 5 |  
ROCNROPE | 4 | 5 | ONLY INTRO PART
ROMPERS | 3 | 5 |  
SECTIONZ | 5 | 8 |  
SHINOBI | 3 | 6 |  
SIDEARMS | 3 | 6 |  
SPLATTER | 3 | 4 |  
SF2CE | 3 | 6 |  
SF2HF | 3 | 6 |  
SF2 | 4 | 6 |  
STRIDER | 2 | 4 |  
SPANG | 1 | 3 |  
TANKFRCE | 1 | 3 |  
TMNT | 3 | 5 |  
WOFCH | 3 | 4 |  
TETRIS | 3 | 5 |  
SIMPSONS | 3 | 5 |  
SRUMBLER | 7 | 8 |  
3WONDERS | 3 | 5 | ONLY INTRO PART
THUNDERX | 3 | 7 | ONLY INTRO PART
TIGEROAD | 4 | 7 |  
TTURFU | 3 | 5 |  
TROJAN | 2 | 4 |  
UNSQUAD | 4 | 6 |  
VARTH | 3 | 6 |  
VENDETTA | 3 | 5 |  
WILLOW | 3 | 4 |  
WNDPLNT | 3 | 5 |  
YIEAR | 3 | 5 |  

**Sounds low**

Game | Minimum volume| Maximum volume| Note
-- | -- | -- | --
ALIENSYN | 1 | 2 |  
ALTBEAST | 1 | 2 |  
AURAIL | 1 | 2 |  
BADDUDES | 1 | 2 |  
BOULDASH | 1 | 2 |  
BULLET | 1 | 2 |  
BERABOWMAN | 1 | 2 |  
CRIMFGHT | 1 | 2 |  
ESWAT | 1 | 2 |  
FANTZN2X | 1 | 2 |  
FPOINT | 1 | 2 |  
GOLDNAXE | 1 | 2 |  
HBARREL | 1 | 2 |  
HIPPODRM | 1 | 2 |  
KICKER | 1 | 2 |  
KNIGHTS | 1 | 2 |  
MMAZE | 1 | 2 |  
MEGAMAN | 1 | 2 |  
MIDRES | 1 | 2 |  
MBOMBRD | 1 | 2 |  
MVP | 1 | 2 |  
MX5000 | 1 | 2 |  
RENEGADE | 1 | 3 |  
ROBOCOP | 1 | 2 |  
SECRETAG | 1 | 2 |  
SF | 4 | 7 | SOUNDS LOW
FROUND | 1 | 2 |  
TNZS | 0 | 0 | IT SOUNDS
PUNISHER | 1 | 2 |  
WB3 | 1 | 2 |  

Commit: a9083e0

jotego, 2024-05-05:
@rp-jt, please adjust the core's sound level where possible so all cores have the same sound level. If the core is loud for some games but quiet for others, do not change it.

Sarvets, 2024-05-24:
Low sounds in games 5/24/24
CPS-2:
slighty quiet-
Marvel super heroes 
SF alpha 3
Super street fighter 2 new challengers

really quiet-
D&D shadow over mystara
Capcom sports club

other low volume games -
Double dragon 2
Fantasy zone 2 (16c system)
crime fighters 
ESWAT
sega ninja

this is what i found so far hope this helps! <3

jotego, 2024-07-03:
@jtmiki add the volume for each game in each core's TOML file using this syntax:

``` TOML
[Audio]
volume = [
    { machine="secretag", value=0x70 },
    { machine="bouldash", value=0x34 },
]
```

jtmiki, 2024-07-04:
[Audio]
volume = [
    { machine="secretag", value=0x70 },
    { machine="bouldash", value=0x34 },
    { machine="1943kai", value=0x60 },
    { machine="1942", value=0x98 },
    { machine="1943", value=0x51 },
    { machine="1943mii", value=0x63 },
    { machine="aceattaca", value=0x8D },
    { machine="afighter", value=0x59 },
    { machine="alexkidd", value=0x43 },
    { machine="astorm", value=0x60 },
    { machine="aliensyn", value=0x92 },
    { machine="aliens", value=0x7A },
    { machine="altbeast", value=0xFF },
    { machine="arknoid2", value=0x93 },
    { machine="aurail", value=0xF4 },
    { machine="baddudesl", value=0xFF },
    { machine="bakutotu", value=0x93 },
    { machine="bayroute", value=0x8F },
    { machine="bionicc", value=0x40 },
    { machine="blktiger", value=0x25 },
    { machine="blastoff", value=0x67 },
    { machine="blazer", value=0x69 },
    { machine="block", value=0xF3 },
    { machine="bloxeed", value=0xA7 },
    { machine="bodyslam", value=0x47 },
    { machine="bouldash", value=0xFF },
    { machine="bublbobl", value=0x83 },
    { machine="bullet", value=0xF3 },
    { machine="cworld", value=0x70 },
    { machine="chelnov", value=0x48 },
    { machine="berabohm", value=0xE7 },
    { machine="cltchitr", value=0x68 },
    { machine="combatsc", value=0x22 },
    { machine="commando", value=0x28 },
    { machine="contra", value=0x46 },
    { machine="cotton", value=0x70 },
    { machine="crimfght", value=0xE6 },
    { machine="ddcrew", value=0x7F },
    { machine="dangseed", value=0xAC },
    { machine="desertbr", value=0x40 },
    { machine="dokaben", value=0x4B },
    { machine="ddragon", value=0x50 },
    { machine="ddragon2", value=0xFF },
    { machine="drtoppel", value=0x66 },
    { machine="dspirit", value=0x4E },
    { machine="dunkshot", value=0xA0 },
    { machine="ddux", value=0xA9 },
    { machine="eswat", value=0xFF },
    { machine="exctleag", value=0xC2 },
    { machine="exedexes", value=0x32 },
    { machine="extrmatn", value=0x71 },
    { machine="f1dream", value=0x1D },
    { machine="f1dream", value=0x1D },
]
I checked until F-1 Dream.
Commit: 61be205

jtmiki, 2024-07-05:
[Audio]
volume = [
{ machine="secretag", value=0x70 },
{ machine="bouldash", value=0x34 },
{ machine="1943kai", value=0x60 },
{ machine="1942", value=0x98 },
{ machine="1943", value=0x51 },
{ machine="1943mii", value=0x63 },
{ machine="aceattaca", value=0x8D },
{ machine="afighter", value=0x59 },
{ machine="alexkidd", value=0x43 },
{ machine="astorm", value=0x60 },
{ machine="aliensyn", value=0x92 },
{ machine="aliens", value=0x7A },
{ machine="altbeast", value=0xFF },
{ machine="arknoid2", value=0x93 },
{ machine="aurail", value=0xF4 },
{ machine="baddudesl", value=0xFF },
{ machine="bakutotu", value=0x93 },
{ machine="bayroute", value=0x8F },
{ machine="bionicc", value=0x40 },
{ machine="blktiger", value=0x25 },
{ machine="blastoff", value=0x67 },
{ machine="blazer", value=0x69 },
{ machine="block", value=0xF3 },
{ machine="bloxeed", value=0xA7 },
{ machine="bodyslam", value=0x47 },
{ machine="bouldash", value=0xFF },
{ machine="bublbobl", value=0x83 },
{ machine="bullet", value=0xF3 },
{ machine="cworld", value=0x70 },
{ machine="chelnov", value=0x48 },
{ machine="berabohm", value=0xE7 },
{ machine="cltchitr", value=0x68 },
{ machine="combatsc", value=0x22 },
{ machine="commando", value=0x28 },
{ machine="contra", value=0x46 },
{ machine="cotton", value=0x70 },
{ machine="crimfght", value=0xE6 },
{ machine="ddcrew", value=0x7F },
{ machine="dangseed", value=0xAC },
{ machine="desertbr", value=0x40 },
{ machine="dokaben", value=0x4B },
{ machine="ddragon", value=0x50 },
{ machine="ddragon2", value=0xFF },
{ machine="drtoppel", value=0x66 },
{ machine="dspirit", value=0x4E },
{ machine="dunkshot", value=0xA0 },
{ machine="ddux", value=0xA9 },
{ machine="eswat", value=0xFF },
{ machine="exctleag", value=0xC2 },
{ machine="exedexes", value=0x32 },
{ machine="extrmatn", value=0x71 },
{ machine="f1dream", value=0x1D },
{ machine="fantzone", value=0x27 },
{ machine="fantzn2", value=0xFF },
{ machine="fastlane", value=0x30 },
{ machine="fpoint", value=0xFF },
{ machine="galaga88", value=0x75 },
{ machine="gbusters", value=0x34 },
{ machine="gng", value=0x7F },
{ machine="goldnaxe", value=0xA3 },
{ machine="gunsmoke", value=0x8C },
{ machine="hamaway", value=0x51 },
{ machine="hcastle", value=0x47 },
{ machine="hbarrel", value=0xFF },
{ machine="hippodrm", value=0xFF },
{ machine="hyperspt", value=0x7B },
{ machine="insectx", value=0xAD },
{ machine="kabukiz", value=0xFF },
{ machine="kageki", value=0x86 },
{ machine="kchamp", value=0x1D },
{ machine="karnov", value=0x82 },
{ machine="kicker", value=0xE3 },
{ machine="pingpong", value=0xC1 },
{ machine="lwings", value=0x40 },
{ machine="mia", value=0xBC },
{ machine="mjleague", value=0x84 },
{ machine="mmaze", value=0x40 },
{ machine="mwalk", value=0x7B },
{ machine="midres", value=0xFF },
{ machine="mikie", value=0x56 },
{ machine="mvp", value=0xF1 },
{ machine="mx5000", value=0xFF },
{ machine="outrun", value=0xEC },
{ machine="pacmania", value=0xA8 },
{ machine="pang", value=0x9A },
{ machine="parodius", value=0xB0 },
{ machine="passsht", value=0xC4 },
{ machine="higemaru", value=0x75 },
{ machine="pistoldm", value=0x6E },
{ machine="wldcourt", value=0xC6 },
{ machine="ws", value=0x8D },
{ machine="punkshot", value=0x60 },
{ machine="quartet2a", value=0x5C },
{ machine="quester", value=0x80 },
{ machine="rastan", value=0x71 },
{ machine="renegade", value=0xFF },
{ machine="riotcity", value=0xFC },
{ machine="roadf", value=0x4B },
]
I checked until ROAD FIGHTER.
Commit: 21fbc03

jtmiki, 2024-07-08:
[Audio]
volume = [
{ machine="secretag", value=0x70 },
{ machine="bouldash", value=0x34 },
{ machine="1943kai", value=0x60 },
{ machine="1942", value=0x98 },
{ machine="1943", value=0x51 },
{ machine="1943mii", value=0x63 },
{ machine="aceattaca", value=0x8D },
{ machine="afighter", value=0x59 },
{ machine="alexkidd", value=0x43 },
{ machine="astorm", value=0x60 },
{ machine="aliensyn", value=0x92 },
{ machine="aliens", value=0x7A },
{ machine="altbeast", value=0xFF },
{ machine="arknoid2", value=0x93 },
{ machine="aurail", value=0xF4 },
{ machine="baddudesl", value=0xFF },
{ machine="bakutotu", value=0x93 },
{ machine="bayroute", value=0x8F },
{ machine="bionicc", value=0x40 },
{ machine="blktiger", value=0x25 },
{ machine="blastoff", value=0x67 },
{ machine="blazer", value=0x69 },
{ machine="block", value=0xF3 },
{ machine="bloxeed", value=0xA7 },
{ machine="bodyslam", value=0x47 },
{ machine="bouldash", value=0xFF },
{ machine="bublbobl", value=0x83 },
{ machine="bullet", value=0xF3 },
{ machine="cworld", value=0x70 },
{ machine="chelnov", value=0x48 },
{ machine="berabohm", value=0xE7 },
{ machine="cltchitr", value=0x68 },
{ machine="combatsc", value=0x22 },
{ machine="commando", value=0x28 },
{ machine="contra", value=0x46 },
{ machine="cotton", value=0x70 },
{ machine="crimfght", value=0xE6 },
{ machine="ddcrew", value=0x7F },
{ machine="dangseed", value=0xAC },
{ machine="desertbr", value=0x40 },
{ machine="dokaben", value=0x4B },
{ machine="ddragon", value=0x50 },
{ machine="ddragon2", value=0xFF },
{ machine="drtoppel", value=0x66 },
{ machine="dspirit", value=0x4E },
{ machine="dunkshot", value=0xA0 },
{ machine="ddux", value=0xA9 },
{ machine="eswat", value=0xFF },
{ machine="exctleag", value=0xC2 },
{ machine="exedexes", value=0x32 },
{ machine="extrmatn", value=0x71 },
{ machine="f1dream", value=0x1D },
{ machine="fantzone", value=0x27 },
{ machine="fantzn2", value=0xFF },
{ machine="fastlane", value=0x30 },
{ machine="fpoint", value=0xFF },
{ machine="galaga88", value=0x75 },
{ machine="gbusters", value=0x34 },
{ machine="gng", value=0x7F },
{ machine="goldnaxe", value=0xA3 }, #Poor balance between background music and weapon sounds
{ machine="gunsmoke", value=0x8C },
{ machine="hamaway", value=0x51 }, #Attack and explosion sounds too low
{ machine="hcastle", value=0x47 },
{ machine="hbarrel", value=0xFF },
{ machine="hippodrm", value=0xFF },
{ machine="hyperspt", value=0x7B },
{ machine="insectx", value=0xAD },
{ machine="kabukiz", value=0xFF }, #Poor balance between background music and weapon sounds
{ machine="kageki", value=0x86 }, #Poor balance between background music and weapon sounds
{ machine="kchamp", value=0x1D },
{ machine="karnov", value=0x82 }, #Poor balance between background music and weapon sounds
{ machine="kicker", value=0xE3 }, #Poor balance between background music and weapon sounds
{ machine="pingpong", value=0xC1 },
{ machine="lwings", value=0x40 },
{ machine="mia", value=0xBC },
{ machine="mjleague", value=0x84 },
{ machine="mmaze", value=0x40 },
{ machine="mwalk", value=0x7B },
{ machine="midres", value=0xFF },
{ machine="mikie", value=0x56 },
{ machine="mvp", value=0xF1 },
{ machine="mx5000", value=0xFF },
{ machine="outrun", value=0xEC },
{ machine="pacmania", value=0xA8 },
{ machine="pang", value=0x9A },
{ machine="parodius", value=0xB0 },
{ machine="passsht", value=0xC4 },
{ machine="higemaru", value=0x75 },
{ machine="pistoldm", value=0x6E }, #Dialogue is very small
{ machine="wldcourt", value=0xC6 },
{ machine="ws", value=0x8D }, #Dialogue is very small
{ machine="punkshot", value=0x60 },
{ machine="quartet2a", value=0x5C },
{ machine="quester", value=0x80 },
{ machine="rastan", value=0x71 },
{ machine="renegade", value=0xFF },
{ machine="riotcity", value=0xFC },
{ machine="roadf", value=0x4B },
{ machine="robocop", value=0xFF },
{ machine="rocnrope", value=0xD6 }, #Poor balance between background music and weapon sounds
{ machine="rompers", value=0x77 },
{ machine="ryukyu", value=0xFF },
{ machine="sdi", value=0xCB },
{ machine="secretag", value=0xFF },
{ machine="sectionz", value=0x21 },
{ machine="shadowld", value=0x7E },
{ machine="shinobi", value=0x7C },
{ machine="sidearms", value=0x41 },
{ machine="sonicbom", value=0xAD }, #Poor balance between background music and weapon sounds
{ machine="splatter", value=0x5E }, #Poor balance between background music and weapon sounds
{ machine="sf", value=0xFF }, #Announcement audio is muffled
{ machine="sbasketb", value=0x7B }, #Announcement audio too low
{ machine="scontra", value=0xD9 },
{ machine="shangon", value=0xD9 },
{ machine="suprleag", value=0xF2 },
{ machine="spang", value=0x65 },
{ machine="tankfrce", value=0x8C },
{ machine="tmnt", value=0x66 },
{ machine="tetris", value=0x4C },
{ machine="fround", value=0xFA },
{ machine="tnzs", value=0xFF },
{ machine="simpsons", value=0x53 },
{ machine="srumbler", value=0x1C },
{ machine="thunderx", value=0x5E }, #Opening sound too loud
{ machine="tigeroad", value=0x21 },
{ machine="timescan", value=0xE1 },
{ machine="tokio", value=0xB2 },
{ machine="toryumon", value=0xFF },
{ machine="tturfu", value=0x8A },
{ machine="trackfld", value=0xB9 },
{ machine="trojan", value=0x7E },
{ machine="toutrun", value=0xA9 },
{ machine="vendetta", value=0x67 }, #Continuation sound effects too loud
{ machine="vigilant", value=0x59 }, #Continuation sound effects too loud
{ machine="vulgus", value=0xB6 },
{ machine="wwallyj", value=0x6E },
{ machine="wb3", value=0xEE },
{ machine="wndrplnt", value=0x60 },
{ machine="wrestwar", value=0x9C },
]
I checked all games except CPS1, CPS15 and CPS2.
Commit: c248da6

closed 2025-02-15 by jotego