jotego/jtcores #626: shouse: 4-player versions of Tank Force won't boot up

jotego, 2024-04-20:
This one was missed when support for other boot-up keys was added in #544. The four-player versions of Tank Force won't boot up. They currently boot to a "KEY ERROR", so it might be the key indeed but it could simply be that missing the joystick connections produce an error in the boot up sequence.

Adding these connections requires having the `JTFRAME_4PLAYERS` macro defined. That will make the other two joysticks appear as ports. MAME's `namcos1.cpp` file in the doc folder details how to connect the new joysticks.

gyurco, 2024-04-20:
The key error is simple, just need to change 'setname' to 'machine' in mame2mra.toml
```
diff --git a/cores/shouse/cfg/mame2mra.toml b/cores/shouse/cfg/mame2mra.toml
index ddb91eb3..1f85d4bb 100644
--- a/cores/shouse/cfg/mame2mra.toml
+++ b/cores/shouse/cfg/mame2mra.toml
@@ -44,7 +44,7 @@ data= [
     { machine="rompers",  data="03 B6 07 FF FF FF FF FF" },
     { machine="blastoff", data="03 B7 00 07 03 05 FF FF" },
     { setname="ws89",     data="03 B8 02 FF FF FF FF FF" },
-    { setname="tankfrce", data="03 B9 05 FF 01 FF 02 FF" },
+    { machine="tankfrce", data="03 B9 05 FF 01 FF 02 FF" },
     { machine="dangseed", data="03 34 06 FF 05 FF 00 04" },
     { machine="pistoldm", data="03 35 01 02 00 FF 04 FF" },
     { machine="ws90",     data="03 36 04 FF 07 FF 03 FF" },
```

However there are several games with missing/custom controls, like Tank Force 4p, Quester (paddle), Face Off (looks like the same as Tank Force 4p), Berabowman.

gyurco, 2024-04-20:
Should I send a PR with the toml change (looks easier to do the edit than preparing the PR).

jotego, 2024-04-20:
> Should I send a PR with the toml change (looks easier to do the edit than preparing the PR).

I just made the change. Thank you for identifying it. Could we keep this issue to implement the missing controls where necessary?

gyurco, 2024-04-20:
> I just made the change. Thank you for identifying it. Could we keep this issue to implement the missing controls where necessary?

Yes, sure.

gyurco, 2024-05-13:
I would accept some advice about controls:
- Faceoff/Tank Force 4p: Faceoff is actually 2 players, two sticks/player. I think it should use either left/right stick on the same controller, or joystick 1/2 and 3/4. Is there any method to select between the two methods?
- Berabowman: jump and attack buttons are actually pressure sensitive. In MAME, there are 3 buttons assigned to each move (like small jump/medium jump/high jump). Should I use 6 buttons, too?
- Quaster is a spinner user, I don't have a specific spinner, and as I see, the MiSTer handles them separately them from analog joystick. Maybe I can look at Arkanoid 2 how to use it.

jotego, 2024-05-13:
> * Faceoff/Tank Force 4p: Faceoff is actually 2 players, two sticks/player. I think it should use either left/right stick on the same controller, or joystick 1/2 and 3/4. Is there any method to select between the two methods?

Letting the user select is indeed a good idea. **JTKCHAMP** is a similar case. The selection is made through a new DIP switch defined in the TOML:

``` TOML
[dipsw]
extra=[
    { name="Link joysticks", options="Off,On", bits="8" },
]
```

To get access to the second analog stick you need to define the macro **JTFRAME_ANALOG_DUAL**. I will probably get rid of these macros soon as the game instance comes via a template now and that makes things much easier to maintain.

> * Berabowman: jump and attack buttons are actually pressure sensitive. In MAME, there are 3 buttons assigned to each move (like small jump/medium jump/high jump). Should I use 6 buttons, too?

That is a good solution. It requires changing the definition for the number of buttons in the .def file and then the button names in the TOML.

> * Quaster is a spinner user, I don't have a specific spinner, and as I see, the MiSTer handles them separately them from analog joystick. Maybe I can look at Arkanoid 2 how to use it.

The spinner is different from the analog joystick as it does not provide an absolute position. It only tells you whether it is spinning or not and the direction.

JTFRAME has a spinner emulator using the joystick and the mouse. So you can connect the spinner input even if you do not have a spinner.

Arcade games are a bit tricky when it comes to spinners. Sometimes there are resets for the spinner counts or other features that make it less generic than it should. A lot of these games used a small spinner controller that you can find instantiated in cores such as JTCPS1. MAME sometimes does not identify this controller correctly. But if it is there, adding support becomes very easy because it is just a matter of connecting it. Maybe there is some PCB picture where you can see the chip?

gyurco, 2024-05-13:
> Letting the user select is indeed a good idea. **JTKCHAMP** is a similar case. The selection is made through a new DIP switch defined in the TOML:
>
> ```toml
> [dipsw]
> extra=[
>     { name="Link [...]

I need only the digital part, as I see, access to these are not standardized (on MiST, it's joystick[19:16]).

gyurco, 2024-05-13:
> The spinner is different from the analog joystick as it does not provide an absolute position. It only tells you whether it is spinning or not and the direction.
>
> JTFRAME has a spinner emulator [...]

Quester seems to expect a 8 bit counter as spinner. But it's not hard to create one from the quadrature input (dial signal), however is there anything in the framework which I can re-use?

jotego, 2024-05-14:
> I need only the digital part, as I see, access to these are not standardized (on MiST, it's joystick[19:16]).

I do not remember seeing those bits in MiSTer/Pocket. [This](https://www.analogue.co/developer/docs/bus-communication#pad) is the reference for Pocket inputs. If it is not supported on MiSTer/Pocket, we should not go down that road.

Is it possible to simply use one joystick for control on Face Off?

> Quester seems to expect a 8 bit counter as spinner. But it's not hard to create one from the quadrature input (dial signal), however is there anything in the framework which I can re-use?

I think you probably need jt4701. Check out how to use it on jtcps1_main. It is also used on other cores.

gyurco, 2024-05-14:
> I do not remember seeing those bits in MiSTer/Pocket. [This](https://www.analogue.co/developer/docs/bus-communication#pad) is the reference for Pocket inputs. If it is not supported on [...]
Only half of the team is controllable with one stick. We can use joy1/2 for player1 and and joy3/4 for player2 then.

jotego, 2024-05-15:
Thank for the PR. @jtmiki, please try out the affected games so we can close the issue:

- faceoff
- quester
- berabohm

jotego, 2024-05-15:
What was the conclusion with 4-player version of Tank Force? I do not edits related to controls? (Although there was an edit for getting the right key)

gyurco, 2024-05-15:
Controls are the same as the 2-player version, just supports more controllers.

jotego, 2024-05-24:
Tank Force 4 player controls are not completed:

- missing credits for players 3/4
- controls for players 2/3 are swapped

You can test it in service mode, advancing with the service key (9) will show the controls window. All 4-player inputs are mapped to the keyboard in JTFRAME.

Credits for players 1/2 were swapped in all games -prior to this issue- I fixed that one in c694259f. Could you finish off these couple of things with Tank Force?

jotego, 2024-06-04:
@jtmiki please test Tank Force 4 player controls in today's compile_all.

jtmiki, 2024-06-05:
Hola Pepe,

I was going to test it but there wasn't Tank Force 4P in today's 
compile_all.

Miki

> @jtmiki [1] please test Tank Force 4 player controls in today's 
> compile_all.
> 
> [3].

-- 

Miki Saito
Gerente de JOTEGO
https://www.patreon.com/jotego

Links:
------
[1] https://github.com/jtmiki
[2] https://github.com/jotego/jtcores/issues/626#issuecomment-2146701506
[3] 
https://github.com/
--=_1e7dda3515b0c4ec0a43e695f583bd66
Content-Type: multipart/related;
 boundary="=_89cbb2514eee8cdf2b5d42381b532d26"

--=_89cbb2514eee8cdf2b5d42381b532d26
Content-Transfer-Encoding: quoted-printable
Content-Type: text/html; charset=UTF-8

<html><head><meta http-equiv=3D"Content-Type" content=3D"text/html; charset=
=3DUTF-8" /></head><body style=3D'font-size: 10pt; font-family: Verdana,Gen=
eva,sans-serif'>
<p>Hola Pepe,</p>
<p><br /></p>
<p>I was going to test it but there wasn't Tank Force 4P in today's compile=
_all.</p>
<p><br /></p>
<p>Miki</p>
<p><br /></p>
<p id=3D"reply-intro">El 2024-06-04 08:20, JOTEGO escribi&oacute;:</p>
<blockquote type=3D"cite" style=3D"padding: 0 0.4em; border-left: #1010ff 2=
px solid; margin: 0">
<div id=3D"replybody1">
<p><br /></p>
<p dir=3D"auto"><a class=3D"v1user-mention v1notranslate" href=3D"https://g=
ithub.com/jtmiki" target=3D"_blank" rel=3D"noopener ***@***.***</a>=
 please test Tank Force 4 player controls in today's compile_all.</p>
<p style=3D"font-size: small; -webkit-text-size-adjust: none; color: #666;"=
otego/jtcores/issues/626#issuecomment-2146701506" target=3D"_blank" rel=3D"=
noopener noreferrer">view it on GitHub</a>, or <a href=3D"https://github.co=
BGQIIUVGVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDCNBWG4YDCNJQGY" target=3D=
"_blank" rel=3D"noopener noreferrer">unsubscribe</a>.<br />You are receivin=
g this because you were mentioned.<img src=3D"https://github.com/notificati=
ons/beacon/A2NOQSHEHP4IJ5MZNUVPSHDZFVMCDA5CNFSM6AAAAABGQIIUVGWGG33NNVSW45C7=
OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTT76QIME.gif" width=3D"1" heig=
ht=3D"1" /><span style=3D"color: transparent; font-size: 0; display: none; =
visibility: hidden; overflow: hidden; opacity: 0; width: 0px; height: 0px; =
tcores/issues/626/2146701506</span><span>@</span><span>github</span><span>.=
</span><span>com&gt;</span></span></p>
</div>
</blockquote>
<p><br /></p>
<div id=3D"signature">-- <br />
<div class=3D"pre" style=3D"margin: 0; padding: 0; font-family: monospace">=
<span style=3D"font-size: 11pt;"><img src=3D"cid:16cfcc068d6e45e7e03d939754=
<div class=3D"pre" style=3D"margin: 0; padding: 0; font-family: monospace">=
<span style=3D"font-size: 11pt;">Miki Saito</span></div>
[... 155 lines cut ...]
eTIPE4haadeuXYk4LIgKPrDBzJkzeZhA1EJJOiyISkRgAxQUMrgRlQarnOnTp8sseYI+PIi6ePGi
GjVqFE9KiZqAoIZVTpIlJmMDbIBiI5Sdd4n0sKpBHWjSJSqwATZCGdyICiGohX4HtFSJWoo2dOjQ
ITV48GCZEaVbEmvVmpK4jC0HLVdwlE2UdghqSaxVa0piAxvgKJvBjdIsF9SSWKvWlEQHNmBwo7RK
a1CDxAc2YHCjtElzUINUBDZgcKO0SHtQg9QENsgFN/zFEyURSjqef/75VAc1SGy5R1PwdgIuz/OG
AiVJmurUmpOqjC0Hn2ZI1Zm5UVIwqDWWysAGCG5I2fENQRQy3P1E91sGtbxULkUbwsV5vMqzcOFC
+QpRONBoFdsq1FjqA1vOkiVL1KxZs2RG5D8chOFAjAoxsDWwe/duNWLECJkR+YnlHM1L7R6bDj79
Xn/9dR4qkLdYzlEaZmwaeNpv/vz53Hcjr6CV97Rp03hIUAIGtiYg3Z8yZYrMiNzACmLZsmXZjjVU
Gi5Fm4DTprq6Oi5NyRksPdFHjUGtZRjYmoG9DHxjsd6NbGN9WutxKdoCXJqSDRMnTlQLFizIPghO
rcOMrQWwND179qyqqKiQrxDFCwcEzzzzDINaGzFjayVmbxQn7OOuWbOGAS0mzNhaKXewwL03aqvK
ysrsPi6DWnyYscVg27Zt6uGHH2YbJGoR7qWZw8AWE1ymX7RokZo7d658hUgPy845c+bw8rpBDGwx
q62tVbNnz1bV1dXyFaI8HDwhqHXq1Em+QiYwsBmCC/UPPvggl6eUhYCG/VguO+1gYDMIy9MtW7Zk
750ywKUT9tEQ1NheyC4GNgsQ4NatW6fKy8vlK5R02EfDnisDmhsMbBaha8jatWvZ0DLBcgcDEyZM
4FUohxjYHGCASx4GNL8wsDmEALdp06ZsW3LuwYUJe2gzZsxQI0eOZEDzCAObB7AHt3PnTrV06VKW
iQQCBwIIatxD8xMDm2cOHTqkNm/ezEJfD2G5OX36dDVp0iS25vYcA5uncstUXLZnFucW6s/uuece
NWTIEC43A8HAFoDTp0+rl19+mXtxFuX2zoYOHcpbAgFiYAsMbjSgEwSXqvHLLTXHjh3LGwKBY2AL
FA4c8FQgHs1dvXo1M7lWwjITJ5rDhg1jMEsQBraEyC1Xcbq6YsUK+Srp4ERzzJgxXGYmGANbAuHg
4dixY9kl68aNG1OfzWG/DIEMr/z37t2bBwApwMCWAsjmampqsi2VDh8+nPiMDhlZ//791Te/+U3V
t29fZmUpxMCWUghyJ0+eDDrYYbN/9OjRnwexq6++mvVllMXARp/DEvaDDz7IBryPP/5YHT9+XJ0/
fz4b/FzV0mFzv2PHjqpHjx6qa9eu6sorr1RdunRR3bt355KSimJgo5LhJBYP2MCFCxfUiRMnsmPI
BcGWyAUr6NChg+rZs2d2DDyhpLZgYCOixOHze0SUOAxsRJQ4DGxElDgMbESUOAxsRJQ4DGxElDgM
bESUOAxsRJQ4DGxElDgMbESUOAxsRJQ4DGxElDgMbESUOAxsRJQ4DGxElDgMbESUMEr9P8W0hIeU
WNNmAAAAAElFTkSuQmCC
--=_89cbb2514eee8cdf2b5d42381b532d26--

--=_1e7dda3515b0c4ec0a43e695f583bd66--

jtmiki, 2024-06-06:
Hola Pepe,

He copiado los ficheros de Debug builds de hoy pero no esta Tank Force 
4P.

un saludo,

Miki

> Hola Pepe,
> 
> I was going to test it but there wasn't Tank Force 4P in today's 
> compile_all.
> 
> Miki
> 
> 
>> @jtmiki [1] please test Tank Force 4 player controls in today's 
>> compile_all.
>> 
>> [3].

Links:
------
[1] https://github.com/jtmiki
[2] https://github.com/jotego/jtcores/issues/626#issuecomment-2146701506
[3] 
https://github.com/
--=_a8c41e346cc48f74ce3010dd4fb00360
Content-Transfer-Encoding: quoted-printable
Content-Type: text/html; charset=UTF-8

<html><head><meta http-equiv=3D"Content-Type" content=3D"text/html; charset=
=3DUTF-8" /></head><body style=3D'font-size: 10pt; font-family: Verdana,Gen=
eva,sans-serif'>
<p>Hola Pepe,</p>
<p><br /></p>
<p>He copiado los ficheros de Debug builds de hoy pero no esta Tank Force 4=
P.</p>
<p><br /></p>
<p>un saludo,</p>
<p>Miki</p>
<p><br /></p>
<p id=3D"reply-intro">El 2024-06-05 10:21, Miki escribi&oacute;:</p>
<blockquote type=3D"cite" style=3D"padding: 0 0.4em; border-left: #1010ff 2=
px solid; margin: 0">
<div id=3D"replybody1">
<div style=3D"font-size: 10pt; font-family: Verdana,Geneva,sans-serif;">
<p>Hola Pepe,</p>
<p><br /></p>
<p>I was going to test it but there wasn't Tank Force 4P in today's compile=
_all.</p>
<p><br /></p>
<p>Miki</p>
<p><br /></p>
<p id=3D"v1reply-intro">El 2024-06-04 08:20, JOTEGO escribi&oacute;:</p>
<blockquote style=3D"padding: 0 0.4em; border-left: #1010ff 2px solid; marg=
in: 0;">
<div id=3D"v1replybody1">
<p><br /></p>
<p dir=3D"auto"><a class=3D"v1v1user-mention v1v1notranslate" href=3D"https=
://github.com/jtmiki" target=3D"_blank" rel=3D"noopener ***@***.***=
</a> please test Tank Force 4 player controls in today's compile_all.</p>
<p style=3D"font-size: small; -webkit-text-size-adjust: none; color: #666;"=
otego/jtcores/issues/626#issuecomment-2146701506" target=3D"_blank" rel=3D"=
noopener noreferrer">view it on GitHub</a>, or <a href=3D"https://github.co=
BGQIIUVGVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDCNBWG4YDCNJQGY" target=3D=
"_blank" rel=3D"noopener noreferrer">unsubscribe</a>.<br />You are receivin=
g this because you were mentioned.<img src=3D"https://github.com/notificati=
ons/beacon/A2NOQSHEHP4IJ5MZNUVPSHDZFVMCDA5CNFSM6AAAAABGQIIUVGWGG33NNVSW45C7=
[... 162 lines cut ...]
RkRxw77aI488EnRQg6ACG+AuKdsbEZmxaNGibAIRuuACG6C9EdJlIooPDuiGDx8us7AFtcfWEDru
jh8/Xu3bt0++QkSthURh+fLlMgtfsIENWN9G1HYh16sVE+RSNAf1begPRUStV1VVlaigBkEHNpg8
eTIPE4haadeuXYk4LIgKPrDBzJkzeZhA1EJJOiyISkRgAxQUMrgRlQarnOnTp8sseYI+PIi6ePGi
GjVqFE9KiZqAoIZVTpIlJmMDbIBiI5Sdd4n0sKpBHWjSJSqwATZCGdyICiGohX4HtFSJWoo2dOjQ
ITV48GCZEaVbEmvVmpK4jC0HLVdwlE2UdghqSaxVa0piAxvgKJvBjdIsF9SSWKvWlEQHNmBwo7RK
a1CDxAc2YHCjtElzUINUBDZgcKO0SHtQg9QENsgFN/zFEyURSjqef/75VAc1SGy5R1PwdgIuz/OG
AiVJmurUmpOqjC0Hn2ZI1Zm5UVIwqDWWysAGCG5I2fENQRQy3P1E91sGtbxULkUbwsV5vMqzcOFC
+QpRONBoFdsq1FjqA1vOkiVL1KxZs2RG5D8chOFAjAoxsDWwe/duNWLECJkR+YnlHM1L7R6bDj79
Xn/9dR4qkLdYzlEaZmwaeNpv/vz53Hcjr6CV97Rp03hIUAIGtiYg3Z8yZYrMiNzACmLZsmXZjjVU
Gi5Fm4DTprq6Oi5NyRksPdFHjUGtZRjYmoG9DHxjsd6NbGN9WutxKdoCXJqSDRMnTlQLFizIPghO
rcOMrQWwND179qyqqKiQrxDFCwcEzzzzDINaGzFjayVmbxQn7OOuWbOGAS0mzNhaKXewwL03aqvK
ysrsPi6DWnyYscVg27Zt6uGHH2YbJGoR7qWZw8AWE1ymX7RokZo7d658hUgPy845c+bw8rpBDGwx
q62tVbNnz1bV1dXyFaI8HDwhqHXq1Em+QiYwsBmCC/UPPvggl6eUhYCG/VguO+1gYDMIy9MtW7Zk
750ywKUT9tEQ1NheyC4GNgsQ4NatW6fKy8vlK5R02EfDnisDmhsMbBaha8jatWvZ0DLBcgcDEyZM
4FUohxjYHGCASx4GNL8wsDmEALdp06ZsW3LuwYUJe2gzZsxQI0eOZEDzCAObB7AHt3PnTrV06VKW
iQQCBwIIatxD8xMDm2cOHTqkNm/ezEJfD2G5OX36dDVp0iS25vYcA5uncstUXLZnFucW6s/uuece
NWTIEC43A8HAFoDTp0+rl19+mXtxFuX2zoYOHcpbAgFiYAsMbjSgEwSXqvHLLTXHjh3LGwKBY2AL
FA4c8FQgHs1dvXo1M7lWwjITJ5rDhg1jMEsQBraEyC1Xcbq6YsUK+Srp4ERzzJgxXGYmGANbAuHg
4dixY9kl68aNG1OfzWG/DIEMr/z37t2bBwApwMCWAsjmampqsi2VDh8+nPiMDhlZ//791Te/+U3V
t29fZmUpxMCWUghyJ0+eDDrYYbN/9OjRnwexq6++mvVllMXARp/DEvaDDz7IBryPP/5YHT9+XJ0/
fz4b/FzV0mFzv2PHjqpHjx6qa9eu6sorr1RdunRR3bt355KSimJgo5LhJBYP2MCFCxfUiRMnsmPI
BcGWyAUr6NChg+rZs2d2DDyhpLZgYCOixOHze0SUOAxsRJQ4DGxElDgMbESUOAxsRJQ4DGxElDgM
bESUOAxsRJQ4DGxElDgMbESUOAxsRJQ4DGxElDgMbESUOAxsRJQ4DGxElDgMbESUMEr9P8W0hIeU
WNNmAAAAAElFTkSuQmCC
--=_f043b658bef561ea049976d08a5dde19--

jotego, 2024-06-13:
tested Tank Force 4p (all 4 coin chutes and controls for all 4 players) and it works correctly. @jtmiki mark it as completed.

closed 2024-06-13 by jotego