jotego/jtcores #628: Simulation stopped working on Toki core

vertrex, 2024-04-21:
I got only black frame when launching simulation on toki core since this commit : 
https://github.com/jotego/jtcores/commit/44f50a555f99273b1ef5dad8997792d40e91759d

if I change the line back to : 
```rd_st[ cur_ba ] = burst_len+1;```
it work again

jotego, 2024-04-22:
That change came after reviewing the SDRAM controller with @gyurco and deciding that platforms that can work with zero shift on the SDRAM clock, should do so. This implied modifiying the SDRAM model used in verilator (hdl/test.cpp). I was worry of the change, but everything was working. I have just simulated JTSF, which also uses 96MHz as _toki_, and it boots up correctly. Other 48MHz cores also work fine.

Do you have the whole repository synchronized or are you picking up files in a different branch?

Could you check whether the CPU boots or not? Maybe this change brought up a weakness either in the framework or this particular core.

vertrex, 2024-04-22:
Normally the full repo is sync. I will check if cpu boot or it fail elsewhere

vertrex, 2024-04-22:
Simulation work in 48mhz with burst_len+2 (it seems more accurate than before) but not with burst_len+1 
For 96mhz it work with burst_len+1 but not burst_len+2.
The cpu doesn't seem to boot properly at first sight, I will  look more in details why.

jotego, 2024-05-07:
Fixed. An old hack in the verilog test bench was no longer needed. Removing it restored the simulation.

commit 499e9f4b (2024-05-07): no need for the hack anymore. Fixes #628