jotego/jtcores #629: Unable to load cartridges in NGP Analogue core

dji85, 2024-04-22:
Since the last few updates the function to load cartridges on the NGP core for Analogue Pocket does not work. After going through the NGP initial setup menu (language selection and setting time) and arriving at the NGP menu, from here in past versions you could load a cartridge from the Analogue menu. In the latest version this does not work, and you end up back at the NGP menu screen. Core version is 274dbaf and Pocket is running OS v2.2.

jotego, 2024-04-22:
Are you using the "virtual power button" to turn off the NGP before inserting the cartridge and then back on?

thunder-rabbit, 2024-04-22:
as soon as you get home I will try it--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

thunder-rabbit, 2024-04-22:
Hi Jotego,I have to go this waystart NGPChoose a RomSettingsChoose a RomRESETRESETNow it works but not all Games (Metal slug)RegardsSascha

jotego, 2024-06-29:
I am able to operate it, but the initial cartridge selection is ignored. If you boot up, then select a game using Analogue's OSD menu, and then turn off and on the core using the _virtual power_ button, the cartridge loads correctly.

closed 2024-06-29 by jotego