jotego/jtcores #631: Pantalla negra Super pang

Kaligulin, 2024-04-22:
Buenas, con el ultimo core jtpang al lanzar el super pang world hace la comprobación de ram ok 8 mhz original board y se queda pantalla negra o reiniciando en bucle. He borrado los mra la rom y el core sin solución, la única solución que encontré fue sustituir el core por uno anterior.
Saludos.

jotego, 2024-04-22:
Prueba a borrar los archivos de configuración. A veces se corrompen e impiden al core arrancar bien.

Kaligulin, 2024-04-29:
Hola, borrado los archivos de configuración y sigue el mismo problema, la solución la he encontrado al borrar los archivos nvram.
Gracias, por el soporte, lo replicare en el telegram de mister en español para futuros .
Saludos.

closed 2024-04-29 by Kaligulin

jotego, 2024-04-30:
Gracias por la información. Me alegro de que lo pudieras solucionar.