jotego/jtcores #636: Namco System-1 - Button names

Toryalai1, 2024-04-26:
The Namco system 1 games currently have "button 1" and "button 2" button labels.
Looking online, I can see that the cabinets had proper names. For example, Dragon spirit button labels are "Fire" and "Bomb".
https://www.flickr.com/photos/10707024@N04/34681730884

jotego, 2024-04-26:
@jtmiki, edit the button names around line number 21 of this file https://github.com/jotego/jtcores/blob/master/cores/shouse/cfg/mame2mra.toml

jtmiki, 2024-05-02:
I edited the button names. 
**Pro yakyuu world stadium** has three buttons to control.
**Face off** and **Pistol daimyo** have two functions on one button. A long press or repeatedly presses changes the type of shot.
Commit: c53a486

jtmiki, 2024-05-03:
All of edited button's names showed up well on the define buttons of each game.
Commit: d9bda02

jotego, 2024-05-06:
Maybe some buttons are swapped:

[image]

jtmiki, 2024-05-06:
Button's names of some games were swapped. I corrected them.
Commit: 579392f

closed 2024-05-10 by jotego