jotego/jtcores #644: s18: button names

jotego, 2024-05-03:
Please add button names for SEGA System 18 games [here](https://github.com/jotego/jtcores/blob/master/cores/s18/cfg/mame2mra.toml)

jtmiki, 2024-05-03:
I added button's names of most of the games.
I can't find **Pontoon** in the SEGA System 18.
I need time to find out the button name of **Where's Wally**. 
Commit: 65e99cb

Toryalai1, 2024-05-15:
>  I need time to find out the button name of **Where's Wally**. Commit: [65e99cb](https://github.com/jotego/jtcores/commit/65e99cb066bc9bd0915914da33d56fc19c00bb94)

Where's Wally uses one button which is called "Confirm". Found it here: https://dl.vgmdownloads.com/soundtracks/wally-wo-sagase-system-18-arcade/W1.png

jtmiki, 2024-05-16:
Thank you for the advice.

closed 2024-06-02 by jotego