jotego/jtcores #650: toki: PCM pitch wrong?

jotego, 2024-05-05:
Several people have reported that PCM/FX pitch is wrong. I have tried listening to it but cannot really tell. Maybe it's worth having a look, @vertrex

vertrex, 2024-05-05:
It's the case on pocket. It should work on mister but I will test to be sure, and try to find why this happened on pocket.

jotego, 2024-05-07:
The parameters for the _sdram.v file are not generated correctly:

``` VERILOG
jtframe_gated_cen #(.W(1),.NUM(1),.DEN(24),.MFREQ(24000)) u_cen0_clk48(
    .rst    ( rst          ),
    .clk    ( clk48 ),
    .busy   ( 1'b0    ),
    .cen    ( { oki_cen } ),
    .fave   (              ),
    .fworst (              )
); 
``

jotego, 2024-05-07:
@vertrex, please give it a go again (using the master branch) after the change I did in 32a2716d. The master clock frequency was not correctly identified for the case `JTFRAME_SDRAM96` was defined.

closed 2024-05-07 by jotego

vertrex, 2024-05-07:
It works on mister and pocket with the last master branch

PR #654 (2024-05-07): Toki : Fixes #650
Fixes #650

In mem.yaml : 

```
clocks:
  clk48:
    - freq:  1000000
      outputs:
       - oki
```
create a 500khz clock insted of a 1Mhz one , replaced the clock from mem.yaml with a derivated one create with jtframe_frac_cen
commit 32a2716d (2024-05-07): Correctly identify the 96MHz clock for CEN generation. Fixes #650