jotego/jtcores #656: shouse: sound RAM2 error

jotego, 2024-05-08:
Entering test mode displays a "SOUND RAM2 ERROR" message. This might be the reason why music/FX gets weird sometimes in rompers or the missing fire sound in bakutotu.

jotego, 2024-05-08:
RAM2 Error seems to refer to the Tri RAM. It currently has a hack around xwe, but removing it conditionally does not make the test pass:

``` VERILOG
wire xwe_aux = debug_bus[0] ? xwe : xwe && (xaddr !=0 || xdout == 8'ha6);

/* verilator tracing_off */
jtframe_dual_ram #(.AW(11)) u_ram(
    // Port 0: main and sub CPUs
    .clk0   ( clk   ),
    .data0  ( bdout ),
    .addr0  ( baddr ),
    .we0    ( bwe ),
    .q0     ( p_bdin  ),
    // Port 1
    .clk1   ( clk   ),
    .data1  ( xdout ),
    .addr1  ( xaddr ),
    .we1    ( xwe_aux ), // see https://github.com/jotego/jtcores/issues/410
    .q1     ( xdin  )
);
```

jotego, 2024-05-10:
This might be the reason why **dangseed** cannot enter service mode and stays trapped in a loop.

gyurco, 2024-05-10:
This:
```
diff --git a/cores/shouse/hdl/jtshouse_game.v b/cores/shouse/hdl/jtshouse_game.v
index f1338832..03b8bbbb 100644
--- a/cores/shouse/hdl/jtshouse_game.v
+++ b/cores/shouse/hdl/jtshouse_game.v
@@ -241,7 +241,7 @@ jtshouse_triram u_triram(
 
     .snd_cen    ( cen_snd   ),
     .mcu_cen    ( cen_mcu   ),
-    .snd_sel    ( snd_sel   ),
+    .snd_sel    ( cen_sndq  ),
 
     .baddr      ( baddr[10:0]   ),
     .mcu_addr   ( eerom_addr    ),
diff --git a/cores/shouse/hdl/jtshouse_triram.v b/cores/shouse/hdl/jtshouse_triram.v
index b2350135..253a63ba 100644
--- a/cores/shouse/hdl/jtshouse_triram.v
+++ b/cores/shouse/hdl/jtshouse_triram.v
@@ -123,7 +123,7 @@ jtframe_dual_ram #(.AW(11)) u_ram(
     .clk1   ( clk   ),
     .data1  ( xdout ),
     .addr1  ( xaddr ),
-    .we1    ( xwe && (xaddr !=0 || xdout == 8'ha6) ), // see https://github.com/jotego/jtcores/issues/410
+    .we1    ( xwe && (xaddr !=0 || xdout == 8'ha6 || xsel) ), // see https://github.com/jotego/jtcores/issues/410
     .q1     ( xdin  )
 );
```

fixes Rompers' service mode, but not dangseed's. I don't think the two issues have the same cause, the sound cpu doesn't even select tri-ram.

gyurco, 2024-05-10:
Not related, but I noticed that JTFRAME_LF_SDRAM_BUFFER is defined for the SiDi128 target. It's only needed for Outrun, isn't it?

gyurco, 2024-05-10:
Dangseed's reset loop is due to the watchdog.

gyurco, 2024-05-10:
But the watchdog kicks in because the CPU goes out of track.

gyurco, 2024-05-10:
If you keep pressing reset, eventually it'll enter service mode.

jotego, 2024-05-10:
> Not related, but I noticed that JTFRAME_LF_SDRAM_BUFFER is defined for the SiDi128 target. It's only needed for Outrun, isn't it?

Currently outrun is the only core using it, yes.

gyurco, 2024-05-10:
> > Not related, but I noticed that JTFRAME_LF_SDRAM_BUFFER is defined for the SiDi128 target. It's only needed for Outrun, isn't it?
> 
> Currently outrun is the only core using it, yes.

But then it was wrong to move it to the global target.def, because now all cores has 'no driver for SDRAM2_XXX'.

gyurco, 2024-05-10:
In this commit:
https://github.com/jotego/jtcores/commit/b649321a93d896936406748ba3482e23bffb9f20

jotego, 2024-05-10:
PR https://github.com/jotego/jtcores/pull/658 fixed the RAM2 error message. Thank you!
I thought this would fix #641 too, but it didn't. Bad luck! At least **dangseed** service mode works.

I have tested the service mode on a few more romsets. Some of them produce errors. Not sure whether these errors were there before the PR or not. This is not a comprehensive list, some games are not here:

dangseed  ok
splatter  ok
quester   ROM NO. 5 ERROR
mmaze     TEST PROGRAM INITIALIZE ERROR
wldcourt  ROM NO. 6 ERROR
ws89      ROM NO. 0 ERROR
bakutotu  ok
rompers   ok
pacmaniao ok
pistoldm  ok
shland    ok
soukobdx  ok
tankfrcej ok

jotego, 2024-05-10:
> > > Not related, but I noticed that JTFRAME_LF_SDRAM_BUFFER is defined for the SiDi128 target. It's only needed for Outrun, isn't it?
> > 
> > 
> > Currently outrun is the only core using it, yes.
> 
> But then it was wrong to move it to the global target.def, because now all cores has 'no driver for SDRAM2_XXX'.

Let's take that discussion to a new issue

gyurco, 2024-05-10:
> PR #658 fixed the RAM2 error message. Thank you! I thought this would fix #641 too, but it didn't. Bad luck! At least **dangseed** service mode works.
>
> I have tested the service mode on a few more [...]

At least not RAM errors :)
In mmaze and rompers, I hear music-like tunes, probably filters or MiSTer related bug?

gyurco, 2024-05-10:
I cannot reproduce the mmaze failure (it worked previously, it still works for me).

jotego, 2024-05-24:
For **ws**, mirroring the ROMs fixes the ROM error. In terms of the MRA file, it should look like:

```
        <part name="ws1_prg0.bin" crc="b0234298"/>
        <part name="ws1_prg0.bin" crc="b0234298"/>
        <part name="ws1_prg0.bin" crc="b0234298"/>
        <part name="ws1_prg0.bin" crc="b0234298"/>
        <part name="ws1_prg0.bin" crc="b0234298"/>
        <part name="ws1_prg0.bin" crc="b0234298"/>
        <part name="ws1_prg0.bin" crc="b0234298"/>
        <part name="ws1_prg0.bin" crc="b0234298"/>
        <part name="ws1_prg1.bin" crc="dfd72bed"/>
        <part name="ws1_prg1.bin" crc="dfd72bed"/>
        <part name="ws1_prg1.bin" crc="dfd72bed"/>
        <part name="ws1_prg1.bin" crc="dfd72bed"/>
        <part name="ws1_prg1.bin" crc="dfd72bed"/>
        <part name="ws1_prg1.bin" crc="dfd72bed"/>
        <part name="ws1_prg1.bin" crc="dfd72bed"/>
        <part name="ws1_prg1.bin" crc="dfd72bed"/>
        <part name="ws1_prg2.bin" crc="bb09fa9b"/>
        <part name="ws1_prg2.bin" crc="bb09fa9b"/>
        <part name="ws1_prg2.bin" crc="bb09fa9b"/>
        <part name="ws1_prg2.bin" crc="bb09fa9b"/>
        <part name="ws1_prg2.bin" crc="bb09fa9b"/>
        <part name="ws1_prg2.bin" crc="bb09fa9b"/>
        <part name="ws1_prg2.bin" crc="bb09fa9b"/>
        <part name="ws1_prg2.bin" crc="bb09fa9b"/>
        <part repeat="0x200000"> FF</part>
        <part name="ws1_prg7.bin" crc="28712eba"/>
        <part repeat="0x10000"> FF</part>
```

ws1_prg7.bin does not need the mirroring to pass the ROM check, though, although mirroring it once does not break either the ROM check or the MRA file.

I am going to check if this mirroring can be done by `jtframe mra`

jotego, 2024-05-24:
The ROM error problem was fixed by duplicating the files. A small change was introduced in `jtframe mra` to allow more files repetition, although for this case it was enough with the 2x already built it.

I cannot see the mmaze error now either, it might be related to the NVRAM.

I think we can close this issue. @jtmiki marked it as done. Thanks @gyurco

closed 2024-05-24 by jotego