jotego/jtcores #665: Toki dont work with 15khz intervideo arcade monitor

vastago, 2024-05-16:
[image]

jotego, 2024-05-17:
That image looks like the scan doubler was set. **Check your MiSTer settings please and make sure that the scan doubler is disabled.**

I have tested the core with a regular CRT TV set and it works for me. The video signals in simulation look reasonable. Horizontal time is a bit too long maybe, almost 65us rather than the typical 64us, but acceptable. The frame period is 16.77ms or 59.7Hz, which is good. I do not see a problem here.

I let the core author, @vertrex, comment further.

vastago, 2024-05-17:
It's strange because almost all the other cores work for me. This is my mister.ini in case someone wants to see my configuration.
[MiSTer.zip](https://github.com/jotego/jtcores/files/15344932/MiSTer.zip)

vertrex, 2024-05-19:
I have access to the board but don't know how can I measure HS & VS on iit. Can you help me to measure it ?

jotego, 2024-05-19:
Yes, you need either an oscilloscope or a "logic" probe. The [discovery 2](https://www.digikey.com/en/videos/d/digilent/analog-discovery-2-the-all-in-one-test-measurement-tool?utm_adgroup=&utm_source=google&utm_medium=cpc&utm_campaign=Pmax_Shopping_Boston%20Metro%20Category%20Awareness&utm_term=&utm_content=&utm_id=go_cmp-20837509568_adg-_ad-__dev-c_ext-_prd-_sig-Cj0KCQjwxqayBhDFARIsAANWRnQOQ6tZIbDxxMxQ8A_aWpogEMNbtVG3Ogcu_dn6VHPZD2xjkQoH1IIaAiouEALw_wcB&gad_source=1&gclid=Cj0KCQjwxqayBhDFARIsAANWRnQOQ6tZIbDxxMxQ8A_aWpogEMNbtVG3Ogcu_dn6VHPZD2xjkQoH1IIaAiouEALw_wcB) is one such device, but there are cheaper solutions that can work. I think you can even use an [Arduino UNO](https://store.arduino.cc/products/arduino-uno-rev3?_gl=1*1px5wm1*_ga*MTQ0NTQ0NDA0MS4xNzE2MTI5OTk5*_ga_NEXN8H46L5*MTcxNjEyOTk5OC4xLjEuMTcxNjEzMDAxMy4wLjAuMTEyNDk2OTc5NQ..*_fplc*YlplUzglMkJ0VVJXN2F5NUNSRVp5WlZXJTJGbEhocmRwUFpQc1IwMXN5VnY3WFNxUlNTa01MVmQlMkJYTXR5bElzcnEyaDl6UDQ4MTZ6YTlLcFM4ZGJlQSUyRnFOZkZIVEdNZDZ5Z0p3bGswMEduWCUyQmpISWNJcGhSNFBlQlZOdlVnR0ExZyUzRCUzRA..) for that with some clever programming.

The CSYNC pin is in the JAMMA connector, it contains the XOR of HS and VS. You need to measure the distance between HS pulses and their width. Same for VS. If you follow the trace back into the board, you will probably find the separated HS and VS signals. But it is possible to just measure CSYNC. CSYNC looks like this:

[image]

Image from this [web site](http://wla.berkeley.edu/~cs150/sp99/sp99/project/compvideo.htm)

Note that all time units will be a multiple of the pixel clock. Look for a crystal quartz in the board. The pixel clock will be a factor of the frequency shown there. Typical frequencies are 6 and 8MHz.

jotego, 2024-09-15:
I have made a shorter sync, which is more compliant with the standard.

Please @vastago try the attached version, which uses a shorter vsync. Both the old and new versions sync correctly at 15kHz with the hardware I have, so I cannot do much more than this.

[toki3v.zip](https://github.com/user-attachments/files/17005054/toki3v.zip)

closed 2024-09-15 by jotego