jotego/jtcores #667: cps1, 3wonders: layer priority problem, Lou's gun shown out of frame

jotego, 2024-05-19:
Core (4f7da4b)
[image]

PCB footage:
[image]

This may be related to #657 and #157.

jotego, 2024-05-19:
This was already reported and fixed once in https://github.com/jotego/jtcps/issues/131

JuJuden, 2024-05-19:
This topic has been debated for years:
https://mametesters.org/view.php?id=7606

It's hard to have a final final answer. Some are assuming the original game should have the gun shown out of frame and only bootlegs/repaired/converts are clipped.

The screen above comes from a board acquired more than 10 years ago when it was really affordable. It was sold as original, but I don't know its complete background history. Converts were already all around for more rare/known/expensive CPS1 games such as Daimakaimura or Strider. It's safe to assume it's not a convert, neither I can't say for sure all three A-B-C board are 100% genuine, if such thing exists in the first place considering how boards could have been swapped or reprogrammed back then. Also there might be some differences between the versions and revisions of the game.

However, assuming Japan may have less bootlegs/converts (which could be wrong as well), I looked for streams from Japanese Game Centers and found at least three videos from different Game Centers where the gun is clipped:
- Braf Soya Shinjuku Ôguchi Annex: https://youtu.be/MwN0coQmPGU?si=DOgcI5GZmcDbOra4&t=708
- Mat House: https://youtu.be/TcXRdGqDsGo?si=j12zgn7gZE2-OvBC&t=51
- Ekuru: https://www.youtube.com/live/F9GeHmWoXoY?si=G21UoYabH8Zp_6ec&t=23

It's also interesting to mention the ports history:
- Saturn Port from Xing (gun is not shown, no "frame" on the right either): https://youtu.be/lAsT3A0LZxk?si=7GpiNmzIMTgi9rGV&t=136
- PS1 Port from Xing (Same as Saturn): https://youtu.be/aevTaC_SZN0?si=KCdC0JmPNTDZc5Qa&t=117
- Capcom Classics Collection Vol.2 PS2 (Gun is shown out of frame): https://youtu.be/d0lGzTv2HAg?si=aID03ahIH4XEJM2W&t=4569
- Capcom Classics Collection Remixed PSP (Gun is shown out of frame): https://youtu.be/_nGRvsFqlN8?si=RgJdy4td-T5N8Gh9&t=337
- Capcom Arcade 2nd Stadium PS4 (gun not shown, frame on the right):  https://youtu.be/e6f9h8dHiqA?si=5Xyuo6y8YpY-A6CF&t=67

At this point, our best chance might be to try getting in touch with the devs and ask what was their original intent :)

jotego, 2024-05-19:
Thank you, very informative. Unless the ports are not emulators, they are not reliable.

The problem with layer priority is that it depended on the small C board if I remember correctly. It looks like over the years some C boards have ended up in the wrong place. The MAME code does not have a single answer for the position of some control bits that affect layer priority. The only way to be sure is to get the game with the right C board connected...

gyurco, 2024-08-06:
IMHO it looks better when the gun is out of the frame.

jotego, 2024-09-15:
Fixed in #802 by @paulb-nl

closed 2024-09-15 by jotego